atar-axis/xpadneo #364: Vibration not working

Massfug, 2022-05-30:
## Version of xpadneo

## Controller Model

- [ ] Xbox Series X|S controller

### Connection mode

- [ ] Bluetooth connection

## Protocol Information

- [ ] Steam Proton games are having issues

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] Fantastic work but ... 

## Describe the Bug

Everything works fine, but there is no vibration.
Previously, everything worked perfectly on ubuntu 20.04 and 22.04 with the same controller. Now I have arch installed and vibration does not work on it.

## System Information

```console
# uname -a
Linux a320m-h 5.18.0-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 24 May 2022 22:00:36 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8800831/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8800832/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/8800833/xpadneo-lsusb.txt)

kakra, 2022-05-30:
First:
```
[ 1848.774830] xpadneo 0005:045E:0B13.000A: BLE firmware version 5.09, please upgrade for better stability
```

Also, there's an issue in Steam Input which seems to prevent rumble from working. Try disabling Steam Input for your controllers in the Steam settings. Your controller will still work in games.

Massfug, 2022-05-30:
> First:
> 
> ```
> [ 1848.774830] xpadneo 0005:045E:0B13.000A: BLE firmware version 5.09, please upgrade for better stability
> ```
> 
Omg, thank you so much, now it works! And thanks for the fast reply!
Sorry for my stupid question :(

closed 2022-05-30 by Massfug

kakra, 2022-05-30:
> Sorry for my stupid question :(

There are no stupid questions, only stupid answers ;-)