atar-axis/xpadneo #366: Xbox Series X|S controller now works with Intel AX200 bluetooth

MageJohn, 2022-06-04:
## Version of xpadneo

AUR xpadneo-dkms-git: 0.9.r121.g727a84f-1

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Description

I'm submitting this report because the Intel AX200 chipset is currently reported as not working at all, and also so I have a record of what I did to make it work in the likely event I forget. I can pair the controller, turn it off, turn it on again, and it connects automatically again. I used the workaround from #336 and it all just works.

## Steps

These steps were performed with all-default settings in `/etc/bluetooth/main.conf`.

1. Start fresh and remove the the controller if it's already paired. In bluetoothctl: `remove <controller mac>`.
3. Put the controller in pairing mode: long press the wireless pairing button.
4. Turn on bluetooth scanning: `scan on` in bluetoothctl.
5. `scan off` when the controller appears. For me it's named `Xbox Wireless Controller`, making it easy to identify.
6. Pair the controller: `pair <controller mac>`. There was one instance where I had to do this twice before it succeeded. 
7. At this point the controller should stop flashing and give a rumble, and it should now be recognised fully as a gamepad by the system. 
8. Disconnect the controller: `disconnect <controller mac>` in bluetoothctl.
9. Edit `/var/lib/bluetooth/<adaptor mac>/<controller mac>/info` and remove the lines: 
    ```
    [IdentityResolvingKey]
    Key=<key>
    ```
10. Restart bluetoothd: `sudo systemctl restart bluetoothd.service`.
11. Make sure the bluetooth adaptor is on again: `power on` in bluetoothctl.
12. Turn on the controller. It should now connect automatically.

## System Information

```console
# uname -a
Linux 5.17.11-xanmod1-rog-1 #1 SMP Mon, 30 May 2022 18:51:12 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8836311/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8836312/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/8836313/xpadneo-lsusb.txt)

1985a, 2022-08-03:
I just update my controller on Windows and after that, I synchronize the controller again, and now is working even the rumble is working fine.

kakra, 2022-08-03:
Yep, it gets more and more clear that the primary issue is the controller firmware itself and since firmware 5.12, it seems rock solid on many Bluetooth chipsets.

ispanos, 2023-08-14:
> I just update my controller on Windows and after that, I synchronize the controller again, and now is working even the rumble is working fine.

I just bought a controller and its not even discoverable (using AX200). Do I need to install Windows to update the xbox controllers firmware? (I will try in a vm and pass-through the controller).

kakra, 2023-08-14:
You can use a Windows eval version booted in a VM to update the controller but it only works reliably using the Xbox dongle - otherwise the controller will disconnect from the VM as soon as it reboots into firmware update mode (it becomes a different USB device in that mode). It may work if your VM software allows to automatically pass through new USB devices.

If your controller is still on firmware 4.x, there's a good chance it will work after updating to 5.x.

ispanos, 2023-08-14:
> You can use a Windows eval version booted in a VM to update the controller but it only works reliably using the Xbox dongle - otherwise the controller will disconnect from the VM as soon as it [...]

I borrowed a Win. laptop and updated the firmware. I have installed xpadneo using a Copr on Fedora, but bluetoothctl doesn't even discover the device.

kakra, 2023-08-14:
Does it find other Bluetooth devices? If not, the radio may be disabled. Maybe come over to our Discord to get some help for this.

ispanos, 2023-08-14:
My headphones are working fine. What discord?

ispanos, 2023-08-14:
@kakra Ready when you are.

closed 2024-04-03 by github-actions[bot]