atar-axis/xpadneo #367: Xbox One S

Skullfurious, 2022-06-11:
## Version of xpadneo
hid-xpadneo-v0.9-96-g4fd620c

## Controller Model
Xbox One S Model 1708

* [ ] Xbox One S controller

### Connection mode

* [ ] Bluetooth connection

## Installed Software
* [ ] Steam Input (enabled by default via Steam Desktop client)
* [ ] (I disabled Xbox Configuration Support)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues

Please describe how it is failing below in the next sections.

## Severity / Impact

I would say the impact can be severe based on the game. For Hades for instance it's not a huge deal to not be able to see the boons screen but in Path of Exile which recently added controller support I cannot access my inventory.

- [ ] I've read the docs and the bug reporting instructions
- [ ] I can code and I want to help

## Describe the Bug

The share button doesn't appear to work as a hotkey in games. Basically games don't even detect it for remapping purposes and if they have mappings assigned to it by default they don't detect it being pressed. The button is not broken when using the native windows drivers and gaming on Windows.

## Steps to Reproduce

Unsure how to reproduce. Sorry.

## Expected Behavior

The button fires the event in game.

## System Information

```console
# uname -a

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
Linux skull-pop 5.17.5-76051705-generic #202204271406~1653440576~22.04~6277a18 SMP PREEMPT Wed May 25 01 x86_64 x86_64 x86_64 GNU/Linux

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8884524/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8884525/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/8884526/xpadneo-lsusb.txt)

Skullfurious, 2022-06-11:
Basically it wasn't working and following the issues to get log files appears to have fixed it. 

That being said I have had the share button not work on several occasions over the past week I just wasn't able to figure out if it was something to do with Steam or if it was something to do with my new dongle I purchased. 

If there's anything I can do the next time it happens please let me know.

kakra, 2022-06-12:
```
[260036.183314] loaded hid-xpadneo v0.9-96-g4fd620c
```
Please retry with the latest version.

Skullfurious, 2022-06-12:
Sure thing, will report if it happens again.

kakra, 2022-06-12:
Alternatively, try the stable branch (see releases, not using master) which had some fixes around the button mappings.

kakra, 2022-06-12:
I didn't realize PoE supports controllers now. I'll test that.

The dongle will have no impact on the functionality of buttons it self, it only affects if the controller connects reliably. For Steam, you might want to disable Steam Input for the controller (if you see mapping problems) but some games require at least the Steam Overlay enabled for controllers to work at all.

Nemirtingas, 2022-07-18:
Hi,

I have a problem too with the share button. It is not sent on the same input device as the gamepad.
For example:
I use evtest on /dev/input/event20: I have all the keys/abs/ff working, but no share button.
I use evtest on /dev/input/event21: I have only the share button working.

With xpad, I don't have the share button when wired connected, but I do have it on the same event device when connected in bluetooth but lacking rumble support :).

I'm using xpadneo on Ubuntu 20: hid-xpadneo, v0.9-121-g727a84f, 5.4.0-120-generic, x86_64

kakra, 2022-07-18:
> I have both EV_KEY/KEY_RECORD/value 0 and EV_KEY/KEY_RECORD/value 1 on press where I expected

I'm not yet sure why that happens. Could you try the v0.9 stable branch instead? It should map the Share button to F12 (for now, can be configured later). You download the stable snapshots from the releases page.

You're on v0.9-121 which probably is the devel branch towards v0.10 and includes some unstable features. v0.10 will probably move towards F12 mapping, too, unless I redo the HID patcher routines, then we might get a better behavior before v0.11.

Nemirtingas, 2022-07-19:
Ok, what about the 2 event devices for the same hardware device ? Is it by design or is it some bug ?
When I plug my XBox One controller via Bluetooth, it creates 2 event files. One for the gamepad (event20), and one for the share key(event21).

kakra, 2022-07-19:
It's by design, we cannot mix keyboard and gaming input in the same device. And actually, there's no official gamepad event for the Share button in the Linux kernel.

Nemirtingas, 2022-07-19:
Ok, cause the xpad driver has gamepad and keyboard in the same device.

kakra, 2022-07-19:
That may be a bug or using some special mode with overflowing event ids which evtest may not properly show... as far as my research went, you are not allowed to mix keyboard and gaming input in one device, and you may overflow some device category event ids but not all (that's actually discouraged as it makes it difficult for user-space to properly identify device classes).

In this case, it may actually be a bug because the share mapping id overflows into the next event category.

kakra, 2022-07-19:
But it looks like such a mapping patch has been accepted into the kernel: https://lore.kernel.org/all/20210414021201.2462114-1-lzye@google.com/

So I may consider it post-v0.9...