atar-axis/xpadneo #368: xpadneo module doesn't seem to be loading

Bucov, 2022-06-18:
## Version of xpadneo
xpadneo-dmks 0.9.1-1

from here https://aur.archlinux.org/packages/xpadneo-dkms

## Controller Model

- [ x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ x] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x ] I don't use Lutris, Bottles, wine or did not try
- [ x] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x ] I've read the docs and the bug reporting instructions
- [ x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I'm want to use xbox one s controller via bluetooth, but button mapping is wrong. After researching I found about this driver and that it is supposed to solve issues. Installed it via package manager, everything seems fine, but it doesn't seem to be loading into the system. (no vibration after connection is established)

```
modprobe xpadneo                                                                                                                                   127 ✘  3s  
modprobe: FATAL: Module xpadneo not found in directory /lib/modules/5.15.48-1-MANJARO
```
```
sudo modprobe hid_xpadneo                                                                                                                                   1 ✘ 
modprobe: FATAL: Module hid_xpadneo not found in directory /lib/modules/5.15.48-1-MANJARO
```
```
sudo modprobe hid-xpadneo                                                                                                                                   1 ✘ 
modprobe: FATAL: Module hid-xpadneo not found in directory /lib/modules/5.15.48-1-MANJARO
```
```
dkms status                                                                                                                                                 1 ✘ 
    ~                                                                                                                                                                ✔ 
```
```
sudo cat /etc/udev/rules.d/99-xpadneo.rules                                                                                                                 1 ✘ 
cat: /etc/udev/rules.d/99-xpadneo.rules: No such file or directory
```

```
sudo dmesg
usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  +0,001457] audit: type=1110 audit(1655555370.517:363): pid=22325 uid=1000 auid=1000 ses=4 msg='op=PAM:setcred grantors=pam_env,pam_fprintd acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  +0,001781] audit: type=1105 audit(1655555370.517:364): pid=22325 uid=1000 auid=1000 ses=4 msg='op=PAM:session_open grantors=pam_systemd_home,pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/0 res=success'
[  +3,940304] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +2,133981] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +1,919746] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +1,920218] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +2,133391] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +1,189209] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B20.0008/input/input26
[  +0,000541] hid-generic 0005:045E:0B20.0008: input,hidraw1: BLUETOOTH HID v5.13 Gamepad [Xbox Wireless Controller] on f8:89:d2:c7:86:74
[  +0,729660] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +4,053311] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +1,919957] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +1,919988] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +2,133371] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
[  +4,053356] rtw_8821ce 0000:03:00.0: firmware failed to leave lps state
```
## Steps to Reproduce

## Expected Behavior

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux milan-vostro153515 5.15.48-1-MANJARO #1 SMP PREEMPT Thu Jun 16 12:33:56 UTC 2022 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
 xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)                                                     1|0 ✔ 
zsh: correct 'xxd' to '_xxd' [nyae]? n
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
    ~  xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)                                                     1|0 ✔ 
zsh: correct 'xxd' to '_xxd' [nyae]? y
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8933544/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8933545/xpadneo-dmesg.txt)

## Additional Context

kakra, 2022-06-18:
It looks like the AUR package installation failed somehow or is partially broken (similar issues have been reported to our Discord for the xone project), please ask AUR to diagnose and fix it. Also, they should bump the version to v0.9.3.

Bucov, 2022-06-18:
Thanks, there are 2 of them
https://aur.archlinux.org/packages/xpadneo-dkms-git
https://aur.archlinux.org/packages/xpadneo-dkms

Tried both, but same problem. I'll try to build package from source to see if that fixes it.

DanielPower, 2022-06-19:
I was running into the same problem. In my case it turns out the dkms module was failing to build/install because I didn't have `linux-headers` installed.

Make sure you have the correct `-headers` package installed for the kernel you're using (`linux-headers`, `linux-lts-headers` or `linux-zen-headers`). Then re-install xpadneo. That fixed it for me.

deathxxx123, 2022-06-30:
The AUR dkms packages are not working well.
But the AUR git version is working good.
But best way is still to install from github archive here.

kakra, 2022-06-30:
I'd say this is completely an issue of the distribution, and the package maintainer needs to fix it. If the package is missing essential dependencies, then those need to be added by the maintainer. If the kernel does not properly provide the needed features (e.g., uhid built-in or autoloaded), then this needs to be addressed by the distribution kernel maintainers.

Thus, I'm closing this now.

closed 2022-06-30 by kakra