atar-axis/xpadneo #369: no vibration over bluetooth

jonfive, 2022-06-18:
## Version of xpadneo

aur/xpadneo-dkms 0.9.1-1
## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x ] Bluetooth connection
- [ ] USB cable (not yet supported) 
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ x -- disabled] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ x] Steam Proton games are having issues
- [ x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ x] I don't know how to ... 
I don't know how to properly search for the issue that i'm having to find relevant results.
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

While connecting over bluetooth, vibration does not work at all, not even on initial connection.
After plugging in the usb cable, all vibration works as expected.
Gamepadtester detects the controller _while plugged in via usb_ as 
"Microsoft Controller (STANDARD GAMEPAD Vendor: 045e Product: 0b12)".
_After_ i unplug the usb cable to return to bluetooth, i get the connection vibration jingle, but nothing more.
On bluetooth, gamepadtester detects the controller as
"Xbox Wireless Controller (STANDARD GAMEPAD Vendor: 045e Product: 02e0)"
## Steps to Reproduce

## Expected Behavior

Vibration no different than plugged in via usb

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
#Linux arch 5.18.5-zen1-1-zen
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

file:///home/jon/xpadneo-btmon.txt

file:///home/jon/xpadneo-dmesg.txt

## Additional Context

Aside the vibration, the controller works great over bluetooth. no disconnects, connects quickly.
Having previous issues with a t-rudder controller - could it possibly just be udev issues? The 60 rules are in place at the moment from install.
steam input checkboxes are all unchecked.

kakra, 2022-06-18:
Usually, I'd say most of the symptoms can be explained by having "Steam Input" enabled. But (a) you state that it is disabled and (b) it won't affect the initial connection vibration.

So let's look for some other clues:

1. Your Bluetooth settings could be incompatible (look for BLE interval settings in "Troubleshooting", and ensure to not enable the old ERTM workaround of xpadneo that was needed before kernel 5.12).
2. Your xpadneo version is too old (it looks like it misses the latest SDL fixes, v0.9.3 is the current version).
3. Your dongle may be incompatible (some Intel chipsets have general issues with the gamepad, it also affects Windows machines). This can be influenced by problem 1.
4. Communication over Bluetooth does not work beyond initial connection (try running `evtest` to see if you get input reports for your gamepad, but you already wrote it works fine except vibration). This is usually related to problem 3.

If you used custom udev rules, please ensure that they don't affect the gamepad itself. If unsure, please post those rules. Our own rules only affect the gamepad itself.

Actually, I believe v0.9.3 may fix your issues, it has fixes for both rumble throttling with 5.x firmware and SDL fixes to properly map all buttons: https://github.com/atar-axis/xpadneo/releases. You should ask AUR to bump to the latest version before trying other troubleshooting steps.

jonfive, 2022-06-18:
I no longer have any custom udev rules that i've personally entered. 
I uninstalled 0.9.2, grabbed the 0.9.3 release you linked to, ./installed it and rebooted. Same issue persists. 

I'll keep searching around and trying different things. Maybe i'll give it a whirl on a live usb and see if that works. If it works i can work through the differences between the 'clean' os vs one i've been working with for quite some time.

Thanks a ton for your explanations and ideas! :)

closed 2022-06-29 by jonfive