atar-axis/xpadneo #370: WARNING: kernel uhid module not found, controller firmware 5.x will not be supported

deathxxx123, 2022-06-22:
## Version of xpadneo
Lastest from git pull manual install.
v0.9-121-g727a84f

## Controller Model
- [ ] Xbox One S controller

### Connection mode
- [ ] Bluetooth connection

## Installed Software
Steam

## Protocol Information

## Severity / Impact

- [ ] I've applied the latest firmware update to the controller

## Describe the Bug (during install output)
WARNING: kernel uhid module not found, controller firmware 5.x will not be supported

## Steps to Reproduce
sudo pacman -S dkms linux-headers bluez bluez-utils
git clone https://github.com/atar-axis/xpadneo.git
cd xpadneo
sudo git config --global --add safe.directory $PWD
sudo ./install.sh

## Expected Behavior

## Screenshots / GIFs / Videos

## System Information
# uname -a
Linux x570saoruseliteax 5.17.15-1-MANJARO #1 SMP PREEMPT Wed Jun 15 07:09:31 UTC 2022 x86_64 GNU/Linux
```

<!-- Paste the output below the line prepended with # -->
```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```
zsh: correct 'xxd' to '_xxd' [nyae]? y
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

```

## Controller and Bluetooth Information

## Additional Context
It's working fine, but i got this warning when install xpadneo from git.
And yes, i updated controller to the lastest firmware with Win PC.
Where i can find and install this 'kernel uhid module'?

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/8955851/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/8955852/xpadneo-dmesg.txt)

kakra, 2022-06-23:
If the controller works fine with, then uhid detection may not be working properly. Currently, it looks for `/sys/devices/virtual/misc/uhid`. Maybe that file is only present when the uhid module was already loaded. We could probably also grep for `CONFIG_UHID` in `/proc/config.gz` but not all kernels show this file.

kakra, 2022-06-23:
Can you please try, from a freshly booted system with the controller turned off, if running `modprobe uhid` before the installation fixes the warning?

deathxxx123, 2022-06-24:
> Can you please try, from a freshly booted system with the controller turned off, if running `modprobe uhid` before the installation fixes the warning?

You are the man!
This is absolute right.
i do:
sudo modprobe uhid
before install, and no warning. I try 2 times to be sure.
Without this i got warning again.

kakra, 2022-06-27:
Your kernel should have the uhid module built in, or it should load it before running bluez.

deathxxx123, 2022-06-27:
Hmm i was confused when install lastest verion(9.3), but shows: v9.2
And i look into VERSION into archive and
v9.2 and v9.3 are reported to be v9.2 in VERSION file.

deathxxx123, 2022-06-27:
I do some checks and i install v9.3 from zip archive, witch reported to be a v9.2 have no warnings about UDEV and when connect rumble only one time. I test on Pop!_OS 22.04 and Manjaro Gnome.
I don't know why
git clone
get the v0.9-121-g727a84f
But lastest from archive seems don't have any problems.

kakra, 2022-06-27:
> Hmm i was confused when install lastest verion(9.3), but shows: v9.2 And i look into VERSION into archive and v9.2 and v9.3 are reported to be v9.2 in VERSION file.

Oops... Will fix.

kakra, 2022-06-27:
> witch reported to be a v9.2 have no warnings about UDEV

Yep, that's fixed, too, now. v0.9.2 and v0.9.3 didn't include that check yet. The latest release does.

deathxxx123, 2022-06-28:
Thank you for fast work! I'm glad i can help in some way. Still best driver anyway.

deathxxx123, 2022-06-28:
I try 9.4 and all looking fine. I got UDEV warning, but after install and reboot, i got only 1 time rumble when connected. And that is cool!

closed 2022-07-04 by kakra