atar-axis/xpadneo #373: [QUESTION] Gamepad Axes Are Swapped, Seemingly Unresponsive AND Strange Behavior on Ubuntu and Chrome

funminion, 2022-07-13:
I use recently purchased Xbox Core controller (official. sold by Microsoft on Amazon) on Ubuntu 18.04 with the latest Chrome. I used your driver to make it work finally. I get all buttons messed up in cloud games (tried several). jstest-gtk and joystick web testing website show the buttons are messed up too.  At the same time Steam games work perfectly fine with valve default configs. Also I tested same controller on my macbook with chrome for cloud games - it works fine for both stadia and xcloud. In your troubleshooting you say "IMPORTANT NOTE: ... There is currently no known work-around." Does it mean I do not need to go through debugging and give up on that?

kakra, 2022-07-14:
> "IMPORTANT NOTE: ... There is currently no known work-around."

You mean gamepad support in Chrome browsers?

I will work on this problem during v0.10, and it should land in v0.11 in the worst case. The current work around is to use the controller in wired mode for the browser.

There is a possible work-around using jscal and remapping the buttons order to match what Chrome expects. But that will mess up any other software which uses the jsdev devices (SDL does not, thus Steam and games running through Proton are not affected, native games using SDL are neither affected). The procedure with jscal is described in troubleshooting but you will most likely need to use a different order of buttons and insert fake buttons to account for the buttons that Chrome skips. When I tried that in the past, jscal usually crashed so I'm not sure if there is a way.

Chrome tries to re-implement most of the same abstraction that SDL does but they do it in an invalid way by reading mappings from evdev, then using some hard-coded transformation tables per gamepad model and finally reading values from jsdev, instead of either using evdev only (which would work perfectly fine because it also returns values), or reading the button and axis names AND values from jsdev only (I've looked at the source code of Chromium, it's a mess).

As a second option, Firefox doesn't seem to show this completely broken behavior but I'm not sure if it supports the cloud gaming services you're using.

A third option may be to try using the Xbox360 emulation branch from this repository. It may fake the model well enough for Chrome to not apply its crazily broken assumptions to the mapping.

funminion, 2022-07-14:
I tried it on Firefox - same issues. it sounds like OS/driver issue. I uninstalled v.9 and installed v.8 of the driver - web works but Steam now has this messy buttons issue! So I decided to use XBox controller for web and old PS3 controller for Steam on Ubuntu for the time being. Looking forward to the next version of your driver!

kakra, 2022-07-15:
BTW: "Xbox Core" - is this a new controller model? Could you post the diagnose data as requested by the issue reporting wizard?

funminion, 2022-07-15:
> BTW: "Xbox Core" - is this a new controller model? 

xbox core - seems to be one of two currently sold, core and elite

> Could you post the diagnose data as requested by the issue reporting wizard?

sorry, I do not have time for that, busy with work + I got 2nd core controller with usb to usb-c cable, it works fine with cable so I ordered another very long cable for 1st controller.

cerealbox, 2022-08-08:
i just wanted to second that i have encountered this issue as well.
gamepad tester (https://gamepad-tester.com) in chrome has some buttons swapped or missing.
appears to works fine in firefox and, interestingly, it used to work in chrome as well so a chromium update in the last year or so could be to blame.

i also encounter the same button mapping problem in sdl2 games as well unless i export the below:
`SDL_JOYSTICK_HIDAPI=0`

kakra, 2022-08-08:
Chrome is mostly unfixable as it considers the Xbox controllers to have dead buttons which our driver does not have. The issue with SDL is known and `SDL_JOYSTICK_HIDAPI=0` is the recommended workaround for the moment, it will be fixed in v0.10 or v0.11. See also here: #286

Darune, 2022-09-03:
In the mean time #286 gets implemented, I successfully mitigate the issue by using #283 with xpad_emulation=1 if anyone is interested

closed 2022-09-15 by kakra