atar-axis/xpadneo #374: Implement proper Impulse Triggers for Xbox controllers.

Giovani1906, 2022-07-30:
## Version of xpadneo

v0.9.4

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [x] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature request

- [ ] I'd like to see support for a specific model
- [x] I'd like to see support for a specific hardware feature
- [ ] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**
I'd like that Xbox controllers to have a proper Impulse Triggers rather than what currently is done which is synthesizes the effect.
The current synthesized effect is inaccurate and the vibrations seem to strong at times.

**Describe the solution you'd like**
I'd like that the Impulse Triggers would be implemented using the following kernel patches:
- https://www.spinics.net/lists/linux-input/msg77720.html
- https://www.spinics.net/lists/linux-input/msg77721.html
- https://www.spinics.net/lists/linux-input/msg77722.html
- https://www.spinics.net/lists/linux-input/msg77723.html
- https://www.spinics.net/lists/linux-input/msg77724.html

**Describe alternatives you've considered**
Currently there are no alternatives that I have heard of.

**Additional context**
The maintainer kakra has asked me to make an issue in other for them to keep track of it.

kakra, 2022-08-01:
> The current synthesized effect is inaccurate and the vibrations seem too strong at times.

You can reduce the amount of vibration effect using `rumble_attenuation=0,50` as a module parameter to reduce the rumble effect for the main rumble by 0% and for the trigger rumble by 50%. This is not a solution to the problem but it may improve your experience until proper support is in place.

The Arch wiki has information on how to use module parameters and set them on boot:
https://wiki.archlinux.org/title/Kernel_module

adamdmoss, 2024-05-24:
Is the trigger rumble strength supposed to be proportional to the body rumble strength?  I have an issue(?) where I can scale the rumble strength in a game's settings (Saints Row in this case) and that appears to work for the body rumble, but the trigger rumble appears to remain full strength.

kakra, 2024-05-24:
The trigger rumble adjusts to the strength of the main rumble. But the trigger rumble motors are pretty intense, you can scale them down with the example from https://github.com/atar-axis/xpadneo/issues/374#issuecomment-1200894085.

So yes, the trigger rumble strength is proportional to the main rumble.

The formula goes like this:

The game sends rumble strength for the weak (W) and strong (S) motor - if you made it weaker within the game, these values will be lower. We take the maximum of both values for our main (M) rumble strength: `M = max(W,S)`.

Now we attenuate the W value by the parameter given to the module (a0), and we do the same for the S value. Both new values will be send to the rumble motors: `(Wa,Sa) = (W,S) * (100-a0)%`.

Now we use the M value to calculate the trigger motor strength by multiplying the trigger pressure with value M, and additionally attenuate by either `a0`, or attenuate by `a0%*a1%`, if the second parameter value is given.

So attenuation `0,50` lowers the main motors by 0%, and the trigger motors by 50%. If you'd use `50,50`, it would effectively lower the main motors by 50%, and the trigger motors by an additional 50% (thus 75% effectively).

I hope this explains it well enough. I've double-checked the code, it should work correctly. But feel free to check yourself, the code starts here:
- https://github.com/atar-axis/xpadneo/blob/73be2eb22ee7b3b3f66b01ad8b9aaa4edc9b93b5/hid-xpadneo/src/hid-xpadneo.c#L300

adamdmoss, 2024-05-24:
Thanks for the explanation!  So it sounds like it's just a lot stronger than I'd expect by default even when the app thinks it's scaled way down, but this is expected. :)

kakra, 2024-05-27:
> Thanks for the explanation! So it sounds like it's just a lot stronger than I'd expect by default even when the app thinks it's scaled way down, but this is expected. :)

The problem is: The app doesn't know anything about the trigger rumble - we synthesize the effect from existing main rumble effects. Maybe we should attenuate it down a whole lot by default. The trigger rumble motors can be really really pronounced, especially at some different pressures you apply to them while they rumble: There's kind of some resonance effect then.

Giovani1906, 2024-06-21:
I've managed to make trigger rumble/impulse triggers work by using the patch below and having SDL patched for support as well. 
```diff
diff --git a/hid-xpadneo/src/hid-xpadneo.c b/hid-xpadneo/src/hid-xpadneo.c
index 1a25631..8660864 100644
--- a/hid-xpadneo/src/hid-xpadneo.c
+++ b/hid-xpadneo/src/hid-xpadneo.c
@@ -290,7 +290,7 @@ static int xpadneo_ff_play(struct input_dev *dev, void *data, struct ff_effect *
 	unsigned long flags, ff_run_at, ff_throttle_until;
 	long delay_work;
 	int fraction_TL, fraction_TR, fraction_MAIN, percent_TRIGGERS, percent_MAIN;
-	s32 weak, strong, direction, max_main;
+	s32 weak, strong, trigger_left, trigger_right, direction;
 
 	struct hid_device *hdev = input_get_drvdata(dev);
 	struct xpadneo_devdata *xdata = hid_get_drvdata(hdev);
@@ -301,6 +301,8 @@ static int xpadneo_ff_play(struct input_dev *dev, void *data, struct ff_effect *
 	/* copy data from effect structure at the very beginning */
 	weak = effect->u.rumble.weak_magnitude;
 	strong = effect->u.rumble.strong_magnitude;
+	trigger_left = effect->u.rumble.trigger_left;
+	trigger_right = effect->u.rumble.trigger_right;
 
 	/* calculate the rumble attenuation */
 	percent_MAIN = 100 - param_rumble_attenuation[0];
@@ -308,7 +310,6 @@ static int xpadneo_ff_play(struct input_dev *dev, void *data, struct ff_effect *
 
 	percent_TRIGGERS = 100 - param_rumble_attenuation[1];
 	percent_TRIGGERS = clamp(percent_TRIGGERS, 0, 100);
-	percent_TRIGGERS = percent_TRIGGERS * percent_MAIN / 100;
 
 	switch (param_trigger_rumble_mode) {
 	case PARAM_TRIGGER_RUMBLE_DIRECTIONAL:
@@ -378,12 +379,6 @@ static int xpadneo_ff_play(struct input_dev *dev, void *data, struct ff_effect *
 		break;
 	}
 
-	/*
-	 * we want to keep the rumbling at the triggers at the maximum
-	 * of the weak and strong main rumble
-	 */
-	max_main = max(weak, strong);
-
 	spin_lock_irqsave(&xdata->ff_lock, flags);
 
 	/* calculate the physical magnitudes, scale from 16 bit to 0..100 */
@@ -394,9 +389,9 @@ static int xpadneo_ff_play(struct input_dev *dev, void *data, struct ff_effect *
 
 	/* calculate the physical magnitudes, scale from 16 bit to 0..100 */
 	update_magnitude(xdata->ff.magnitude_left,
-			 (u8)((max_main * fraction_TL + S16_MAX) / U16_MAX));
+			 (u8)((trigger_left * fraction_TL + S16_MAX) / U16_MAX));
 	update_magnitude(xdata->ff.magnitude_right,
-			 (u8)((max_main * fraction_TR + S16_MAX) / U16_MAX));
+			 (u8)((trigger_right * fraction_TR + S16_MAX) / U16_MAX));
 
 	/* synchronize: is our worker still scheduled? */
 	if (xdata->ff_scheduled) {

```
The SDL2 patch can be found at CachyOS/CachyOS-PKGBUILDS#247.
It contains a Linux headers patch as well, otherwise SDL2 won't compile properly.
I've also made a usable patch file at CachyOS/kernel-patches#64.

kakra, 2024-06-21:
Please create a PR for this patch: It will have some issues we need to address.

peterix, 2025-07-03:
Well, the idea would be to tell the game/user application that it can use trigger rumble somehow, because if I have it, I can use it (hopefully well), and if I don't have it, I can fall back to normal rumble or something else (visual feedback, sound, etc.).

kakra, 2025-07-03:
> I've managed to make trigger rumble/impulse triggers work by using the patch below and having SDL patched for support as well.

@Giovani1906 This patch is wrong. If you get direct values for trigger rumble, don't scale them with the trigger pressure fractional... This fractional is purely used for synthesizing a trigger rumble effect by using the pressure applied to the trigger so it becomes more dynamic, and so it doesn't rumble when not touched.

So your patch should either remove the synthesizing part completely and only provide trigger rumble from user-space, or it should flip a boolean once it detects trigger rumble from user-space and use a non-synthesizing code path. The latter is what I'm going to implement.

Thanks for the patch, tho. The general idea is useful.

@peterix On another note, I wonder if we could expose a sysfs file instead (I'm going to add a sysfs interface anyways) to detect features. Or if we could add an IOCTL to query trigger rumble, and maybe even support multi-motor rumble through such an IOCTL interface. I've never add my own IOCTL to a driver yet, tho.

mr-falafel, 2026-08-24:
Hi, I've noticed that xpadneo seems to apply trigger rumble/impulse trigger in a way that's different than on Windows. My understanding is that, on Windows, impulse triggers is only activated in a small selection of games that natively support it, with the purpose of simulating specific vibrations of, for example, a gun trigger or a car pedal. Using xpadneo, I have found impulse triggers is frequently activated when using programs such as RPCS3, in a way that I find quite unexpected and unappealing. Is that because it is activated whenever the regular small motors are activated? Or indeed the large motors? I would have thought it would be better / more accurate to only activate impulse triggers when a game/application specifically signals for it, rather than tying it to either of the other motors. If that means it's never activated, I would find that preferably to having it activated by other rumble triggers.

Am I misunderstanding the situation? Is there a possibility of making that an option for users like me?

kakra, 2026-08-24:
The Linux kernel lacks the API to support trigger rumble through its standard force feedback drivers. In theory, using hidraw mode, wine could offer native trigger rumble support. I've not looked if SDL or wine actually support this, but currently hidraw is disabled in xpadneo. I will look into that after we migrated xpadneo to support hidraw mode (post v0.11/v0.12).

For now, xpadneo synthesizes trigger rumble by combining your trigger pressure with the higher value of both main rumble motors. If the effect is too strong, you can attenuate it. The motors feeling strong mostly result from their small size (they can spin fast, and spin up very fast) and the very small weight of the triggers itself, so these tiny motors do not need to vibrate a lot of mass. You can adjust their attenuation with the second optional attenuation parameter:

```bash
# reduce trigger strength by 33%
echo 0,33 | sudo tee /sys/module/hid_xpadneo/parameters/rumble_attenuation
```

If a future kernel starts supporting a trigger rumble interface, I have ideas how to support both modes at the same time.

Thinking about it, it may make sense to make the pressure parameter non-linear. But floating point math cannot be used in the kernel, so I need to look up if there's an integer-based power function.

mr-falafel, 2026-08-25:
> For now, xpadneo synthesizes trigger rumble by combining your trigger pressure with the higher value of both main rumble motors. If the effect is too strong, you can attenuate it. The motors feeling [...]

Ah, thank you for clarifying that. I'm happy to hear that the effect can be configured. I will probably disable it for now. The issues with Impulse Triggers, as they are currently implemented, are three-fold for me.

Firstly, I dislike the idea of getting a rumble effect that wasn't intended by the games' developers. 
Secondly, I dislike that the rumble effect in a game differs depending on whether I'm connected via USB or Bluetooth.
Lastly, It's quite distracting to receive feedback from the triggers when they are not being used in a game.

I hope that comes across as useful feedback rather than ungrateful criticism! Xpadneo is a fantastic project.

kakra, 2026-08-25:
> Lastly, It's quite distracting to receive feedback from the triggers when they are not being used in a game.

Unless you press the triggers, they should not rumble at all. The rumble strength of the triggers is the product of the following multiplication:

```
trigger_rumble = max(weak_motor, strong_motor) * trigger_pressure * (100% - attenuation)
```

So if you don't press the trigger, one factor is zero, so the trigger rumble strength becomes zero. This formula is run once per trigger (left and right).