atar-axis/xpadneo #375: Lose rumble after reconnect controller

deathxxx123, 2022-08-01:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Ruble is not working after reconnection.

## Steps to Reproduce

Open some game, with controller connected. Now disconnect and reconnect again.

## Expected Behavior

I expect to get ruble working again when reconnect.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
5.18.14-1-MANJARO #1 SMP PREEMPT_DYNAMIC Sun Jul 24 09:00:58 UTC 2022 x86_64 GNU/Linux
```

This require **xxm** to be installed witch is **gvim** dependency.
```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20  U..|..&..u......... 
00000140: 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0                    ....&..u......
3560615638 1558

```

## Controller and Bluetooth Information

## Additional Context

I tested in Elden Ring and Stardew Walley. So it's one Proton game and one native game. Results are the same.

deathxxx123, 2022-08-01:
Restarting the game fix rumbling.

kakra, 2022-08-01:
I've also seen this, it is similar to the behavior when connecting an additional controller: Some games will pick up the second controller while running, some ignore it. Both won't rumble any longer, or the wrong controller rumbles. I think this is an issue in Proton/wine and should be reported there. It may be intentional behavior to not rumble a controller that's sitting idle on a table but it may not work properly. If it is intentional behavior, I'd say the controller should rumble again as soon as a button press was detected so the wine-drivers and/or SDL know that a hand is holding it.

Please make sure you disabled Steam Input as it interferes with Xbox controller rumble.

From my personal experience, I can say that an intermittent loss and reconnect of the Bluetooth connection should not disable rumble, it worked for many games in the past. But I remember some games that did not continue to rumble after such an event. Thus, it's not an xpadneo issue. After updating the controller firmware, I didn't see any more intermittent disconnects, so this issue went mostly invisible. But it's probably still there and sits either in SDL (which is used by Proton and Steam Input) or in Proton/wine. Because you also see it with a native game, I suspect it's an SDL issue first, or the game isn't handling this event well.

deathxxx123, 2022-08-01:
Yes im afraid from this. Elden Ring has disabled steam input anyway.
I was not try some outside Steam games. Maybe some GOG titles.
And Stardew Walley is native title. Maybe steam client itself cause something.

deathxxx123, 2022-08-01:
I confirm that a native linux prt Stardew Walley from GOG does have exactly same issue with losing rumble after reconnect.
And just after restarting the game, rumble works.

deathxxx123, 2022-08-01:
I have some other controllers and will try with them also with wireless connections.

kakra, 2022-08-01:
> I confirm that a native linux port Stardew Walley from GOG does have exactly same issue with losing rumble after reconnect.
> And just after restarting the game, rumble works.

Can you retry with the Steam client completely closed (select quit from systray)? It will stop the Steam Input client from running in the background. Do you use other software like AntiMicro to emulate/map game input devices?

If the problem persists, it's probably in SDL or the game itself.

While the game is still running but the controller does not rumble anymore, you can try `fftest` (run `evtest` first to know the device path). Keep in mind that not all rumble effects are compatible with the controller but if `fftest` still makes the controller rumble, the fault is somewhere up the layers and the driver itself is fine.

I'd recommend to check `fftest` function before trying the game so you know what is expected to work. Then reproduce the issue and try again. After a reconnect, the device path will most likely change so you have to use the new path then. I suspect that the game won't re-detect the capabilities of the device properly after reconnect, so it won't send rumble commands any longer. Whether this issue is in the game itself or in SDL, I'm not sure. Since I've seen games that properly continued to rumble after a re-connect, I'd say the game is missing some code to detect and handle the event. Maybe the game continues to send rumble to the old device path while it receives input events from any device path connected. From xpadneo, we cannot prevent the device path change, it's how the kernel and udev work. Maybe udev could setup a persistent device path using symlinks but I'm not sure SDL games would properly pick that up: SDL looks for `eventXX` only.

deathxxx123, 2022-08-01:
I do not use any other programs than Steam.
fftest already cannot pick right device, bur game working with rumble in same time.
Wirh reconnect all is same exept no more rumble in game.

```
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Power Button
/dev/input/event2:	SIGMACHIP USB Keyboard
/dev/input/event3:	SIGMACHIP USB Keyboard Consumer Control
/dev/input/event4:	SIGMACHIP USB Keyboard System Control
/dev/input/event5:	Logitech MX Master 3
/dev/input/event6:	Xbox Wireless Controller
```
`sudo fftest
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event0 opened
Features:
  * Absolute axes: 
    [00 00 00 00 00 00 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: 
    Force feedback periodic effects: 
    [00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
  * Number of simultaneous effects: 0

Uploading effect #0 (Periodic sinusoidal) ... Error:: Function not implemented
Uploading effect #1 (Constant) ... Error: Function not implemented
Uploading effect #2 (Spring) ... Error: Function not implemented
Uploading effect #3 (Damper) ... Error: Function not implemented
Uploading effect #4 (Strong rumble, with heavy motor) ... Error: Function not implemented
Uploading effect #5 (Weak rumble, with light motor) ... Error: Function not implemented
Enter effect number, -1 to exit`

deathxxx123, 2022-08-01:
```
sudo fftest /dev/input/event6
Force feedback test program.
HOLD FIRMLY YOUR WHEEL OR JOYSTICK TO PREVENT DAMAGES

Device /dev/input/event6 opened
Features:
  * Absolute axes: X, Y, Z, RX, RY, RZ, Hat 0 X, Hat 0 Y, Misc ,
    [3F 00 03 00 00 01 00 00 ]
  * Relative axes: 
    [00 00 ]
  * Force feedback effects types: Periodic, Rumble, Gain, 
    Force feedback periodic effects: Square, Triangle, Sine, 
    [00 00 00 00 00 00 00 00 00 00 03 07 01 00 00 00 ]
  * Number of simultaneous effects: 16

Setting master gain to 75% ... OK
Uploading effect #0 (Periodic sinusoidal) ... OK (id 0)
Uploading effect #1 (Constant) ... Error: Invalid argument
Uploading effect #2 (Spring) ... Error: Invalid argument
Uploading effect #3 (Damper) ... Error: Invalid argument
Uploading effect #4 (Strong rumble, with heavy motor) ... OK (id 1)
Uploading effect #5 (Weak rumble, with light motor) ... OK (id 2)
Enter effect number, -1 to exit
5
Now Playing: Weak Rumble
Enter effect number, -1 to exit
```

deathxxx123, 2022-08-01:
Now rumble is working from terminal while game is opened and don't have working rumble with Steam opened.

deathxxx123, 2022-08-01:
I test Stardew Walley native game with steam quit/exit.
After reconnect rumble in game stopped working, but in terminal is working.

deathxxx123, 2022-08-01:
I continue testing.
Now i try with PS4 controller with no software/driver installed. Just with steam.
Rumble is working and after disconnect still working.

deathxxx123, 2022-08-01:
I try with xpadneo v.9.1, still same efect, but reconnectoin is faster.

kakra, 2022-08-01:
> Now i try with PS4 controller with no software/driver installed. Just with steam.
> Rumble is working and after disconnect/reconnect still working.

Then I suspect the problem within SDL... It has a very unique code path for Xbox controllers that differs from the other input drivers.

deathxxx123, 2022-08-01:
I test again:
Uninstall xpadneo driver and use default kernel driver.
Then start steam, and test with 2 games.

Elden Ring with Proton
and
Streets of Rage 4 (native)

It's no problem for rumble and disconnecting/reconnecting now.

kakra, 2022-08-01:
If you uninstall xpadneo, hidraw will be enabled. That's a different code path in den SDL drivers: In this mode, it won't use the Linux kernel drivers. There's nothing in xpadneo kernel driver that could cause such issues, running `fftest` directly on the low-level kernel driver interface showed it: It can still rumble even when the game does not. This is because we no longer get rumble packets from SDL after the re-connect. The problem is at SDL or higher levels.

Future versions of xpadneo are planned to be compatible with SDL hidraw expectations, so we can probably enable it by default then. It may avoid your re-connect issues because SDL is then using the hidraw code path.

deathxxx123, 2022-08-01:
Yes, this is only thing left for perfection.

kakra, 2022-10-19:
@orbea It's probably the same problem if user-space is using SDL.

@deathxxx123 You may be able working around it by disabling the connection notification rumble. It is currently a blocking operation during driver initialization and will be reworked for v0.10.

closed 2023-06-14 by kakra