atar-axis/xpadneo #376: Support gyro on 8BitDo Pro 2

Yasand123, 2022-08-06:
## Version of xpadneo
v0.9-121-g727a84f

## Controller Model
- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [X] Other: 8BitDo Pro 2

### Connection mode
- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature request
- [ ] I'd like to see support for a specific model
- [X] I'd like to see support for a specific hardware feature
- [X] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**
Gyro does not work under Linux. There's a workaround, which is to disable the kernel module `hid_nintendo` as it interferes with Switch 3rd party controllers such as 8BitDo Pro 2. However that workaround barely works. I tried CEMU under Wine, Steam, Ryujinx as well as Dolphin emulator and they do not recognize motion inputs regardless of which API I choose. Only exception being Yuzu and it has to be connected via USB (I'm not sure why). I'm not sure what Yuzu does differently.

Controller is set to S (Switch mode).

**Describe the solution you'd like**
For Gyro inputs to be recognized by the system and applications properly.

**Describe alternatives you've considered**
There are no alternatives that I'm aware of. I think I tried some suggested solution in the past but I can't remember what it was. I apologize for that. It didn't work anyway. 

**Additional context**
In case it's relevant. I'm on Arch Linux.

kakra, 2022-08-06:
xpadneo does not support controllers in Switch mode. And I'm guessing the gyro won't be exposed by the interface if the controller is in Xbox mode.

kakra, 2022-11-16:
The GuliKit controllers also have a gyro, and such a controller has been donated. I'll look into supporting these types of controllers in Switch mode but this won't happen during v0.10 (tho I may start working on it).

anzix, 2025-01-08:
You can use the gyroscope in the UDP protocol using [joycond-cemuhook](https://github.com/joaorb64/joycond-cemuhook/). Download and run the command `sudo joycond-cemuhook` and connect the gamepad (it works both via Bluetooth and by wire). Usually, the gyroscope in the yuzu emulator (and its forks) works great with my 8Bitdo Pro 2 in Switch mode. The main thing in the emulator Settings is in the Control - Advanced section, you must check the box "Enable UDP controllers" and everything will work.