atar-axis/xpadneo #377: Xbox Series X|S Controller won't connect

RoyvanEmpel, 2022-08-06:
## Version of xpadneo

v0.9.4

## Controller Model

- [x] Xbox Series X|S controller

## Connection mode

- [x] Bluetooth connection

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

```
filename:       /lib/modules/5.15.6-051506-generic/updates/dkms/hid-xpadneo.ko
version:        v0.9-121-g727a84f
description:    Linux kernel driver for Xbox ONE S+ gamepads (BT), incl. FF
author:         Kai Krakow <kai@kaishome.de>
author:         Florian Dollinger <dollinger.florian@gmx.de>
license:        GPL
srcversion:     ABAAFAFCF3EE0C50890D9B5
alias:          hid:b0005g*v0000045Ep00000B13
alias:          hid:b0005g*v0000045Ep00000B22
alias:          hid:b0005g*v0000045Ep00000B05
alias:          hid:b0005g*v0000045Ep00000B20
alias:          hid:b0005g*v0000045Ep000002E0
alias:          hid:b0005g*v0000045Ep000002FD
depends:        hid,ff-memless
retpoline:      Y
name:           hid_xpadneo
vermagic:       5.15.6-051506-generic SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         roy-pc Secure Boot Module Signature key
sig_key:        4D:7A:A5:F2:C2:D1:28:D9:E8:82:3C:11:54:5B:3D:8B:3A:B8:94:54
sig_hashalgo:   sha512
signature:      5B:6A:78:53:A2:11:93:6C:11:6B:14:71:8A:6B:11:B3:2A:B2:C2:03:
                D6:2F:AF:A4:87:0A:90:4D:C8:F7:42:18:4B:05:74:8C:1C:16:FD:9C:
                9B:24:B5:7B:E3:60:CD:B9:55:EE:B9:1B:EF:F5:12:DE:20:63:B2:24:
                C6:F0:DE:40:5A:C2:FC:DF:F3:DF:8E:A7:A9:7B:E9:52:D0:FB:0E:D2:
                85:AF:36:A8:63:8D:FC:5E:A6:BC:74:E1:B0:37:45:72:9F:61:BE:74:
                3A:EE:B1:2C:A2:0B:0E:1E:FD:A2:D0:9D:4C:AD:6D:F3:D5:31:91:7D:
                73:7E:D2:71:E2:8D:20:30:E8:4B:DF:4E:7F:58:2C:B4:3A:A8:3F:94:
                EC:70:14:4D:30:A9:CC:FA:C5:FD:E9:A5:69:D6:33:C4:EE:71:ED:4A:
                76:79:F8:81:50:E4:05:7F:68:61:E1:F6:57:56:B6:D9:6D:D4:84:E7:
                B7:A6:F1:87:05:D4:70:40:AA:71:0A:5D:FE:C2:EB:D2:95:6E:78:93:
                57:0E:A9:FC:D9:EE:61:DD:14:2E:B6:9D:FF:B4:EC:3B:15:1A:FA:B9:
                66:50:7B:E7:1D:68:28:1F:56:CB:F4:8E:EE:4F:CA:38:AE:12:64:16:
                B2:1E:F7:9D:30:49:D4:A8:96:ED:17:B3:D7:EA:B5:0D
parm:           trigger_rumble_mode:(u8) Trigger rumble mode. 0: pressure, 2: disable. (byte)
parm:           rumble_attenuation:(u8) Attenuate the rumble strength: all[,triggers] 0 (none, full rumble) to 100 (max, no rumble). (array of byte)
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
parm:           gamepad_compliance:(bool) Adhere to Linux Gamepad Specification by using signed axis values. 1: enable, 0: disable. (bool)
parm:           disable_deadzones:(bool) Disable dead zone handling for raw processing by Wine/Proton, confuses joydev. 0: disable, 1: enable. (bool)
parm:           quirks:(string) Override or change device quirks, specify as: "MAC1{:,+,-}quirks1[,...16]", MAC format = 11:22:33:44:55:66, no pulse parameters = 1, no trigger rumble = 2, no motor masking = 4, use Linux button mappings = 16, use Nintendo mappings = 32, use Share button mappings = 64 (array of charp)
```

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've tried disabling or running without above mentioned software
- [x] It does not work at all

## Describe the Bug

The Xbox controller keeps blinking and doesn't connect. Sometimes the PC sees it as connected but the light on the controller keeps blinking and it is unresponsive.

## Steps to Reproduce

I installed xpadneo using the `install.sh` and after that tried to connect the controller to the pc.

## Expected Behavior

The controller to stop blinking and connect to the pc.

## System Information

```console
# uname -a
Linux roy-pc 5.15.6-051506-generic #202112010437 SMP Wed Dec 1 09:47:58 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/9275910/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/9275924/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/9275925/xpadneo-lsusb.txt)

kakra, 2022-08-07:
Your kernel is really old, please check with your distribution if there's a newer version which fixes some issues with Bluetooth in general. The current LTS version is 5.15.59, you're running 5.15.6 which is a very early LTS version.

Also, your Xbox controller is using a firmware with known problems:
```
[24839.181354] xpadneo 0005:045E:0B13.000E: BLE firmware version 5.09, please upgrade for better stability
```

This is known to result in connection loops for some combinations of chipsets/kernel/bluez versions. You can upgrade the firmware using a Windows machine and running the Xbox Accessories app from the app store.

RoyvanEmpel, 2022-08-07:
Ooh i will try upgrading the kernel to a newer version. Is there a tutorial to update the firmware of the xbox controller on linux? Or do i need an xbox..x

RoyvanEmpel, 2022-08-07:
I do not have any windows devices in my house.

kakra, 2022-08-07:
An Xbox will also work for upgrading. Also, you can download an eval copy of Windows and upgrade the firmware from a VM but it can be tricky: when the controller reboots into firmware update mode, its USB id will change, so you might need to fiddle around with auto USB passthrough, I never got it to work properly. Some people reported success using Virtualbox. If you have the original Xbox dongle, you can upgrade through that using a VM. I'm doing it that way.

RoyvanEmpel, 2022-08-07:
Well, the windows in the VM sees the controller but the Xbox Accessories App only shows a big "help" button and doesn't detect it. So I guess I am going to have to get hold of a windows machine somewhere. 

Thanks for the help for now :)

closed 2023-06-14 by kakra