atar-axis/xpadneo #378: Controller Becomes unresponsive

tmayoff, 2022-08-07:
## Version of xpadneo
0.9.3 (pretty sure same was happening on 0.9.4

## Controller Model

- [ ] Xbox One S controller
- [X] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [X] I don't use Lutris, Bottles, wine or did not try
- [X] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [X] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool
```
[LOG] SDL2 Gamepad Tool v1.1.2 by General Arcade (compiled with SDL version 2.0.4, DLL version 2.0.22)
[LOG] Website: http://generealarcade.com/gamepadtool/

[LOG] Searching gamepads...
[LOG] Found 1 gamepad(s):
[LOG] "Xbox One S Controller", 050000005e040000fd02000030110000 (mapping available)
[LOG] Environment variable "SDL_GAMECONTROLLERCONFIG" is not defined
[LOG] Checking if new mappings available from github: https://github.com/gabomdq/SDL_GameControllerDB
```

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [X] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [X] It used to work in a previous version
- [X] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
When left alone for a couple of seconds (2-5) then trying to use it not receive inputs for a couple of seconds (10+). (It also seems play the last input whenever it 'wakes up' then becomes responsive)

## Steps to Reproduce
Don't touch any inputs, wait for a couple of seconds, then try to press buttons or use the joystick

## Expected Behavior
Input should always work

## System Information

```console
# uname -a
Linux malcolm 5.18.16-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Aug 3 15:44:49 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 a1 01 0a 85 00  ...%.u..............
000000c8: 15 00 26 ff 00 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95  ..&....u........&...
000000dc: 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 0a 81 00 15 00  .u.....%...u........
000000f0: 26 ff 00 95 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 c0  &....u.....%...u....
00000104: 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02  ...!........%.u.....
00000118: 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04  ..%.u......p..%du...
0000012c: 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02  ...Pf..U...&..u.....
00000140: 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15  ....&..u.....e.U..|.
00000154: 00 26 ff 00 75 08 95 01 91 02 c0 05 01 09 06 a1 01 85 05 05  .&..u...............
00000168: 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81  ...)...%.u.......u..
0000017c: 03 95 06 75 08 15 00 25 65 05 07 19 00 29 65 81 00 c0 05 0c  ...u...%e....)e.....
00000190: 09 01 85 0c a1 01 0a 9e 00 15 00 26 ff 00 95 01 75 08 81 02  ...........&....u...
000001a4: 0a a1 00 15 00 26 ff 00 95 01 75 08 81 02 0a a2 00 15 00 26  .....&....u........&
000001b8: ff 00 95 01 75 08 81 02 0a a3 00 15 00 26 ff 00 95 01 75 08  ....u........&....u.
000001cc: 81 02 c0 c0                                                  ....
1410450731 2192
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/9277642/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/9277644/xpadneo-dmesg.txt)

## Additional Context
Games I tried were Stray via steam and Proton and this [project](https://github.com/open-goal/jak-project)

kakra, 2022-08-07:
Please upgrade the firmware of the controller (but you seem to run the latest version, just in case others come across this report) and try the latency fix from the troubleshooting guide:
https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md#high-latency-or-lost-button-events-with-bluetooth-le

tmayoff, 2022-08-08:
So 'updated' firmware no updates available, and applied the changes in the troubleshooting guide, no luck. Not sure what else I can send to help debug this

kakra, 2022-08-08:
Okay, so there may still be problems with the Intel Bluetooth chipset. Maybe the wrong latency parameters are still cached in Bluetooth and you should completely wipe the pairing from Bluetooth and pair the controller again.

Otherwise, could you, by chance, try a dedicated Bluetooth dongle, CSR-chipsets preferred?

tmayoff, 2022-08-08:
I'd have to look around to see if I have one, I'll let you know

tmayoff, 2022-08-25:
Sorry I got distracted and forgot to respond unfortunately I don't have a dedicated bluetooth dongle

closed 2023-06-14 by kakra