atar-axis/xpadneo #383: Kernel panic on Linux 4 Tegra r35.1 (Ubuntu 20.04.4 LTS, aarch64)

makinbacon21, 2022-08-22:
## Version of xpadneo
Tried on v0.9.4 RE-RELEASE and current master (should be the same though)

## Controller Model
- [ ] Xbox One S controller
- [x] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

Can't get to any of this--it insta-fails right after displaying as connected in `bluetoothctl`

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [x] I can code (not super well-versed on programming kernel drivers but I work with them) and I want to help

## Describe the Bug
On running `connect <mac>` in `bluetoothctl`, the kernel panics.

## Steps to Reproduce
Run stock Jetson Linux 35.1 on an AGX Orin (I expect on other Tegra devices this also happens), install the driver via dkms, and try to pair the controller. The output will appear successful, indicating it is connected, but then the whole system will freeze up. Kernel logs indicate a panic has occurred.

## Expected Behavior
Obviously I expected it to work lol

## Screenshots / GIFs / Videos
No need for screenshots

## System Information

```console
# uname -a
Linux tmakin-orin 5.10.104-tegra #1 SMP PREEMPT Wed Aug 10 20:17:07 PDT 2022 aarch64 aarch64 aarch64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth Information
None of this is really relevant. My BT adapter is a PCIe combo bt/wifi adapter (rtl8822ce) that comes built in on the devkit.

## Additional Context
dmesg log gathered via UART:
```console
[  624.479058] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
[  624.488235] Mem abort info:
[  624.491193]   ESR = 0x96000005
[  624.494411]   EC = 0x25: DABT (current EL), IL = 32 bits
[  624.499948]   SET = 0, FnV = 0
[  624.503152]   EA = 0, S1PTW = 0
[  624.506441] Data abort info:
[  624.509449]   ISV = 0, ISS = 0x00000005
[  624.513445]   CM = 0, WnR = 0
[  624.516535] user pgtable: 4k pages, 48-bit VAs, pgdp=00000001eca94000
[  624.523212] [0000000000000000] pgd=00000001ecb05003, p4d=00000001ecb05003, pud=0000000000000000
[  624.532213] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[  624.537956] Modules linked in: nvidia_modeset(O) fuse xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter lzo_rle lzo_compress zram overlay ramoops reed_solomon bnep binfmt_misc snd_soc_tegra210_ope snd_soc_tegra186_dspk snd_soc_tegra186_asrc snd_soc_tegra210_iqc snd_soc_tegra186_arad snd_soc_tegra210_mvc snd_soc_tegra210_afc snd_soc_tegra210_dmic snd_soc_tegra210_adx snd_soc_tegra210_amx input_leds snd_soc_tegra210_i2s snd_soc_tegra210_admaif snd_soc_tegra210_mixer aes_ce_blk snd_soc_tegra210_sfc snd_soc_tegra_pcm crypto_simd cryptd aes_ce_cipher rtk_btusb ghash_ce btusb sha2_ce btrtl rtl8822ce sha256_arm64 btbcm snd_soc_tegra210_adsp ucsi_ccg sha1_ce btintel snd_soc_tegra_machine_driver typec_ucsi snd_soc_tegra_utils snd_soc_spdif_tx typec ofpart snd_hda_codec_hdmi snd_soc_simple_card_utils nct1008 cmdlinepart cfg80211 snd_hda_tegra snd_soc_tegra210_ahub snd_hda_codec
[  624.538057]  qspi_mtd nvadsp snd_hda_core mtd userspace_alert tegra_bpmp_thermal tegra210_adma spi_tegra114 nvidia(O) ina3221 pwm_fan loop nvgpu nvmap ip_tables x_tables
[  624.645734] CPU: 0 PID: 123 Comm: kworker/u25:0 Tainted: G           O      5.10.104-tegra #1
[  624.654508] Hardware name:  /, BIOS 1.0-d7fb19b 08/10/2022
[  624.660165] Workqueue: hci0 hci_rx_work
[  624.664112] pstate: 20c00009 (nzCv daif +PAN +UAO -TCO BTYPE=--)
[  624.670303] pc : vli_mult+0x74/0x100
[  624.673985] lr : vli_mult+0x2c/0x100
[  624.677664] sp : ffff8000117bb820
[  624.681074] x29: ffff8000117bb820 x28: 0000000000002010
[  624.686540] x27: ffff78010f326010 x26: 0000000000000040
[  624.692007] x25: ffffd04f80767380 x24: ffff7801ce1a7100
[  624.697473] x23: 0000000000000000 x22: ffff8000117bb898
[  624.702940] x21: 0000000000000000 x20: ffff7801ce1a7100
[  624.708405] x19: 0000000000000004 x18: 0000000000000018
[  624.713868] x17: ffff8000117bb798 x16: 0000000000000003
[  624.719334] x15: ffff8000117bb750 x14: ffff8000117bb7b0
[  624.724799] x13: ffff8000117bb790 x12: ffff8000117bb738
[  624.730264] x11: 0000000000000000 x10: 0000000000000000
[  624.735729] x9 : b55171de7242460c x8 : 0000000000000000
[  624.741195] x7 : 3260eb02a6086b19 x6 : 0000000000000000
[  624.746661] x5 : 0000000000000000 x4 : 0000000000000000
[  624.752126] x3 : 00000000fffffffd x2 : 0000000000000001
[  624.757591] x1 : 0000000000000000 x0 : 0000000000000007
[  624.763057] Call trace:
[  624.765570]  vli_mult+0x74/0x100
[  624.768897]  vli_mod_mult_fast+0x58/0xa0
[  624.772931]  ecc_is_pubkey_valid_partial+0x114/0x1e0
[  624.778036]  ecc_is_pubkey_valid_full+0x30/0xb0
[  624.782700]  ecc_make_pub_key+0xb4/0x140
[  624.786733]  ecdh_compute_value+0x164/0x180
[  624.791036]  generate_ecdh_public_key+0xe0/0x1a0
[  624.795787]  generate_ecdh_keys+0x3c/0x50
[  624.799913]  sc_send_public_key+0xf0/0x1a0
[  624.804125]  smp_recv_cb+0x13c4/0x16e0
[  624.807985]  l2cap_recv_frame+0xb40/0x16a0
[  624.812197]  l2cap_recv_acldata+0x20c/0x210
[  624.816496]  hci_rx_work+0x230/0x290
[  624.820179]  process_one_work+0x1c4/0x4a0
[  624.824304]  worker_thread+0x54/0x430
[  624.828071]  kthread+0x148/0x170
[  624.831400]  ret_from_fork+0x10/0x24
[  624.835084] Code: 7a441260 54000409 d280000a 4b040165 (f8645aa9)
[  624.841371] ---[ end trace fa2a9de23c9d6c38 ]---
[  624.851137] Kernel panic - not syncing: Oops: Fatal exception
[  624.857048] SMP: stopping secondary CPUs
[  624.861313] Kernel Offset: 0x504f6e940000 from 0xffff800010000000
[  624.867583] PHYS_OFFSET: 0xffff880000000000
[  624.871891] CPU features: 0x0040006,4a80aa38
[  624.876287] Memory Limit: none
[  624.884440] ---[ end Kernel panic - not syncing: Oops: Fatal exception ]---
```

kakra, 2022-08-23:
The xpadneo module is not involved here, actually, xpadneo doesn't interact with Bluetooth at all, xpadneo is purely a HID driver, the HID API is provided by the Bluetooth stack. It looks like the bug you are seeing is in the HCI driver, it should be reported to the bluez team.

closed 2022-08-23 by kakra

makinbacon21, 2022-08-23:
Thanks for the help! Good to be able to rule out xpadneo as the issue. I think it's not likely bluez' fault--more likely nvidia modified the hci driver (again) and screwed it up (again). This isn't even the only driver issue for this card in the latest L4T release...no wonder they keep pushing the final release.

russelltg, 2023-05-12:
@makinbacon21 Did you end up opening an issue with Nvidia about this? I'm hitting this as well.