atar-axis/xpadneo #384: Left and Right are reversed for right stick in Dolphin for True Crime Streets of LA

ipkpjersi, 2022-08-26:
## Version of xpadneo
0.9.4

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [X] I've tried disabling or running without above mentioned software
- [X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
When running True Crime Streets of LA in Dolphin 5.0 snap on Ubuntu 22.04, moving the camera left and right is reversed for the C-Stick (I have it bound to the right stick). 

True Crime New York City works exactly as expected in Dolphin 5.0 snap, so I'm not sure if it's a game-specific issue, or if it's a Dolphin issue, or an xpadneo issue, or something else entirely. I'm not sure if I should file an issue here, or with Dolphin, or if this is somehow possibly working as expected (I don't think it is, I'll explain more below later) or something.

## Steps to Reproduce
1. Install xpadneo 0.9.4
2. Connect Xbox Series X controller
3. Install dolphin 5.0 snap with `sudo snap install dolphin-emulator`
4. Open True Crime Streets of LA
5. Move the camera around with the C-Stick (aka the right stick), turn left and it goes right, turn right and it goes left.

## Expected Behavior
I expect turning left to go left and right to go right.

## Screenshots / GIFs / Videos
Funnily enough, here's a video of someone else having the same problem: https://youtu.be/ikypRCyyMDI?t=580
Granted, the chances of them also using xpadneo back in 2018 are pretty unlikely, but maybe xpadneo could possibly workaround this somehow? I swear I remember playing this game back on Gamecube and it never had this problem unless I'm misremembering somehow - also Googling this problem doesn't seem to return any results either, if it were more common and not simply some sort of bug you think there would be more results of people just dumbfounded by this like I am right now.

## System Information

```console
# uname -a
Linux fast-cottage 5.15.0-46-generic #49-Ubuntu SMP Thu Aug 4 18:03:25 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0

```

## Controller and Bluetooth Information
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/9434155/xpadneo-btmon.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/9434157/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/9434159/xpadneo-dmesg.txt)

## Additional Context
I'm really not sure what to do about this issue but I'm sure there's some solution or workaround somehow? I would greatly appreciate any advice or suggestions or anything.

ipkpjersi, 2022-08-26:
Okay, I just tested it on PCSX2 and it has the same behaviour - right and left on the right control stick are swapped. Maybe it is just the game? It doesn't make sense to me though, because I don't remember the game behaving this way, and I'm literally not seeing anything about it on Google, or in any reviews, or anything. There's no way something this weird could be "normal", right?

ipkpjersi, 2022-08-26:
I found a decent workaround, I can just bind the right and left movements in reverse in Dolphin settings, so I can go ahead and close this issue since it's probably not an xpadneo issue. It's a shame controller settings can't be customized per game because then I wouldn't need to change them depending on which game I play. Oh well, this workaround works well enough anyway.

closed 2022-08-26 by ipkpjersi

kakra, 2022-08-26:
This sounds very strange, I believe this is a specific behavior of the game. To me, it sounds like the developers implemented in a way that you move the camera position behind your character instead of the direction of turning, it may be intended.