atar-axis/xpadneo #388: Does it detect xbox controller motion?

gustavosr98, 2022-09-24:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature reqeust

- [ ] I'd like to see support for a specific model
- [ ] I'd like to see support for a specific hardware feature
- [ ] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when ...

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kakra, 2022-09-26:
I'm not sure what this is supposed to ask for... Are you asking for motion controls? None of the current Xbox controller models support that feature. It seems that MS has patented some ideas for motion and VR accessories but Xbox controllers itself (neither the Xbox Adaptive controller) have support for this, they are lacking the internal sensors for such a feature.

gustavosr98, 2022-09-27:
I am sorry. I just bought an Xbox Controller and assumed It had a gyroscope like PS5 Controllers. Thank you

closed 2022-09-27 by gustavosr98