atar-axis/xpadneo #389: Series S|X controller updated to 5.13.3146.0 firmware disconnects after a minute or two, even when wired

kode54, 2022-10-06:
## Version of xpadneo
xpadneo-dkms-git 0.9.r121.g727a84f-1

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [x] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [x] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [x] I can code and I want to help

## Describe the Bug
The controller worked just fine on the previous 5.13.3143.0 firmware, but now that I'm on the recently released 5.13.3146.0 firmware, it randomly disconnects after a few minutes. And since you mention that USB support is not there, I cannot get it to work with that, either.

## Steps to Reproduce
1. Update the controller firmware with Xbox Accessories app from Windows 11.
2. Switch back to Linux with xpadneo.
3. Controller randomly disconnects while gaming. This is indicated by the controller doing the little rumble dance it does every time I power it on, and my game ceasing to respond to the controller.

## Expected Behavior
I expect the controller to stay connected and working.

## Screenshots / GIFs / Videos
I have no screen shots to demonstrate it. I do have some dmesg log output, though.

```
Oct 04 22:23:38 mrgency bluetoothd[6336]: src/device.c:device_set_wake_support() Unable to set wake_support without RPA resolution
```

## System Information

```console
# uname -a
Linux mrgency 6.0.0-270-tkg-cfs #1 TKG SMP PREEMPT_DYNAMIC Wed, 05 Oct 2022 03:32:37 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/9721115/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/9721116/xpadneo-dmesg.txt)

```
Bus 001 Device 025: ID 04ca:3006 Lite-On Technology Corp.
```

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/9721123/xpadneo-lsusb.txt)

## Additional Context

As I said, this issue only started after I briefly used Windows 11 and made the mistake of updating my controller's firmware. Windows does not allow me to revert it, either.

kakra, 2022-11-16:
Does this happen when rumble is involved?

bezirg, 2022-12-02:
I have also disconnects happening randomly after some minutes. The controller is disconnected and immediately reconnected by doing the "welcome rumble" dance.

I am not sure if it is the gamepad's fault or the bluetooth hci0 controller's fault. I am saying that because I had previously other bluetooth problems with my hci0 controller (wifi ax200).

kakra, 2022-12-02:
The AX200 is known to have problems with this controller.

Overflwn, 2023-08-03:
Sorry for necro'ing this issue, didn't know if I really should open a new (similar) one

I also have the same problem with my Series X controller and a TP-Link UB500 dongle. The controller keeps getting disconnected after some time between 1-15 minutes and immediately reconnects with the init rumble. I'm only using it while playing Steam games and at least these games recognize the re-connection and I can continue playing after waiting for the reconnect.

xpadneo-dkms-git (0.9.r144.g9b3b696-1)
Archlinux 7.4.7-arch1-3

Don't know how to check what exact firmware my controller has installed, however, last time I updated was approx. 2-3 weeks ago, so I guess it should be the newest firmware available..

Dmesg log (here I connected the controller after booting, which got re-connected at approx 130.8 and then it re-connected again at about 280.0):

```
[    8.008364] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    8.008366] Bluetooth: BNEP filters: protocol multicast
[    8.008368] Bluetooth: BNEP socket layer initialized
[    8.008834] Bluetooth: MGMT ver 1.22
[    8.014114] NET: Registered PF_ALG protocol family
[    8.076898] RTL8226B_RTL8221B 2.5Gbps PHY r8169-0-2a00:00: attached PHY driver (mii_bus:phy_addr=r8169-0-2a00:00, irq=MAC)
[    8.163482] Bluetooth: hci0: Bad flag given (0x1) vs supported (0x0)
[    8.273978] r8169 0000:2a:00.0 enp42s0: Link is Down
[    8.300227] Generic FE-GE Realtek PHY r8169-0-400:00: attached PHY driver (mii_bus:phy_addr=r8169-0-400:00, irq=MAC)
[    8.490358] r8169 0000:04:00.0 enp4s0: Link is Down
[    8.544757] userif-4: sent link down event.
[    8.544760] userif-4: sent link up event.
[    9.486995] userif-4: sent link down event.
[    9.486999] userif-4: sent link up event.
[   10.823730] IPv6: ADDRCONF(NETDEV_CHANGE): enp42s0: link becomes ready
[   10.823818] r8169 0000:2a:00.0 enp42s0: Link is Up - 1Gbps/Full - flow control rx/tx
[   11.023842] userif-4: sent link down event.
[   11.023846] userif-4: sent link up event.
[   11.254477] memfd_create() without MFD_EXEC nor MFD_NOEXEC_SEAL, pid=965 'Xorg'
[   12.568736] rfkill: input handler disabled
[   13.417079] Bluetooth: RFCOMM TTY layer initialized
[   13.417086] Bluetooth: RFCOMM socket layer initialized
[   13.417087] Bluetooth: RFCOMM ver 1.11
[   21.002697] logitech-hidpp-device 0003:046D:4074.0009: HID++ 4.2 device connected.
[   24.707696] systemd-journald[338]: File /var/log/journal/b04f138d26264ec68ce4ea2d7b7077ef/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[   25.191957] rfkill: input handler enabled
[   26.765962] rfkill: input handler disabled
[   32.910751] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.000A/input/input30
[   32.910863] hid-generic 0005:045E:0B13.000A: input,hidraw9: BLUETOOTH HID v5.17 Gamepad [Xbox Wireless Controller] on e8:48:b8:c8:20:00
[   32.919426] loaded hid-xpadneo 0.9.r144.g9b3b696
[   33.000704] xpadneo 0005:045E:0B13.000A: BLE firmware version 5.17
[   33.000709] xpadneo 0005:045E:0B13.000A: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[   33.000710] xpadneo 0005:045E:0B13.000A: working around wrong SDL2 mappings (changed version from 0x00000517 to 0x00001130)
[   33.000712] xpadneo 0005:045E:0B13.000A: report descriptor size: 283 bytes
[   33.000713] xpadneo 0005:045E:0B13.000A: fixing up Rx axis
[   33.000714] xpadneo 0005:045E:0B13.000A: fixing up Ry axis
[   33.000715] xpadneo 0005:045E:0B13.000A: fixing up Z axis
[   33.000715] xpadneo 0005:045E:0B13.000A: fixing up Rz axis
[   33.000716] xpadneo 0005:045E:0B13.000A: fixing up button mapping
[   33.000786] xpadneo 0005:045E:0B13.000A: gamepad detected
[   33.000787] xpadneo 0005:045E:0B13.000A: enabling compliance with Linux Gamepad Specification
[   33.000812] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.000A/input/input31
[   33.000889] xpadneo 0005:045E:0B13.000A: input,hidraw9: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on e8:48:b8:c8:20:00
[   33.000908] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.000A/input/input32
[   33.000928] xpadneo 0005:045E:0B13.000A: consumer control added
[   33.000945] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.000A/input/input33
[   33.000965] xpadneo 0005:045E:0B13.000A: keyboard added
[   33.000967] xpadneo 0005:045E:0B13.000A: controller quirks: 0x00000050
[   33.000968] xpadneo xpadneo_welcome_rumble start
[   33.991677] xpadneo xpadneo_welcome_rumble took 990ms
[   33.991680] xpadneo 0005:045E:0B13.000A: Xbox Wireless Controller [44:16:22:d5:6f:ad] connected
[   34.041149] xpadneo 0005:045E:0B13.000A: reverting to original version (changed version from 0x00001130 to 0x00000517)
[   34.041154] xpadneo 0005:045E:0B13.000A: reverting to original product (changed PID from 0x028E to 0x0B13)
[   34.160599] xpadneo 0005:045E:0B13.000A: BLE firmware version 5.17
[   34.160603] xpadneo 0005:045E:0B13.000A: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[   34.160605] xpadneo 0005:045E:0B13.000A: working around wrong SDL2 mappings (changed version from 0x00000517 to 0x00001130)
[   34.160607] xpadneo 0005:045E:0B13.000A: report descriptor size: 283 bytes
[   34.160608] xpadneo 0005:045E:0B13.000A: fixing up Rx axis
[... 45 lines cut ...]
[... cut ...]
[  291.121362] xpadneo 0005:045E:0B13.000C: controller quirks: 0x00000050
[  291.121363] xpadneo xpadneo_welcome_rumble start
[  292.112083] xpadneo xpadneo_welcome_rumble took 990ms
[  292.112088] xpadneo 0005:045E:0B13.000C: Xbox Wireless Controller [44:16:22:d5:6f:ad] connected
[  345.063670] usb 1-2.3: reset full-speed USB device number 6 using xhci_hcd
[  346.604234] usb 1-2.3: reset full-speed USB device number 6 using xhci_hcd
[  348.150879] usb 1-2.3: reset full-speed USB device number 6 using xhci_hcd

```

Overflwn, 2023-08-10:
@Arian8j2 Regarding USB, try installing xone aswell. Now, before plugging in make sure that the controller is shut off or disconnected from bluetooth. Then connect the USB cable. That way xone should take over and not xpadneo (you will notice this by checking if there's the "welcome rumble", it should not trigger if you do it this way and that's when xone actually takes control)

This way your controller shouldn't have a problem anymore, at least it worked for me. However, I bought myself a xbox controller dongle on Amazon, which works perfectly fine with xone.

It's a shame though, I wouldn't need the dongle if the bluetooth connection just worked stabily..

kakra, 2023-08-11:
> @Overflwn I have exactly same problem with exactly same controller and bluetooth dongle, Controller disconnect after some times, and the time it takes to disconnect is pretty much random, One time i could play for 3 hours without any disconnects but some times it will disconnect every 15 minutes or so. I don't think it's a problem from bluetooth dongle because problem still exists if i hook it up via usb. I even hooked up controller with a Windows machine and updated it's firmware to latest one but the problem still exists on Linux. xpadneo version: `0.9.r144.g9b3b696` archlinux kernel version: `6.4.7.arch1-2`

Well, that's exactly the problem with incompatible Bluetooth hardware (tho, whether it's the controller or the dongle which is incompatible can be left as an unsolved discussion). With some Bluetooth dongles, the problem even persists in Windows.

If you are seeing the exact same problem with USB, things become strange:

- chances are your controller didn't detect the cable and the cable is broken.
- the controller always prefers cable connections over wireless (wireless will shut down if cable is connected).
- your games trigger the firmware rumble crash, try turning rumble off and see if it changes things.
- connections over USB cable don't use the xpadneo driver, xpadneo isn't involved at all in this case.

You could try:

- using the controller via Bluetooth in Windows may show the same problems, then it's the Bluetooth hardware.
- it may be a combination of both: Bluetooth hardware and rumble crash triggered by games.

Rumble crash can be avoided by inhibiting raw HID access to the controller device file. The udev file shipped with xpadneo does this but if you're running OpenRGB (or other software which operates on devices in raw mode), it may bypass this protection even for hardware it doesn't support. Try uninstalling such software temporarily. Take note: This only applies for the controller in Bluetooth mode, not in USB mode (USB is not HID).

Rumble crash seems to be triggered by games:

- in USB mode if rumble is sent faster than 10ms intervals (100 hz), unlikely because the kernel uses 50 hz but due to design of ff_memless, lower than 10ms may still be seen
- in GIP mode if rumble is sent faster than 20ms intervals (50 hz), xone prevents this by maintaining an interval of at least 20ms
- in HID mode (Bluetooth) if rumble is sent faster than 50ms intervals (20 hz), xpadneo prevents this by maintaining an interval of at least 50ms but user-space can by-pass this protection by writing to the HID device directly

For the last mode (HID), SDL has received a patch to stay at intervals above 50ms (it used 10-20ms previously). You may need to update your SDL version. Some native games ship with bundled older versions of SDL which don't have this fix. Disabling raw HID access is a reliable work-around. Disabling rumble is a reliable work-around, too. Also, try disabling Steam Input.

Later firmware versions seem to prefer longer intervals: We found that xone needed to go from 10 to 20ms, and xpadneo needed to go from 20 to 50ms after some firmware update. USB may be unaffected but it's more likely that it needs 20ms intervals, too.

Arian8j2, 2023-08-11:
@kakra I tried turning off rumble by setting `trigger_rumble_mode` to 2 and `rumble_attenuation` to 100 and tested out and rumble was successfully turned off then played a game for an hour and the issue was still there so i don't think the problem is due to rumble crash.

> connections over USB cable don't use the xpadneo driver, xpadneo isn't involved at all in this case.

I tried with usb and powered off bluetooth completely with `bluetoothctl`, then played the same game for an hour and **there was no issues**, Maybe last time it was still connected to bluetooth for some reason, idk.  
  
It seems the **UB500** is the problem, If i understand correctly even #432 has the same problem.

Overflwn, 2023-08-11:
> Maybe last time it was still connected to bluetooth for some reason, idk.

Yup, that‘s why I wrote that you first have to shut the controller (or the bt connection) off before plugging in.
If you are connected via BT and then connect the cable, it seems like the controller keeps using the BT connection.

kakra, 2023-08-11:
> If you are connected via BT and then connect the cable, it seems like the controller keeps using the BT connection.

It should not do that. I have 6 or 7 different models and vendors of Xbox compatible controllers, all support Bluetooth, three are genuine MS controllers. Each off them turns Bluetooth off and disconnects automatically from xpadneo as soon as USB is plugged in. So I rather suspect a flaky cable which doesn't properly establish a USB data connection and maybe only provides power if anything at all. Or you're connecting through a USB hub which doesn't always properly establish connection due to bandwidth or power constraints. Many non-powered USB hubs only support low-power USB devices as per spec, only a few ignore the spec and then it depends on your mainboard chipset (AMD usually allows running out of spec while Intel shuts down USB early).

A flaky cable may also explain why the controller intermittently re-connects. Also, using a passive USB hub may explain why the controller disconnects during rumble because it may exceed the spec power limit when rumble motors are running.

Try using a USB connector directly on your motherboard, or use an actively powered USB hub. If using a black USB connector (USB 2.1) try going to a blue USB 3 connector instead (provides more power). Or otherwise: If a lot of devices are already connected to either blue or black connectors, try isolating the controller on the other color to see if there are bandwidth/power limits you can circumvent that way. Also, if you're using a USB hub with more than 4 connectors, try switching to one with only 4 connectors (hubs with more than 4 connectors most likely have a cascaded hub inside with some connectors attached directly to the bus, some connectors handled by the internal hub chip).

Regarding the Bluetooth dongle: I originally suggested using BT400. Unfortunately, there are a lot of China clones out there which only mimic the CSR chipset but don't behave exactly the same. The kernel works around a lot of those issues but some devices may have problems still. I'm not sure which chipset the BT500 uses. Running `lsusb` may reveal it.

Arian8j2, 2023-08-11:
@kakra @Overflwn Thanks for the helps, I will try to get my hands on **UB400**, as soon as i got it i will test it and share my result.

Overflwn, 2023-08-11:
@kakra I will try again later today. I have used my USB3 ports on the pc case, I‘ll try it with MB ports aswell.

I haven‘t actually checked if GNOME shows the BT connection still being active, I just noticed that when I connect the cable while the BT connection is active, the random reconnect (with init rumble) kept happening, unless I first disconnect BT or shut the controller of and then connect the cable.

I think my USB cable should be fine though, as the controller works fine with it and I can also use it to copy files from phones and stuff like that.

kakra, 2023-08-12:
> I think my USB cable should be fine though, as the controller works fine with it and I can also use it to copy files from phones and stuff like that.

Yeah sounds like the cable might be okay, or the connector for that matter. That behavior is still strange and should not happen: The controller is supposed to drop wireless connections as soon as the wire is plugged in and established a USB data connection.

Arian8j2, 2023-08-13:
Used  **UB400** for two days, Played games for about 6 hours in this two days and had no disconnect, So now i'm pretty sure the problem is something with **UB500**, I also sens a little more latency (just a little) with **UB400** than **UB500** but i can't say for sure.

It is better to have a file like `tested-bluetooth-devices.md` or something like that, that people write their experience with different bluetooth devices and send pr, So people know which bluetooth dongle is better to buy or which devices has problems.

Again, Thank you @kakra  @Overflwn for all the helps.

kakra, 2023-08-14:
@Arian8j2 Try https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md#high-latency-or-lost-button-events-with-bluetooth-le

Arian8j2, 2023-08-14:
@kakra 
> Try https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md#high-latency-or-lost-button-events-with-bluetooth-le

I already had these configured.

> Also, see here:
https://github.com/atar-axis/xpadneo/blob/master/docs/BT_DONGLES.md

Oh cool i missed that :), Please add these too:
### Cambridge Silicon Radio
* [TP-Link USB Bluetooth Adapter Bluetooth 4.0 (UB400)](https://www.amazon.com/gp/product/B07V1SZCY6/)
  * Chipset: CSR ???
  * `ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`
  * Performance:
    * Connection flawless
  * Reported by @Arian8j2
  
### TP-Link
* [TP-Link USB Bluetooth Adapter Bluetooth 5.0 (UB500)](https://www.amazon.com/gp/product/B07V1SZCY6/)
  * Chipset: CSR ???
  * `ID 2357:0604 TP-Link TP-Link UB500 Adapter`
  * Performance:
    * Disconnects after some random interval and reconnects
    * When it's connected, It's good
  * Reported by @Arian8j2

Also these are `btmgmt info` output if needed:
**UB500**:
```
hci1:	Primary controller
	addr E8:48:B8:C8:20:00 version 10 manufacturer 93 class 0x6c0104
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr phy-configuration wide-band-speech
	current settings: powered ssp br/edr le secure-conn privacy
	name arch #2
	short name
```
**UB400**:
```
hci0:	Primary controller
	addr 00:1A:7D:DA:71:15 version 6 manufacturer 10 class 0x6c0104
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr phy-configuration
	current settings: powered ssp br/edr le secure-conn privacy
	name arch
	short name
```

rautesamtr, 2023-08-18:
I am having the same issue with random disconnects and immediate reconnects with `TP-Link UB500 Adapter` on Linux (Arch) but not Windows 11. The issue very often occurs immediately at the beginning of lego games after the menu and the level/hub loads and then randomly.

According to kernel source TP-LINK UB500 is Realtek 8761BUV Chipset.

Arian8j2, 2023-08-18:
@rautesamtr If you can, buy **UB400** it doesn't have that problem.

kanedait, 2023-08-31:
> I am having the same issue with random disconnects and immediate reconnects with `TP-Link UB500 Adapter` on Linux (Arch) but not Windows 11. The issue very often occurs immediately at the beginning [...]

thanks for info chipset "TP-LINK UB500 is Realtek 8761BUV"
i don't know how to see the chipset of my TP-LINK.
I think I have found a solution, since it works on Fedora 36 copied these 4 files, and put them on Fedora 38
rtl8761b_config.bin.xz
rtl8761b_fw.bin.xz
rtl8761bu_config.bin.xz
rtl8761bu_fw.bin.xz
add link
https://drive.google.com/drive/folders/1EDSdTsvKfLmJ2cJHkuf1TS5Q_tzjdhfN?usp=drive_link
if I go against the regulation I remove

rautesamtr, 2023-08-31:
Found the chipset trough other bug reports but you can also find it refered in kernel source `linux/drivers/bluetooth/btusb.c` when you look for the usb id.

Testing your firmware and no disconnects so far.
Looks like realtek introduced new bugs with their [firmware update].(https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/rtl_bt/rtl8761bu_fw.bin?id=1de22a39246513ce431de0971a6376093a94c5c1). Firmware problems are [known](https://bugzilla.kernel.org/show_bug.cgi?id=214111) but maybe we should open another ticket specifically regarding the disconnects in the current firmware.

CosmicHeron, 2023-09-15:
I have started encountering the same issue with the ASUS USB-BT500, here is the info `btmgmt info` outputs:
```
hci0:   Primary controller
    addr 04:42:1A:5B:DA:F4 version 10 manufacturer 93 class 0x7c0104
    supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr phy-configuration wide-band-speech 
    current settings: powered bondable ssp br/edr le secure-conn 
    name arch
    short name 
```
When using the same BT adapter and controller on Windows I don't seem to get any random disconnects/reconnects. And oddly enough I don't seem to get reconnects if using a PS5 DualSense controller on Arch. Maybe the driver/firmware update @rautesamtr mentioned changed/broke whatever part xpadneo relies on?

kanedait, 2023-09-18:
> I have started encountering the same issue with the ASUS USB-BT500, here is the info `btmgmt info` outputs:
> 
> ```
> hci0:   Primary controller
>     addr 04:42:1A:5B:DA:F4 version 10 manufacturer 93 class 0x7c0104
>     supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr le advertising secure-conn debug-keys privacy static-addr phy-configuration wide-band-speech 
>     current settings: powered bondable ssp br/edr le secure-conn 
>     name arch
>     short name 
> ```
> 
> When using the same BT adapter and controller on Windows I don't seem to get any random disconnects/reconnects. And oddly enough I don't seem to get reconnects if using a PS5 DualSense controller on Arch. Maybe the driver/firmware update @rautesamtr mentioned changed/broke ~whatever part xpadneo relies on~?
> 
> [Edit] After testing the gamepad without xpadneo, except for RetroArch games seem to detect the controller fine without mapping issues, but the Bluetooth connection still drops occasionally so it's definitely a bluez/driver/kernel problem.

Try the firmware files I posted
your BT adapter probably has the same chipset

kode54, 2023-09-18:
There's an outstanding issue in bluez 5.69 affecting Xbox controllers (and off-topic, Playstation DualShock controllers), fixed in master, not in a release yet. Best to either downgrade to 5.68, or roll your own updated version with the fix cherry-picked.

CosmicHeron, 2023-09-18:
Downgrading Bluez to 5.68 fixed all connectivity issues.

As far as RetroArch goes, messed around some more with and without xpadneo. With it loaded the controller is recognized as an "Xbox Wireless Controller" and works with the `udev` controller driver though the default mapping is all wrong and has to be manually reassigned.
Without xpadneo however the controller is correctly recognized as an "Xbox Wireless S Controller" but inputs simply do not do anything in the frontend (even keyboard presses are ignored, somehow). The controller works in games but the mapping is again all wrong. Switching the controller driver to `sdl2` (which should be using 2.28+ on Arch Linux at least) forced the emulator to use a fallback profile and the controller works albeit without rumble.

I've made sure to put myself into the `input` group as suggested by the Arch wiki to avoid any permission issues.

The DualSense on the other hand works fine on the default controller driver, without mapping issues, so I'm going to chalk this one up to something not being right with RetroArch and its controller profiles.

CosmicHeron, 2023-10-01:
Has anyone else updated to BlueZ 5.70 and checked if the issue still persists for them? Upgrading didn't seem to fix the random disconnects and reconnects with the Xbox Series S controller (using the latest firmware), and I started getting them even when using the downgraded 5.68 version. With and without xpadneo. As previously, these don't happen when using the same controller and USB dongle on Windows 11 or when using the DualSense on Linux so a hardware fault is unlikely.

Annoyingly, using `dmesg -w` no useful message is logged that would indicate what happens, only this:
```
[ 1547.009376] xpadneo 0005:045E:0B13.0010: reverting to original version (changed version from 0x00001130 to 0x00000517)
[ 1547.009381] xpadneo 0005:045E:0B13.0010: reverting to original product (changed PID from 0x028E to 0x0B13)
[ 1550.358823] xpadneo 0005:045E:0B13.0011: BLE firmware version 5.17
[ 1550.358827] xpadneo 0005:045E:0B13.0011: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[ 1550.358829] xpadneo 0005:045E:0B13.0011: working around wrong SDL2 mappings (changed version from 0x00000517 to 0x00001130)
[ 1550.358831] xpadneo 0005:045E:0B13.0011: report descriptor size: 283 bytes
[ 1550.358832] xpadneo 0005:045E:0B13.0011: fixing up Rx axis
[ 1550.358833] xpadneo 0005:045E:0B13.0011: fixing up Ry axis
[ 1550.358834] xpadneo 0005:045E:0B13.0011: fixing up Z axis
[ 1550.358835] xpadneo 0005:045E:0B13.0011: fixing up Rz axis
[ 1550.358836] xpadneo 0005:045E:0B13.0011: fixing up button mapping
[ 1550.358879] xpadneo 0005:045E:0B13.0011: gamepad detected
[ 1550.358880] xpadneo 0005:045E:0B13.0011: enabling compliance with Linux Gamepad Specification
[ 1550.358912] input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0011/input/input55
[ 1550.359009] xpadneo 0005:045E:0B13.0011: input,hidraw13: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on 04:42:1a:5b:da:f4
[ 1550.359032] input: Xbox Wireless Controller Consumer Control as /devices/virtual/misc/uhid/0005:045E:0B13.0011/input/input56
[ 1550.359059] xpadneo 0005:045E:0B13.0011: consumer control added
[ 1550.359079] input: Xbox Wireless Controller Keyboard as /devices/virtual/misc/uhid/0005:045E:0B13.0011/input/input57
[ 1550.359099] xpadneo 0005:045E:0B13.0011: keyboard added
[ 1550.359101] xpadneo 0005:045E:0B13.0011: controller quirks: 0x00000050
[ 1550.359103] xpadneo xpadneo_welcome_rumble start
[ 1551.349714] xpadneo xpadneo_welcome_rumble took 991ms
[ 1551.349719] xpadneo 0005:045E:0B13.0011: Xbox Wireless Controller [3c:fa:06:50:cc:62] connected
```

From what testing I did the disconnects happen exclusively while in-game, never while just connected but on the desktop or something like that. The only exception was when trying to use the game in in the Yamagi Quake 2 source port so it might be that it's reading the inputs in some different way that doesn't cause the connection to drop? But it also doesn't implement rumble in any way afaik if that's a factor, however I got disconnects even when playing games that don't use rumble (Panzer Paladin, for example).

kakra, 2023-10-01:
> From what testing I did the disconnects happen exclusively while in-game, never while just connected but on the desktop or something like that. The only exception was when trying to use the game in [...]

There's a problem with the rumble implementation of the controller: If the game sends rumble too fast, the firmware will crash and the controller restarts, resulting in a disconnect, eventually followed by a reconnect. xpadneo itself has a countermeasure for this bug by reducing the rumble effects to fixed 50ms intervals. But this can only work if your games actually use rumble through the driver.

If your games use hidraw to directly interface with the controller, bypassing xpadneo, xpadneo cannot do its job. If this happens via SDL, that has a similar countermeasure - at least with later version which adjusted the interval from 10ms or 20ms to 50ms. If games use other libraries to directly interface with the controller in raw mode, no such countermeasure will exist.

You can prevent raw access to the controller by finding the hidraw number in the log (in your log above, it's `hidraw13`). Take note that this number will change every time the controller reconnects. Then run `getfacl /dev/hidraw13` to find if your user has read and/or write access and revoke any permissions to that device: `chmod a-rw /dev/hidraw13`.

When properly installed, xpadneo should prevent access to the hidraw device. But we had issues in the past where other software would unconditionally overwrite that behavior and permit world read/write access to any hidraw device (which is a security disaster, btw, because it allows keylogging by any background process). The software we found is OpenRGB and QMK Firmware. Both should have fixed the behavior in latest versions but if you use such software, uninstall it, and then it works, you should report a bug there.

Even if you use software that doesn't implement rumble, the used software layer may still send rumble commands to the controller, even if the values are 0. If that doesn't pass through xpadneo but hidraw instead, there's nothing xpadneo could fix. SDL 2.28 or higher should be safe.

CosmicHeron, 2023-10-01:
That might have been it! I did try `getfacl /dev/hidraw13` and got back:
```
getfacl: Removing leading '/' from absolute path names
# file: dev/hidraw13
# owner: root
# group: root
user::---
group::---
other::---
```
which like you said is expected, the user does not have read permissions. But I was also running [polychromatic](https://github.com/polychromatic/polychromatic) which depends on [openrazer](https://github.com/openrazer/openrazer), which requires the user to be in the `plugdev` group. So I guess it might have been bypassing those permissions entirely? Either way, uninstalled both, rebooted, tried a few games that are rumble heavy and were guaranteed to disconnect within the first of minutes of play. After about 20 minutes I didn't get any dropped connections.

I'll do some more testing to make sure and if that was indeed the cause I'll log an issue on openrazer with a link to this thread.

kakra, 2023-10-01:
Maybe openrazer adds the `plugdev` group to hidraw devices unconditionally, without looking at the vendor or capabilities of the raw device. If that happens, your user will get read/write access to the device and allows games to bypass the driver. You can also try setting `SDL_JOYSTICK_HIDAPI=0` and/or `SDL_JOYSTICK_RAWINPUT=0` so SDL would bypass raw access. I'm not sure which variable does what, I've simply set both.

CosmicHeron, 2023-10-08:
And I'm back at square one, what I thought was the solution stopped working after a few days. Now the Xbox controller will lose connection and reconnect even when just sitting at the desktop. If it does so during a game rumble will stop working and won't work until the controller is manually disconnected and connected again. Setting the above mentioned environment variables doesn't seem to have any effect, unfortunately.

The same controller works with the same dongle plugged into the same USB port fine in Windows, and the DualSense controller does not appear to have any connectivity issues in Linux so this either has to be an ongoing regression with BlueZ and/or the kernel. But until it is fixed the controller is rendered essentially useless in Linux.

kakra, 2023-10-08:
> If it does so during a game rumble will stop working and won't work until the controller is manually disconnected and connected again.

This may be fixed by setting modparam `ff_connect_notify=0`. I'm planning a rework of that feature in v0.10 so there's no longer a delay between detecting the input device and offering rumble.

liberodark, 2023-10-24:
Hi,

Same issue with all Wireless Xbox Controller & TP Link UB500 on Manjaro.
Is Disconnecting after randonmly.

Best Regards

kakra, 2024-03-04:
Closing as reports are not very focused on a single cause. Many of the issues are already fixed, or proven third-party bugs and documented in the troubleshooting docs, or will be fixed with the new rumble infrastructure coming in v0.10.

If you still feel like your individual problem is not solved, please check the other open reports and subscribe or comment there, or wait for v0.10 going live. If no open bug reports fits your problem, feel free to open a new request.

closed 2024-03-04 by kakra