atar-axis/xpadneo #392: Unknown symbol input_ff_create_memless

edt-xx, 2022-11-06:
Hi

On an up to date Arch install (kernel 6.0.7) using xpadneo-dkms-git the module produced will not load.  I get the following message doing insmod as root:

 3419.418610] hid_xpadneo: loading out-of-tree module taints kernel.
[ 3419.418660] hid_xpadneo: module verification failed: signature and/or required key missing - tainting kernel
[ 3419.418739] hid_xpadneo: Unknown symbol input_ff_create_memless (err -2)
[ 3494.717227] hid_xpadneo: Unknown symbol input_ff_create_memless (err -2)

Ideas?

edt-xx, 2022-11-06:
[root@grover ~]# mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode

edt-xx, 2022-11-06:
[root@grover ~]# zcat /proc/config.gz | grep CONFIG_UHID
CONFIG_UHID=m

edt-xx, 2022-11-06:
reverting to kernel 6.0.2 lets the module load again (its still not working - digging to find out what is happening)

kakra, 2022-11-06:
It seems the kernel does not include the ff-memless module. Does it work if you load it manually? We had a similar issue before with the uhid module, ff-memless is not meant for plug&play auto-loading and should be included statically in the kernel.

edt-xx, 2022-11-07:
If I modprobe ff-memless and then hid-xpadneo it works.  Did not realize that ff-memless was a module too.

edt-xx, 2022-11-07:
looks like arch updated its mkinitcpio.conf, this left a .pacnew file here since I have modules and binaries that need to be in the initrd.  Merging my stuff into the new conf file, and reinstalling linux seems to have fixed things here.  Sorry for the noise

closed 2022-11-07 by edt-xx