atar-axis/xpadneo #394: Is it possible to support to build as android kernel module

v-gu, 2022-11-17:
Just wandering, thank you everyone that could leave a comment.

kakra, 2022-11-17:
In theory, it's perfectly possible. But you need the kernel headers and a proper tool chain for compiling the module. Also, you may need features that are not shipped by the stock Android kernel. Building for Android is probably very similar to building for Raspberry Pi.

If you compile the module without the stock kernel headers/sources, you may not be able to load the module due to security restrictions. This would be completely out of scope of this project. Module signing is part of the distribution which is Android or one of its community forks in this case.

Also, to load the module at all, you'd need a jail-broken Android to gain root permissions. Please respect to not ask about jail-breaks here or in our community unless unlocking the boot loader of your device is officially supported by the manufacturer (which then probably allows to easily gain root access).

v-gu, 2022-12-01:
I tried on my phone(it's my own discretion), the build was success, and the kernel was replaced successfully too. After BT connect, the controller rumble just as on Linux. But it seems the Android system didn't pickup the device with rumble support, apps can't recognize the device's rumble feature.

kakra, 2022-12-01:
I'm not sure if Android apps are even supposed to support rumble. It could make sense to look up the SDK documentation to see how it is supposed to work, then look at the AOSP to see how it's actually implemented. Android may filter rumble support by actual device IDs or device strings.

KayJay7, 2024-06-12:
> I tried on my phone(it's my own discretion), the build was success, and the kernel was replaced successfully too. After BT connect, the controller rumble just as on Linux. But it seems the Android system didn't pickup the device with rumble support, apps can't recognize the device's rumble feature.

@v-gu How did you built it on your phone? AFAIK there is no dkms on android and I can't get my computer to cross compile it for arm.

Regarding the rumble features, android applications usually do not use the interfaces exposed by this driver directly. But I'm interested in trying if something like sdl and moonlight can pick it up.

kakra, 2024-06-12:
> if something like sdl

It can if the system allows direct write access to the hidraw device. This is probably not allowed in Android. OTOH, SDL supports rumble through the input API, too, but the Android kernels are probably to old for hid-microsoft to support rumble via input API, or hid-microsoft even supporting the device (and then it's handled by hid-generic which doesn't support rumble).

I'd still not recommend trying xpadneo on Android for general usage because the button mapping differs and the system APIs and/or apps may not expect that.

KayJay7, 2024-06-12:
AFAICT, Android allows accessing hidraw if selinux is set to permissive, which on some (probably all) devices requires root access, but so would dkms, so it's nothing new. For the mappings, there's apps that take care of that. I don't think it should be recommended, but on older android devices it might be the only way to use rumble via Bluetooth.

kakra, 2024-11-21:
Closing because support on Android it out of scope of this project.

closed 2024-11-21 by kakra