atar-axis/xpadneo #395: 8btido ultimate bluetooth/wireless

antonio258, 2022-11-22:
Would it be possible to add support for 8bitdo ultimate? Both in the bluetooth version and in the version with dungle 2.4ghz.

kakra, 2022-11-22:
Does it have support for an Xbox mode? If it does, it should already be supported via Bluetooth.

For adding other connection methods (e.g., dongles) and modes (e.g. Switch mode), I'd need to have such a controller. At least Switch mode support may follow in the next development cycle, I got donated a GuliKit controller and Switch Bluetooth mode looks easy enough to add support.

kakra, 2022-12-06:
This project is neither about xboxdrv nor xpad. I'd expect that dinput would be the mode to use for xpadneo, and I cannot guess why rumble should not work then. Without providing log data for xpadneo connecting the device, there's not much I could guess from this report.

Are you asking for support in xpadneo? If so, there are plans to support native Switch mode in the future (see above). Until then, dinput mode should do but xinput mode should also work as long as it can connect via Bluetooth. If you're using a special donge from 8BitDo, we may be out of luck unless I can effort such a controller including the dongle to test myself.

You're the second person to tell that dinput doesn't support dongle although all protocol support seems to be in place. This looks like an artificial restriction of the dongle to me. But then again, I'm not even sure if xpadneo is actually used here.

All your udev rules seem to load user-space helpers, so it's bypassing xpadneo completely.

btegs, 2022-12-17:
@LuisArtDavila - There was a merge in xpad recently that may help: https://github.com/paroj/xpad/commit/1821d2822783e8c24733b05138e06d9a9f4d309d

kakra, 2022-12-18:
@btegs Thanks, that looks interesting. I could add a similar patch to xpadneo but we need to ensure that the device identified by those IDs is actually a HID device. Since that patch is in xpad which is not a HID driver, this is probably not a HID device and you rather use xpad or xone for it (the latter may need a similar patch). We currently do not support the GIP protocol in xpadneo (though, this is on my feature list but not planned yet).

With the beta firmware, probably only Switch mode is HID protocol, maybe Android mode (but that seems to miss rumble support). If you guys could share the HID descriptors and device IDs of those modes, we may add some early alpha support in xpadneo. If a mode doesn't offer HID, this has to wait.

btegs, 2022-12-18:
@kakra - Thanks for the update. I also recently got an 8bitdo Pro 2 and using the latest xpadneo makes the rumble run constantly and have to turn off the controller to make it stop. My Xbox Elite 2 does work great though and would be nice to just use xpadneo if i get the Ultimate Wireless down the road as well.

If i were to get the Ultimate controller today, what would be the commands I would run in the terminal to get the information you need? Could try the DirectInput and Xinput modes: https://download.8bitdo.cn/Manual/Controller/Ultimate/Ultimate-2.4g-wireless-Controller.pdf

kakra, 2022-12-18:
> I also recently got an 8bitdo Pro 2 and using the latest xpadneo makes the rumble run constantly

Yep, this is why I have not backported the last change to the stable branch yet. I think I have to get that controller myself to figure out what's going wrong.

I'll look into the links later.

btegs, 2022-12-19:
@kakra - Forgot to mention this, but here is an AUR package if you want to reference some udev rules: https://aur.archlinux.org/packages/8bitdo-ultimate-controller-udev

djust270, 2023-01-19:
> @btegs Thanks, that looks interesting. I could add a similar patch to xpadneo but we need to ensure that the device identified by those IDs is actually a HID device. Since that patch is in xpad [...]

According to USBDeview in Windows

Devicename : XBOX_360_DEVICE_00:00
VendorID : 2dc8
ProductID: 3106

HTML report here https://justgeeks-my.sharepoint.com/:u:/g/personal/dave_justgeeks_co/EV1EX42ktidNrpWijSCrCEsBra4U30fzj1M3W5V6qwa3xA?e=nwn9dA

btegs, 2023-01-19:
@djust270 - It seems like 8bitdo reuses IDs: https://github.com/paroj/xpad/commit/4a64f5caefb2e571d90240f9b1a2bbe1a8950a74