atar-axis/xpadneo #397: Series Controller shows up as Xbox 360 in Steam

TiagodePAlves, 2022-11-30:
## Version of xpadneo

0.9.5 (from AUR [xpadneo-dkms](https://aur.archlinux.org/packages/xpadneo-dkms) 0.9.5-2)

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller (firmware 5.15)
- [ ] Other:

### Connection mode

- [X] Bluetooth connection (through a CSR `0a12:0001` dongle)
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [X] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [X] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [X] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [X] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [X] Fantastic work but ... 
- [X] I can code and I want to help

## Describe the Bug

When xpadneo is installed and loaded, my Xbox Series Controller is recognized in Steam as a Xbox 360 Controller, even named like that in General Controller Settings and Big Picture Configuration. This only happens in Bluetooth, the controller is recognized correctly with a wired connection. The odd part for me is that without xpadneo installed, the controller is labelled correctly in Steam (tested wired and Bluetooth connections). 

Not sure if this expected or not.

## Steps to Reproduce

## Expected Behavior

Controller should show up as Xbox One Controller in Steam settings.

## Screenshots / GIFs / Videos

[image]

## System Information

```console
# uname -a
Linux marmis-arch 6.0.10-zen2-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Sat, 26 Nov 2022 16:51:24 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/10119490/xpadneo-btmon.txt)

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/10119491/xpadneo-dmesg.txt)

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/10119493/xpadneo-lsusb.txt)

## Additional Context

Although the labeling is weird, the controller works perfectly.

man-stein, 2022-11-30:
The same thing happen for me.

yswtrue, 2022-11-30:
The same thing happen for me.

kakra, 2022-11-30:
This works as designed in the current versions. It's just cosmetic, the controller is still fully supported. We cannot currently change that. See also #286

closed 2022-11-30 by kakra