atar-axis/xpadneo #400: 8BitDo Pro 2 non-stop rumble on connect

orbea, 2022-12-13:
## Version of xpadneo
https://github.com/atar-axis/xpadneo/commit/01503462e5a480f28a638f1d64ad7399446c22d3

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: `8BitDo Pro 2`

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
When connecting the controller the rumble never stops.

## Steps to Reproduce
1. Turn on the controller.
2. It doesn't stop rumbling.

## Expected Behavior
The rumble notification when connecting the controller should stop after a short period.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux gentoo 5.15.82-x86_64 #1 SMP PREEMPT Mon Dec 12 21:42:15 PST 2022 x86_64 Intel(R) Core(TM) i5-7600K CPU @ 3.80GHz GenuineIntel GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/10215482/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/10215483/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/10215484/xpadneo-lsusb.txt)

## Additional Context
I bisected this issue to commit 94ad82a4f5be2173dd663c36ea8fd8dbf4c013bd,
```
94ad82a4f5be2173dd663c36ea8fd8dbf4c013bd is the first bad commit
commit 94ad82a4f5be2173dd663c36ea8fd8dbf4c013bd
Author: Kai Krakow <kai@kaishome.de>
Date:   Tue Nov 15 22:01:18 2022 +0100

    xpadneo, init: Fix rumble testing logic on connect
    
    There was a flaw in the rumble testing logic during connect which
    caused controllers with motor masking bits but without rumble envelope
    parameters to rumble indefinitely after connecting. This became
    apparent while adding support for the GuliKit controller family.
    
    See-also: https://github.com/atar-axis/xpadneo/issues/391
    Signed-off-by: Kai Krakow <kai@kaishome.de>

 hid-xpadneo/src/hid-xpadneo.c | 21 +++++++--------------
```

kakra, 2022-12-13:
Somehow I was suspecting this... I'll look into it.

praxis88, 2022-12-21:
Ill throw in on this one

Controller connects, non-stop rumble, and doesnt register any other input

seems to work with the same setup with xpadneo out of the equation

dmanlfc, 2023-02-24:
what is the progress on this one @kakra is there a workaround for the time being?

praxis88, 2023-03-12:
I tested this today, It still does the vibrate thing, just softly now.

The good news is the inputs work whereas before it was just strong virbration and no controll

kakra, 2023-04-05:
Can you guys please share an Amazon link of the exact model you're using?

orbea, 2023-04-05:
@kakra I have this one.

https://www.amazon.com/gp/product/B07WRCHV8S

dmanlfc, 2023-04-06:
Thanks @kakra the problem occurs with Bluetooth

orbea, 2023-04-06:
I also use bluetooth.

praxis88, 2023-04-06:
This is the one i have, used over bluetooth

8BitDo Pro 2 Bluetooth Controller for Switch, PC, Android, Steam Deck, Gaming Controller for iPhone, iPad, macOS and Apple TV (Gray Edition) https://a.co/d/2IbXcl6

kakra, 2023-04-14:
I've ordered the controller, and it arrived. Will test soonish.

closed 2023-04-14 by kakra

praxis88, 2023-04-14:
good stuff, I just tested this and everything seems to be working as intended

kakra, 2023-04-15:
Thanks for confirming, so backporting to v0.9 should also work. It's already queued, I'm just waiting for some feedback like yours.

dmanlfc, 2023-04-15:
Works here too @kakra thanks.

orbea, 2023-04-15:
@kakra Works on my end too, thanks!