atar-axis/xpadneo #401: Add support for 8BitDo Ultimate (2.4Ghz proprietary protocol with receiver) controller

adam900710, 2022-12-16:
## Version of xpadneo
0.9.5

## Controller Model
- [ ] Other:

8BitDo Ultimate 2.4Ghz Controller with Charging Dock
https://www.8bitdo.com/ultimate-2.4g-wireless-controller/

Please don't get confused with the bluetooth version, this one only supports its proprietary 2.4Ghz protocol.

### Connection mode

- [ ] Other
Proprietary 2.4Ghz receiver OR wired USB connection.

## Describe your feature reqeust

- [ ] I'd like to see support for a specific model

In fact, there is already working method using xboxdrv:
https://aur.archlinux.org/packages/8bitdo-ultimate-controller-udev

**Describe the solution you'd like**

Like to see xpadneo just adds the extra device id and needed quirks for that gamepad.

**Describe alternatives you've considered**
https://aur.archlinux.org/packages/8bitdo-ultimate-controller-udev
Which uses xboxdrv to handle the device.

**Extra info**

The receiver and the gamepad USB connection have the same USB id(s):
```
2dc8:3106: 8BitDo 8BitDo Ultimate wireless Controller for PC
2dc8:3109: 8BitDo IDLE
```

The hid id is `0003:2DC8:3109`, from the hid-generic output (only on IDLE):
```
[  105.822181] hid-generic 0003:2DC8:3109.0008: hiddev98,hidraw5: USB HID v1.10 Device [8BitDo IDLE] on usb-0000:00:14.0-1/input0
```

Tried modifying the `modprobe.d` rules, but not working:

```
# For 8bitdo Ultimate 2.4G recevier/gamepad
alias hid:b0003g*v00002DC8p00003106 hid_xpadneo
alias hid:b0003g*v00002DC8p00003109 hid_xpadneo
```

kakra, 2022-12-16:
xpadneo does not currently support USB devices. This is technically a USB device due to the receiver, xpadneo currently only talks to Bluetooth devices (or rather: HID devices). If you connect the controller in Bluetooth mode, it should already be supported.

 But still, it is a HID device. So it may be possible to unbind the device from hid-generic, then rebind to hid-xpadneo via sysfs. But unless it's using an Xbox-compatible HID protocol, xpadneo won't support it. hid-generic should already support the controller by default.

What's the real problem with using hid-generic? Are any functions missing?

That said, if your controller is Switch-compatible, future updates to xpadneo may support it.

adam900710, 2022-12-16:
> But still, it is a HID device. So it may be possible to unbind the device from hid-generic, then rebind to hid-xpadneo via sysfs. But unless it's using an Xbox-compatible HID protocol, xpadneo won't support it. hid-generic should already support the controller by default.

I guess since xboxdrv can handle it, the hid protocol should be mostly XBOX compatible?

The hid-generic problem is, no joystick interface at all, thus most programs can not support the device.

Anyway, thanks for the detailed explanation, I totally understand it's not going to be supported anytime soon, thus I'm fine using the xboxdrv solution for now.

closed 2022-12-16 by adam900710