atar-axis/xpadneo #402: Xbox controller connects, but does not rumble. Steam is not detecting it

jdiadno, 2022-12-21:
Hello,

I'm trying to connect controller "Gamepad Xbox Wireless Controller Carbon Black" to Ubuntu 20.04, kernel 5.15.0-56-generic, but unsuccessfully.

version hid-xpadneo-v0.9-131

I followed the installation instructions

```console
/opt/xpadneo$ sudo ./install.sh 
* creating dkms.conf
* adding hid-xpadneo-v0.9-131-g0150346 folder to /usr/src
* installing module (using DKMS)

Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-131-g0150346/source ->
                 /usr/src/hid-xpadneo-v0.9-131-g0150346

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.0-56-generic -C /lib/modules/5.15.0-56-generic/build M=/var/lib/dkms/hid-xpadneo/v0.9-131-g0150346/build/src VERSION=v0.9-131-g0150346 modules...
Signing module:
 - /var/lib/dkms/hid-xpadneo/v0.9-131-g0150346/5.15.0-56-generic/x86_64/module/hid-xpadneo.ko
Adding '/var/lib/shim-signed/mok/MOK.der' to shim:
cleaning build area...

DKMS: build completed.

hid-xpadneo.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.15.0-56-generic/updates/dkms/

Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...

depmod....

DKMS: install completed.
```

Then, connecting with bluetooth. It's correctly connected, the light on the controller is still, not blinking, ubuntu correctly recognizes it, the battery indicator is working, but it did not rumble during the pairing and Steam is not showing it when I go to general controller settings, no controller detected. 

I also tried xboxdrv, but with that, 4 controllers were listed there all the time, even when I powered controller down.

I'm not using any other software mentioned in the list. 

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

The controller works with iPhone

I don't have USB dongle

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/10273682/xpadneo-btmon.txt)

kakra, 2022-12-21:
This is a known issue with the Intel BT chipset, it also happens on Windows. Actually, xpadneo will see the device but it immediately disconnects after this in the Bluetooth layer and the driver doesn't even notice this until some time later when it eventually times out in the BT layer.

There's currently no known work-around for this other than using a different USB chipset, the CSR 4.0 dongles are known to work (if they are genuine and not China clones). I'm using a TP-LINK UB400 dongle successfully.

jdiadno, 2022-12-27:
Thank you for taking time to reply

closed 2022-12-27 by jdiadno