atar-axis/xpadneo #403: Add link to module signing instructions in the installation instruction

KayJay7, 2022-12-30:
If secure boot is enabled, the module won't load unless it's signed and the key is installed on the system; apparently dkms *should* take care of all this by default, but it did not for me (on fedora 37, which did not come with dkms preinstalled).
The current signing process is fairly easy and explained on the [DKMS's readme](https://github.com/dell/dkms) in the `module signing` and `secure boot` sections.

I saw you had some instructions on how to setup signing and removed because they weren't needed anymore, but apparently they sometimes are, so it might be good to put them back.

An inexperienced user might not understand why it's not working and give up. And if they find the problem and look it up on google, they will find a lot of tutorials suggesting solutions similar to your old solution, which with current dkms are suboptimal.

I'm not suggesting to write the entire signing instructions in the readme, but you could add a couple lines saying "if you get this problem, follow the instruction at this \[dkms's readme\] link".

kakra, 2022-12-30:
Usually, such modules should ship with a native package which correctly signs the module. In the real world, tho, packages often do not exists and DKMS can be used instead. It is probably a bug that DKMS doesn't sign the modules in Fedora and should be reported there. Nevertheless, I'll add the link. Please keep this issue opened until it will be closed by that commit.

KayJay7, 2022-12-30:
Thanks for the quick response.
Dkms does try to sign modules but it fails, it needs to be set up as per the link (providing a key pair and importing the certificate in shim). I assumed this was because I had to install dkms myself (while on other distros it comes installed by default).
Anyway, I will get in touch with the fedora dkms package maintainer to see if it's a mistake on their part or what.

closed 2024-03-04 by kakra