atar-axis/xpadneo #404: Debian repo support suggestion

Apacelus, 2023-01-05:
Hey, have you ever thought about making a debian repo? You could even host it via github pages, like we do at the eupnea project ([apt repo](https://github.com/eupnea-linux/apt-repo)). 

Should pe pretty easy to setup, if you just clone that repo and adjust it to your needs.

kakra, 2023-01-05:
If we can have a deployment bot which automatically generates dpkg packages and repo meta data, we can in theory have this. But we would also need proper CI, and it should probably also build packages for other distributions.

As a first step, I could look into generating dpkg packages right from the Makefile so we could easily attach those to the releases page.

OTOH, what's wrong about DKMS? This should work out of the box for most distributions. The single downside is that the source code isn't managed via the package manager.

signed-log, 2023-01-08:
@kakra 

I could help with that, I am in the process of porting xpadneo to openSUSE TW and the [OpenSUSE Build Service](https://build.opensuse.org) can act as a Debian repo integrated with the Debian build

I already did that, and I am the maintainer for https://github.com/jarun/nnn, which is accessible from many distros from its [OBS](https://build.opensuse.org/project/show/home:stig124:nnn) repo

Now, I don't know how well that would play out, but if you'd be interested, I can give it a try

kakra, 2023-01-09:
@signed-log Yeah sure, why not? We could add a link to the OpenSUSE Build Service then. Any contribution is appreciated. You should look at how other packages build kernel modules, maybe skipping DKMS completely if such packages do that, too. Please check module signing in that case, it seems to be an important thing on some distros which support secure boot.

signed-log, 2023-01-09:
Great, will have a go when I can,

Debian packaging is quite tricky in general, but I think I can make it work