atar-axis/xpadneo #405: Can't install the kernel module

sBinott0, 2023-01-10:
## Version of xpadneo

v0.9-131-g0150346

## Controller Model

-Xbox Series X|S controller

### Connection mode

 Bluetooth connection

## Severity / Impact

- I've read the docs and the bug reporting instructions
- I've applied the latest firmware update to the controller
- It does not work at all
- I probably didn't figure it all out but it's too early to give up

## Describe the Bug

When installing the module this is the feedback:

 creating dkms.conf
 adding hid-xpadneo-v0.9-131-g0150346 folder to /usr/src
 installing module (using DKMS)
Sign command: /lib/modules/6.0.17-300.fc37.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-131-g0150346/source -> /usr/src/hid-xpadneo-v0.9-131-g0150346

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.0.17-300.fc37.x86_64 -C /lib/modules/6.0.17-300.fc37.x86_64/build M=/var/lib/dkms/hid-xpadneo/v0.9-131-g0150346/build/src VERSION=v0.9-131-g0150346 modules...
Signing module /var/lib/dkms/hid-xpadneo/v0.9-131-g0150346/build/src/hid-xpadneo.ko
Cleaning build area...

hid-xpadneo.ko.xz:
Running module version sanity check.
**modinfo: ERROR: could not get modinfo from 'hid_xpadneo': Invalid argument**
 - Original module
   - Found /lib/modules/6.0.17-300.fc37.x86_64/extra/hid-xpadneo.ko.xz
   - Storing in /var/lib/dkms/hid-xpadneo/original_module/6.0.17-300.fc37.x86_64/x86_64/
   - Archiving for uninstallation purposes
 - Installation
   - Installing to /lib/modules/6.0.17-300.fc37.x86_64/extra/

Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...
depmod.....

After that, I try to sign hid-xpadneo the same way I signed VBox modules.

/usr/src/kernels/$(uname -r)/scripts/sign-file sha256 /home/rafael/.certificados/MOK.priv /home/rafael/.certificados/MOK.der /lib/modules/6.0.17-300.fc37.x86_64/extra/hid-xpadneo.ko.xz

Then I try to enable the kernel module and I obtain this:

sudo modprobe hid-xpadneo
modprobe: ERROR: could not insert 'hid_xpadneo': Invalid argument

## Steps to Reproduce

It happens on install

## Expected Behavior

The installation should be succesfull

## System Information

# uname -a

Linux PC-TUX 6.0.17-300.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Jan 4 15:58:35 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0

## Additional Context
I think the main problem is that I have secure boot enabled, but this is compulsory due to some anticheats.

kakra, 2023-01-10:
Fedora seems to need special instrucations, see also https://github.com/atar-axis/xpadneo/issues/403

kakra, 2024-11-21:
Maybe related: #431

kakra, 2024-12-18:
Please test #517

closed 2024-12-23 by kakra