atar-axis/xpadneo #407: Series X|S Controller Not Recognized (Error -22)

BorisBrock, 2023-02-12:
## Version of xpadneo
0.9.5-2

## Controller Model
- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software
- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information
Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [X] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [X] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [X] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [X] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact
- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
The controller pairs fine via bluetooth. But after that it is not usable as a gamepad.
sudo dmesg shows the following error:

> [  656.036595] xpadneo 0005:045E:0B13.000B: BLE firmware version 5.13
> [  656.036601] xpadneo 0005:045E:0B13.000B: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
> [  656.036604] xpadneo 0005:045E:0B13.000B: working around wrong SDL2 mappings (changed version from 0x00000513 to 0x00001130)
> [  656.036606] xpadneo 0005:045E:0B13.000B: report descriptor size: 110 bytes
> [  656.036608] xpadneo 0005:045E:0B13.000B: fixing up Rx axis
> [  656.036609] xpadneo 0005:045E:0B13.000B: fixing up Ry axis
> [  656.036610] xpadneo 0005:045E:0B13.000B: fixing up Z axis
> [  656.036612] xpadneo 0005:045E:0B13.000B: fixing up Rz axis
> [  656.036663] xpadneo 0005:045E:0B13.000B: unbalanced collection at end of report description
> [  656.036669] xpadneo 0005:045E:0B13.000B: parse failed
> [  656.036671] xpadneo: probe of 0005:045E:0B13.000B failed with error -22 

## Steps to Reproduce
1. Turn on series X|s controller
2. Pair gamepad via Bluetooth

## Expected Behavior
After pairing the gamepad I would expect it to be visible in the system and fully usable for gaming.

## Screenshots / GIFs / Videos
N.a.

## System Information

```console
# uname -a
Linux gaming-pc 6.1.9-1-MANJARO #1 SMP PREEMPT_DYNAMIC Wed Feb  1 20:38:52 UTC 2023 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: correct 'xxd' to 'xsd' [nyae]?
```

## Controller and Bluetooth Information

## Additional Context

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/10717264/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/10717265/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/10717266/xpadneo-lsusb.txt)

kakra, 2023-02-12:
Is this the latest firmware? I didn't try it yet...

BorisBrock, 2023-02-13:
> Is this the latest firmware? I didn't try it yet...

I've updated my controller the last time I was on a Windows machine, which was roughly one month ago (if I remember correctly). So the firmware should be pretty up-to-date.

kakra, 2023-02-13:
Okay, so I'll need to update my XBXS controller to reproduce... The version (some months) old I still have installed on it works just fine.

grafgooseman, 2023-04-14:
Even the wired Xbox X controller connection doesn't work for me. (doesn't matter that the Wiredly Connection Xbox X controller is identified as XBox 360 by Gamepad Tester Website)

kakra, 2023-04-14:
For some unknown reason, I couldn't update the firmware yet via my KVM Windows. All my Windows VMs are somehow broken by complaining not being able to create the swap file - resulting in all kinds of errors and stalling when starting applications. I'll test it as soon as updating works again for me.

kakra, 2023-04-14:
Okay, I managed to find what was causing the issues with my VMs (disk IO timeouts, vioscsi seems to not be usable currently in Windows). I successfully booted my Windows 11 VM and updated all my controllers to the latest version. Each one (XB1S, XBE2, XBXS) seems stable, they reliably connect to Bluetooth, connection mode switching (by double-clicking the connect button) works flawlessly (controller goes back and forth to the Windows VM via Xbox dongle and Linux via Bluetooth), connecting via USB works fine, there's no input delay or dropped packets, no errors or warnings in dmesg.

@grafgooseman I wonder if there's an issue with your Bluethooth chipset. For using USB, that doesn't go through xpadneo at all, and gamepad tester websites may not be compatible with some of the custom drivers, e.g. it doesn't play well with Chrome and xpadneo). xpadneo only supports HID devices, and that is only for Bluetooth connections, xpadneo even doesn't know about Bluetooth, it just sees the HID interface provided by Bluetooth user-space (5.x firmware) or by the kernel input layer (4.x firmware).

@BorisBrock I cannot reproduce your issue here (unbalanced collection at end of report description) on the same kernel branch (6.1) and latest Xbox Acc updates from Microsoft. Can you retry with the xpadneo master branch? Distributions may have a package xpadneo-git which installs the latest development version. Maybe your Bluetooth stack is improperly patched in the HID code, so maybe check if you can upgrade or downgrade Bluez.

OTOH, I do not see this log in the dmesg file you've attached:
```
[ 656.036671] xpadneo: probe of 0005:045E:0B13.000B failed with error -22
```

In your dmesg log, everything looks fine. According to the timestamps, the log even seems to be from another boot or another system.

kakra, 2023-06-14:
Closing due to no feedback, this is a driver issue, cannot reproduce here with current firmware.

closed 2023-06-14 by kakra

BogdanTheGeek, 2024-04-29:
I was running into the same issue, only 1 of my 3 xbox one s controllers would connect and work. I managed to (somehow) fix it by just plugging them into a windows 10 machine and waiting for the drivers to install. After that, they just connected on linux.

kakra, 2024-04-29:
> I was running into the same issue, only 1 of my 3 xbox one s controllers would connect and work. I managed to (somehow) fix it by just plugging them into a windows 10 machine and waiting for the drivers to install. After that, they just connected on linux.

This is strange... But I think I remember the early firmware versions did something similar: After being connected to Windows once, they worked differently.