atar-axis/xpadneo #408: Xbox one S controller no more get detected by BT

deathxxx123, 2023-02-14:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Just controller cannot be seen in BT list. It works on Win11. Last time i was using on Win11 and maybe he receive some firmware update, because the blinking are little different now. I try roll back firmware, but no option for that. Just update. I used this controller for long plays in Linux long time. I have PS4 controller and is discoverable in BT and it's working on linux.
With cable is working without any problem.

## Steps to Reproduce

Just update to the lastest firmware and test in on linux.

## Expected Behavior

I expect BT to detect controller, when is in pairing mode.

## Screenshots / GIFs / Videos

## System Information

Pop!_OS 20.04

```console
# uname -a
Linux pop-os 6.0.12-76060006-generic #202212290932~1674139725~22.04~ca93ccf SMP PREEMPT_DYNAMIC Thu J x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

## Controller and Bluetooth Information

## Additional Context

Still not tested in other clean linux. Will do now.

deathxxx123, 2023-02-14:
I tested on my other Manjaro distro, where i left all working with this controller. It's still in BT list, but can't connect and can't see in refresh list as pairing mode.
Definitely its from lastest controller Firmware.
But not only xpadneo, but with a clean linux now showing in list too, but it should.
Maybe that is MS attempt to lock down controller to use only with Win ..

kakra, 2023-02-14:
Please check if #407 is what you're seeing.

deathxxx123, 2023-02-14:
Yea, i just test it on another live distro and Xbox controller is not discoverable for BT.
But PS4 again is discoverable instantly.
I check again from Win11 for Xbox controller, and yes, it is discoverable and Win see it immediately and ask for connection to a new device. So the controller is working.
To be clear - i do not really care about this controller, i use PS4 one, but i just try my long not used Xbox controller and... surprice - i cannot use in with BT on linux.
It's a good controller but my hands start to pain with this one after 4-5 hours. With PS4 is more resting.

deathxxx123, 2023-02-14:
> Please check if #407 is what you're seeing.

It is not the same, i cannot pair the controller, because BT won't see it.
And cannot roll back firmware.. because there are no such option.
It's Xbox One S controller: model 1708

kakra, 2023-02-14:
It's a known problem, nothing to worry about how MS might try to block the controller on Linux - it's not that way. It looks like they changed the (somewhat broken) HID descriptor, and our patching repairing the descriptor now goes wrong. I'll have to look into it. I'm closing this as duplicate of #407, please follow up there.

kakra, 2023-02-14:
> > Please check if #407 is what you're seeing.
> 
> It is not the same, i cannot pair the controller, because BT won't see it. And cannot roll back firmware.. because there are no such option. It's Xbox One S controller: model 1708

Okay, so your previous "yes" wasn't actually a reply to my previous reply? ;-)

Please generate the dmesg and bluetoothctl output as asked for in the bug reporting instructions.

closed 2023-02-14 by kakra

deathxxx123, 2023-02-14:
dmesg just shows noting

Bluetooth monitor ver 5.64
= Note: Linux version 6.0.12-76060006-generic (x86_64)                 0.051292
= Note: Bluetooth subsystem version 2.22                               0.051293
= New Index: F8:89:D2:7A:09:3E (Primary,USB,hci0)               [hci0] 0.051294
= Open Index: F8:89:D2:7A:09:3E                                 [hci0] 0.051294
= Index Info: F8:89:D2:7A:09:3E (MediaTek, Inc.)                [hci0] 0.051294
@ MGMT Open: bluetoothd (privileged) version 1.22             {0x0001} 0.051295
@ RAW Open: blueman-manager version 2.22                      {0x0002} 0.089736
@ RAW Close: blueman-manager                                  {0x0002} 0.089741
@ RAW Open: blueman-manager version 2.22                      {0x0002} 1.095422
@ RAW Close: blueman-manager                                  {0x0002} 1.095432
@ RAW Open: blueman-manager version 2.22                      {0x0002} 2.100948
@ RAW Close: blueman-manager                                  {0x0002} 2.100956
@ RAW Open: blueman-manager version 2.22                      {0x0002} 3.106654
@ RAW Close: blueman-manager                                  {0x0002} 3.106664
@ RAW Open: blueman-manager version 2.22                      {0x0002} 4.112151

deathxxx123, 2023-02-14:
It sees everything, but not Xbox controller. Just not exist.

deathxxx123, 2023-02-14:
I can try something...
I have one Dongle BT 4.0

deathxxx123, 2023-02-14:
Wow intresting.. its not from controller or maybe is from firmware.
With this USB BT 4.0 dongle, controller is detected and even connected and paired.
hmmmmm

deathxxx123, 2023-02-14:
I have X570S AORUS ELITE AX motherboard with integrated BT and wi-fi.
It is working with my other hardware, and on Win11.
And was working with linux with this controller long time without problem.
Not this little dongle see my controller, but not the one in MB.

deathxxx123, 2023-02-14:
So i will investigate further tomorrow.

deathxxx123, 2023-02-14:
So finally i guess the lastest controller `firmware` regress with my integrated MB bluetooth.
It is not from kernel updates, because i try my old distro non-touched  and not updated since last time i used for gaming with same hardware and controller.

deathxxx123, 2023-02-14:
If there is some way to roll back firmware version on controller... That will fix this stupidness..

kakra, 2023-02-15:
> With this USB BT 4.0 dongle, controller is detected and even connected and paired.

The latest firmware needs a dongle capable of BLE. I think that usually requires BT 4.0 or later. Also, some onboard BT controllers are really crappy under Linux for some reason. This would be a bug outside of xpadneo's scope, xpadneo really doesn't interact with Bluetooth at all, it's just a HID driver.

There's a way to roll back, I think you'd need to run (via Win+R in Windows) `xboxaccessories://firmwareupdate?legacyDowngrade=true` but I think this only works for models that initially came with firmware 4.x, like XB1S or XBE2, so your's should work as you initially reported it to be XB1S.

deathxxx123, 2023-02-15:
> The latest firmware needs a dongle capable of BLE. I think that usually requires BT 4.0 or later. Also, some onboard BT controllers are really crappy under Linux for some reason. This would be a bug [...]

Yes i think mine was version 4xx when come.
The MB-bt working great, but not with this controller anymore. When i succes downgrade firmware i will post here results.

deathxxx123, 2023-02-15:
Wow reverting firmware to v3xxx is working now!
So my MB-bt can recognize pairing mode controller and ofcource it's connected and vibrated from my linux.
So the problem is the lastest update to v5xxx that regress with my MB-bt. It was working with v4xxx too, but now i cannot install that version. I only can install the initial version when controller came, or the lastest.
I will look where i was post info about my MB- bluetooth exact model.

deathxxx123, 2023-02-15:
Yes, there is link about my integrated BT model (AMD RZ608 / MediaTek MT7921K):
https://github.com/atar-axis/xpadneo/issues/93#issuecomment-1114205181

So no one can reproduce the issue, but only one that have this BT model or this Motherboard.

deathxxx123, 2023-02-15:
Any suggestions, where i can report that to MS? Maybe they will willing fix this...

deathxxx123, 2023-02-15:
Or this can be fixed via the linux BT driver?

deathxxx123, 2023-02-15:
I report this on Bluez https://github.com/bluez/bluez/issues/478 
So i hope this can be fixed here..

kakra, 2023-02-15:
> Or this can be fixed via the linux BT driver?

Since Windows does work, it's probably an issue with the Linux BT drivers. It's not an issue with the controller firmware, it does actually work in BLE (Bluetooth Low Energy) mode on Linux (actually, it's working much better for me in BLE mode which comes with firmware 5.x).

deathxxx123, 2023-02-16:
Yea... i will try my best to resolve this problem, but will be a hard one, because no program detecting anything with lastest firmware.

deathxxx123, 2023-02-16:
But why BT drivers work with other devices, but not with my integrated MB bluetooth.. but working with previous firmware. Maybe driver is different for my BT.

kakra, 2023-02-16:
It's most likely that the driver for your integrated motherboard BT cannot properly handle Bluetooth Low Energy mode. The Windows driver can, the Linux driver can't. As long as you use devices without BLE, they will work. But the 5.x firmware uses BLE, so the driver for your motherboard BT cannot handle the device.

The Bluetooth driver just provides the data transport layer. The actual Bluetooth devices (controller, mouse, keyboard, headphones etc) usually don't know about Bluetooth at all, they are provided a data bus by the lower level Bluetooth driver and cannot even interact with Bluetooth properties. And if that lower driver can handle standard BT, upper layer drivers will work, if the lower driver cannot handle BLE, upper layer drivers won't work.

In reality, things are even more complex. xpadneo doesn't even access the data bus directly, it gets a HID bus from an intermediate driver layer that sits between Bluetooth driver and input driver layer that xpadneo uses.

deathxxx123, 2023-02-16:
How to check if uses BLE mode? I have PS4 controller witch working with my MB bluetooth.

kakra, 2023-02-16:
PS4 controllers are classic BT, not BLE.

deathxxx123, 2023-02-23:
Working with BLE with:
Intel Wireless-AC 9260 Bluetooth Adapter

kakra, 2023-02-25:
> Yes, there is link about my MB (Gigabyte X570S AORUS ELITE AX) integrated BT model (AMD RZ608 / MediaTek MT7921K):
> [#93 (comment)](https://github.com/atar-axis/xpadneo/issues/93#issuecomment-1114205181)

I've actually locked that thread because it is way outdated, has been closed already, and it is difficult to track individual comments for new issues outside of the original scope of an issue (iow, that issue report was way too generic). If there's a documentation change request for any BT support, a new issue should be raised dedicated to that chipset, otherwise I cannot properly track documentation updates.

This issue has also been closed because the discussion indicates that it is an issue outside of xpadneo's scope or has been outlined in another issue already (tho, the latter turned out to be wrong). But if you have individual findings that should be detailed in the documentation, I'd happily add these to the documentation through opening a new issue report.

To be clear, I don't want nullify your testing efforts, I very much appreciate it. But we really should be opening a new issue for documentation updates about your findings, it's hard to track in a report if the findings deviated from the original report too much. That said, thanks a lot for your efforts testing the failure conditions.

deathxxx123, 2023-03-05:
Ty for this respond. I just remove my other BT adapters and stay with my integrated on motherboard, because having a multiple BT devices, my linux is confusing and gives a dmesg errors. There is no way in Bios to disable MB bluetooth.
So i stay with my MB bluetooth and PS4 controller that works absolutely outstanding. Never had a disconnect or problem with connecting.
And my Xbox controller will collect dust until something is happen. (it's a heavy controller anyway)

deathxxx123, 2023-03-05:
Should i open new issue? But this is not xpandeo issue, but MT7921K BLE driver function issue.

deathxxx123, 2023-03-05:
I think this need to added to BT model that normal BT is working, but BLE not working on linux.
Should looks like this:

Model: `AMD Wi-Fi 6E RZ608 + BT/BLE (MT7921K) 5.2 combo`
Working: `BT`
Not working: `BLE`