atar-axis/xpadneo #409: missing kernel headers when installing from AUR

rmullin7286, 2023-02-15:
## Version of xpadneo
0.9.5

## Controller Model

- [x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [x] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
When installing the driver using `yay -S xpadneo-dkms` from the AUR, The following error message is printed

```
(1/1) checking keys in keyring                                                                                                  [##############################################################################] 100%
(1/1) checking package integrity                                                                                                [##############################################################################] 100%
(1/1) loading package files                                                                                                     [##############################################################################] 100%
(1/1) checking for file conflicts                                                                                               [##############################################################################] 100%
(1/1) checking available disk space                                                                                             [##############################################################################] 100%
:: Processing package changes...
(1/1) installing xpadneo-dkms                                                                                                   [##############################################################################] 100%
:: Running post-transaction hooks...
(1/3) Reloading device manager configuration...
(2/3) Arming ConditionNeedsUpdate...
(3/3) Install DKMS modules
==> ERROR: Missing sbin kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing var kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing lost+found kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing usr kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing srv kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing run kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing boot kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing home kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing tmp kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing lib64 kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing mnt kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing etc kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing root kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing dev kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing proc kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing lib kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing sys kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing opt kernel headers for module hid-xpadneo/0.9.5.
==> ERROR: Missing bin kernel headers for module hid-xpadneo/0.9.5.
```

## Steps to Reproduce
On an arch linux install, run `yay -S xpadneo-dkms`

## Expected Behavior
No error messages on install

## System Information

```console
# uname -a
Linux ryan-arch 6.1.11-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 09 Feb 2023 20:06:08 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
4294967295 0
```

## Controller and Bluetooth Information
N/A, issue is about installation

## Additional Context
It appears this issue was also reported by another user on the AUR in January: https://aur.archlinux.org/packages/xpadneo-dkms

rmullin7286, 2023-02-15:
Turns out installing linux-headers beforehand resolved the issue. Closing.

kakra, 2023-02-15:
Thanks. We should add that to the documentation, re-opened.

TiagodePAlves, 2023-03-01:
I've updated the [PKGBUILD on AUR](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=xpadneo-dkms&id=3020cb40913addf011de9e55af86aa80fd7d03e7) with a warning pointing to the [Arch Wiki on DKMS](https://wiki.archlinux.org/title/Dynamic_Kernel_Module_Support) if the Linux headers can't be found for the current running kernel (`uname -r`). The warning will probably trigger if the package is installed or updated after a kernel update, but before the computer reboots, so it isn't fail-proof. Hope it helps to avoid these errors in the future, though.

CaptainCoward, 2023-10-21:
Don't know if this info is even needed. But i just ran into the same issue when installing via pamac (manjaro) installation went through.. so i thought it should work. Controller paired just fine.. but no input was detected afterwards. After installing headers first and then xpadneo it worked.

kakra, 2024-03-04:
I just verified this: Our README already points out that Arch requires `linux-headers` to be installed. It seems to be a general special behavior of distributions that DKMS packages do not automatically depend on the required kernel headers.

Arch documents this in its wiki, and we document the kernel header dependencies. Closing.

closed 2024-03-04 by kakra