atar-axis/xpadneo #410: Xbox Wireless Controller Axis 2 and 5 calibration

mariomaz87, 2023-02-21:
## Version of xpadneo
v0.9-131-g0150346

## Controller Model
- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x ] Xbox Series X|S controller
- [ ] Other:

### Connection mode
- [x ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [x] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact
- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
Axis 2 and Axis 5 (shoulder triggers) are always pressed

## Steps to Reproduce
I added the controller via bluetooth on Linux Mint and I noticed a weird behaviour. Double checking with Jstest I noticed the bug above.

## Expected Behavior
I expected the Axis 2 and 5 to be a 0 while not pressed.

## Screenshots / GIFs / Videos
[image]

## System Information

```console
# uname -a

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth Information

## Additional Context

mariomaz87, 2023-02-21:
Sorry, my bad. I can't reproduce the issue now, everything is working as expected!

closed 2023-02-21 by mariomaz87

kakra, 2023-02-21:
Don't calibrate the Xbox controllers, they calibrate themselves: Just turn them on without pressing any triggers or pushing any sticks. If you feel like something is off, press each trigger two or three times, and rotate the stick axes a few times and it should be properly calibrated.

The old jsdev calibration interface is only there for old devices which are queried via a time-based analogue interface (like old-school game ports). Modern input controllers are almost always self-calibrating and report digital input values properly scaled.