atar-axis/xpadneo #411: paired and connected but still fast blinking

KUKARAF, 2023-03-09:
## Version of xpadneo

clone from github version v0.9-131-g0150346-dirty 

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [x] I can code and I want to help (mainly python)

## Describe the Bug

Even when the device is shown as connected the fast blinking continues and the input is still not recognized. (the controller seems to work out of the box via usb)
```
Pairing successful
[CHG] Device 68:6C:E6:7A:DE:F4 Modalias: usb:v045Ep0B13d0509
[Xbox Wireless Controller]# trust 68:6C:E6:7A:DE:F4
[CHG] Device 68:6C:E6:7A:DE:F4 Trusted: yes
Changing 68:6C:E6:7A:DE:F4 trust succeeded
[Xbox Wireless Controller]# connect 68:6C:E6:7A:DE:F4
Attempting to connect to 68:6C:E6:7A:DE:F4
Connection successful
[DEL] Device 40:75:A4:78:C8:6F 40-75-A4-78-C8-6F
[DEL] Device 48:E5:93:49:76:69 48-E5-93-49-76-69
[DEL] Device 47:F1:BE:B7:E9:35 47-F1-BE-B7-E9-35
[DEL] Device 98:DA:20:01:E9:08 RuiDengDPS
[Xbox Wireless Controller]# info 68:6C:E6:7A:DE:F4
Device 68:6C:E6:7A:DE:F4 (public)
        Name: Xbox Wireless Controller
        Alias: Xbox Wireless Controller
        Appearance: 0x03c4
        Icon: input-gaming
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000001-5f60-4c4f-9c83-a7953298d40d)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v045Ep0B13d0509
        ManufacturerData Key: 0x0006
        ManufacturerData Value:
  03 00 80                                         ...             
        Battery Percentage: 0x32 (50)
[Xbox Wireless Controller]# 

```

When not in pairing mode connecting is also impossible: 

```
Agent registered
[bluetooth]# connect 68:6C:E6:7A:DE:F4
Attempting to connect to 68:6C:E6:7A:DE:F4
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local

```

## Steps to Reproduce

Pair via bluetoothcli 
connect 
be baffled as to why its still blinking but shows as connected

## Expected Behavior

I'd like it to stop blinking and properly connect?

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

Not a file or directory for some reason? 
```
drivers/: directory
rafa@rafa-Surface-Book-2:/sys/module/hid_xpadneo$ cd drivers/
rafa@rafa-Surface-Book-2:/sys/module/hid_xpadneo/drivers$ ls
hid:xpadneo
rafa@rafa-Surface-Book-2:/sys/module/hid_xpadneo/drivers$ file hid\:xpadneo
hid:xpadneo: symbolic link to ../../../bus/hid/drivers/xpadneo
rafa@rafa-Surface-Book-2:/sys/module/hid_xpadneo/drivers$ pwd
/sys/module/hid_xpadneo/drivers
rafa@rafa-Surface-Book-2:/sys/module/hid_xpadneo/drivers$ 

```

```

## Controller and Bluetooth Information

(i assume that you do not mean unpair but dissconnect which it already is)

[xpadneo-dmesg.txt.gz](https://github.com/atar-axis/xpadneo/files/10936398/xpadneo-dmesg.txt.gz)

N/A

N/A

## Additional Context

Newest version of controller purchased today. I did not try updating the software. I am running a [custom  kernel for my surface book 2](https://github.com/linux-surface/linux-surface). It can be janky at times. 

Sometimes it randomly connects and disconnects.

fennecinspace, 2023-03-13:
I have the same issue. I have this controller : https://www.amazon.nl/Xbox-Wireless-Controller-Carbon-Black/dp/B07SDFLVKD/ref=sr_1_6?__mk_nl_NL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=2UN78U4L0IA23&keywords=xbox&qid=1678737256&sprefix=xbox%2Caps%2C97&sr=8-6

kakra, 2023-03-13:
Your controller firmware is known to have stability issues, from your logs:
```
[ 1326.373994] xpadneo 0005:045E:0B13.000C: BLE firmware version 5.09, please upgrade for better stability
```

Please upgrade the controller firmware.

closed 2023-03-14 by kakra