atar-axis/xpadneo #413: unable to load kernel module

MNS26, 2023-03-28:
## Version of xpadneo

 Latest version on github
 
## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other: None

### Connection mode

- [ ] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ x ] I've read the docs and the bug reporting instructions
- [ x ] I've applied the latest firmware update to the controller
- [ x ] I've tried disabling or running without above mentioned software
- [ x ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

upon loading the kernel module it gives a "Unknown symbol in module" error

## Steps to Reproduce

```sh
make -C /lib/modules/6.2.7-200.fc37.x86_64/build/ LD=ld.bfd M="/mnt/data/Containers/dev/xpadneo/hid-xpadneo/src" VERSION="" modules
sudo insmod src/hid-xpadneo.ko
```

## Expected Behavior

No errors when loading

## Screenshots / GIFs / Videos

[image]
[image]
[image]

## System Information

```console
# uname -a
Linux dev.fedora 6.2.7-200.fc37.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Mar 17 16:16:00 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
```
[image]

## Controller and Bluetooth Information

## Additional Context

I tweaked the build method because it runs in a container and it cant access /lib/src/linux
i used the `make` same build command (other than directory) to build a different kernel module that does work

kakra, 2023-03-28:
Does the container have different kernel headers?

Otherwise I'd suggest using the Makefile provided in the `hid-xpadneo` directory and tweaking the `KERNEL_SOURCE_DIRECTORY` variable. Your script is missing building the version file, tho that probably doesn't fix the issue.

Also, `insmod` won't auto-load module dependencies, so maybe another module is missing?

As far as I can see, we have no explicit module dependencies but your kernel may be missing some input related stuff. Does `dmesg` show which symbol is unknown?

MNS26, 2023-03-28:
no it has the same kernel.
I used to use insmod directly on my laptop in the passed.
dmesg doesnt show anything sadly

MNS26, 2023-03-28:
oh... just had another look. i fucked up the signing of it... oops
just fixed it and now it loads

closed 2023-03-28 by MNS26