atar-axis/xpadneo #414: 8BitDo uses Nintendo layout

Shatur, 2023-04-01:
## Version of xpadneo
0.9.5

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: 8BitDo SN30 Pro

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I have XBox Series X gamepad and it works perfectly with xpadneo. However if I try to connect 8BitDo SN30 Pro - it uses Nintendo layout. It doesn't happen without xpadneo. And the documentation says that this feature is [disabled by default](https://github.com/atar-axis/xpadneo#8bitdo-controllers).

## Steps to Reproduce

Just connect the gamepad with xpadneo installed.

## Expected Behavior

It should use XBox layout by default.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux AcerAspire5 6.1.22-1-lts #1 SMP PREEMPT_DYNAMIC Thu, 30 Mar 2023 14:10:04 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/11127998/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/11127999/xpadneo-dmesg.txt)

## Additional Context

kakra, 2023-04-01:
The v0.9 branch of the driver actually defaults to Nintendo mappings, so what you're seeing is expected behavior. You can add `quirks=e4:17:d8:37:fc:7c:0x07` as a modprobe parameter to disable this mapping. v0.10 (still in development) will default to Nintendo mappings off because too many people complained although I think the driver should return what is printed on the buttons.

Unfortunately, the current v0.10 branch has issues with rumble for 8BitDo which I could fix once I get hands on such a controller myself.

Shatur, 2023-04-01:
> The v0.9 branch of the driver actually defaults to Nintendo mappings, so what you're seeing is expected behavior.

Got it, thank you! Then we probably should update the `README.md`.

> You can add quirks=e4:17:d8:37:fc:7c:0x07 as a modprobe parameter to disable this mapping

Thanks, it works.

>  too many people complained although I think the driver should return what is printed on the buttons.

Makes sense and I personally don't mind using Nintendo layout. But this way it works inconsistently when I connect the gamepad over the USB (it uses Xbox layout) because xpadneo driver doesn't handle USB connection.

kakra, 2023-04-01:
The docs should be correct in the v0.9 branch: https://github.com/atar-axis/xpadneo/tree/release/v0.9

If you find it does not, then yes, it should be fixed. But I'm pretty sure you looked at the master branch docs, not at those coming within the download archive for v0.9, maybe?

Shatur, 2023-04-01:
Oh, you are right, my bad, thank you!

A bit off-topic question, but if I want to use Nintendo layout, to make it consistent with the USB mode I should use something like `xboxdrv`?

closed 2023-04-01 by Shatur

kakra, 2023-04-01:
You can ask offtopic questions in our Discord channel. I think it should be possible to use Steam Input to remap the USB mode - at least if you have that installed (and Steam is running). Otherwise, there would also be anti-micro. I'm not sure if xboxdrv would have any benefits here. OTOH, Steam Input should currently not be used with xpadneo (but that may be partially fixed with latest Proton versions, I didn't check yet, I just had a chat with a developer and we exchanged some ideas how to make it compatible).