atar-axis/xpadneo #423: Xbox Wireless Controller is seen as Xbox 360 controller and cannot use share button

elirasza, 2023-05-03:
## Version of xpadneo

0.9.5

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [X] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [X] Steam Proton games are having issues
- [X] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [X] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [X] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [X] I've tried disabling or running without above mentioned software
- [] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [X] I probably didn't figure it all out but it's too early to give up
- [X] I don't know how to ...
- [ ] It's too complicated
- [X] Fantastic work but ...
- [X] I can code and I want to help

## Describe the Bug

My Xbox Series X|S, connected to my computer with a Bluetooth dongle, works perfectly. With your module, I was even able to get back the rumble feature I lost by switching to Linux. However, I still have two problems which I think are linked:
- the controller is shown in `xboxdrv`, Steam and in the last `dmesg` line as an Xbox 360 controller ; thus cannot bind a manual input to the Share button
- the `showkey` utility does not receive any signal from the Share button (but does for all other non-directional buttons)

## Steps to Reproduce

Connect the controller, go in `xboxdrv` or Steam controller settings.
Exit, try to use the Share button in `showkey`.

## Expected Behavior

The module should expose the controller as a Xbox X|S controller.

## Screenshots / GIFs / Videos

[image]

[image]

## System Information

```console
# uname -a
Linux stagcrown 6.2.13-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 26 Apr 2023 20:50:14 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/11384792/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](h
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/11384794/xpadneo-lsusb.txt)
ttps://github.com/atar-axis/xpadneo/files/11384793/xpadneo-dmesg.txt)

## Additional Context

Thank you for your awesome work ! I did not try to listen to the Share button input manually via some C++ code but I think there should be an easier way. I saw in the doc this specific button was bound to the KEY_RECORD keyboard input, but cannot find a way to get to it.

kakra, 2023-05-03:
Current xpadneo is not compatible with Steam Input, you should disable Xbox controller support in Steam. That being said, the documentation is wrong about this, I forgot to change it: It currently maps to a `BTN_TRIGGER_HAPPY1` in the 0.9 branch, and it maps to `KEY_F12` in the development branch. `KEY_RECORD` never worked and could not be seen by applications.

Steam Input doesn't expect to see more than 10 buttons (or 11 buttons if you count the Guide button) from an Xbox controller, thus it won't see the button.

Thanks for pointing it out, I'll update the documentation. Future version will be made compatible with Steam Input but I'm not sure if it lands in v0.10 or v0.11. If you play Steam games exclusively, you might want to use `hid-microsoft` (part of the kernel) with Steam Input instead of xpadneo but you may need to disable rumble to not crash the controller firmware then.

elirasza, 2023-05-03:
Thank you very much for your answer ! Changing the binding to `KEY_F12` is smart as it is the default shortcut for a screenshot in Steam ; and hearing that a full support could exist in the future is heartwarming. Making "short press to screenshot" and "long press to clip" on Linux thanks to it would be really nice.

Moreover, would there be any chance to be able to customize what the share key binds to in the future, in a configuration file for instance ? Or does it have to be hardcoded ?

kakra, 2023-05-03:
> Changing the binding to `KEY_F12` is smart as it is the default shortcut for a screenshot in Steam

Yes, that was the idea behind that change for upcoming v0.10.

> and hearing that a full support could exist in the future is heartwarming.

User-space has matured enough compared to when the "modern" Bluetooth controllers came out that xpadneo should stop doing tricks to bypass SDL while making jsdev happy at the same time. SDL completely detects all models now and supports runtime loading of newer versions into old binary-only software, Proton has gained full support, wine has learned to handle the device, most user-space ships with fixes to work around rumble-introduced firmware crashes... Yes, I think it's time.

> Making "short press to screenshot" and "long press to clip" on Linux thanks to it would be really nice.

This actually was the idea behind using `KEY_RECORD` but except for some very rare exceptions, most user-space completely ignores this event, probably because it needs to be sent via consumer-control devices - which is not a keyboard and thus invisible to most desktop apps.

> Moreover, would there be any chance to be able to customize what the share key binds to in the future, in a configuration file for instance ? Does it have to be hardcoded ? Or is it just not your priority / goal ?

It will be configurable, probably coming with a sysfs interface and a small rust-based tool to remember your controller mappings and load them back when the controller is re-connected, and maybe some GUI to do that stuff and switch presets. But most of this will be handled after v0.10 which is supposed to lay the ground-work for customizations. At that point, xpadneo will probably move over to a completely new project incorporating multiple tools as sub-projects, leaving the atar-axis repository as an archive. As the maintainer of this project (but not owner) I do not have full control over the repository which complicates some things like using CI pipelines, atar-axis stopped developing xpadneo some years ago and handed it over to me.

elirasza, 2023-05-04:
Thank you very much for your work and for your insight ! It is very instructive. It's been ages since I developed in C++ / low-level programming, as I'm more in web nowadays, but maybe I could help in the near future.

For now, I will stick with this repo version and the F12_KEY, as it already suits my needs. Thanks again !