atar-axis/xpadneo #424: Xbox Wireless Controller, via Bluetooth, disconnects/reconnects only after fresh boot.

benkei-kuruma, 2023-05-06:
## Version of xpadneo

0.9.5-2 (installed from [copr](https://copr.fedorainfracloud.org/coprs/atim/xpadneo/))

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [X] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [X] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [X] I've read the docs and the bug reporting instructions
- [X] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Upon fresh boot, after connecting my Xbox Wireless Controller via Bluetooth, the controller will disconnect after about 30 seconds, then immediately reconnect. I'm trying to figure out why it is doing this.

## Steps to Reproduce

1. Boot into OS.
2. Connect Xbox Wireless Controller via Bluetooth.
3. Wait ~30 seconds.

## Expected Behavior

The controller shouldn't disconnect. 
1. I have the same OS and network card installed on three different systems, with the same OS, but only one of them has this problem.
2. On the same system, this problem does not occur in Windows 11.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

```
6.2.14-300.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Mon May  1 00:55:28 UTC 2023 x86_64 GNU/Linux

I'm running Fedora Silverblue 38 if it matters.

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```

## Controller and Bluetooth Information

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/11412393/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/11412394/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/11412395/xpadneo-lsusb.txt)

## Additional Context

If it helps, here are my system specs:

[image]

kakra, 2023-05-07:
Actually, xpadneo is not a Bluetooth driver so it has absolutely no responsibility. The layer above xpadneo is HID, and HID in turn is provided by the Bluetooth layer. So your problem is really with the Bluetooth drivers of the system.

> xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
> 4294967295 0

This needs the xpadneo driver loaded and bound to the HID subsystem. Otherwise it won't work. It might work if you replace `hid:xpadneo` with `*`. But this information is unimportant anyways for your problem.

The 30 seconds interval indicates that the Bluetooth subsystem does not fully establish the connection, eventually it runs into a timeout, then tries to reconnect. If it works this second time, I suspect there might be a race condition within the protocol handshake. This issue should be inspected by the Bluetooth team of the kernel.

```
> HCI Event: Disconnect Complete (0x05) plen 4            #152 [hci0] 64.953775
        Status: Success (0x00)
        Handle: 3585 Address: 3C:FA:06:40:CD:A6 (Microsoft Corporation)
        Reason: Remote User Terminated Connection (0x13)
```
It looks like the firmware is actively disconnecting, it's probably missing parts of the handshake. This is usually due to bluez not being able to complete the full handshake.

```
[  122.902531] xpadneo 0005:045E:0B13.0006: BLE firmware version 5.15
```
This firmware version is supposed to work correctly for most Bluetooth chipsets.

```
= Index Info: 78:2B:46:50:08:E7 (Intel Corp.)                   [hci0] 0.197871
```
But from the reports I'm seeing the Intel chipsets still seem to have problems especially with some devices. From personal experience, I can confirm that Intel chipsets are even problematic with Xbox controllers in Windows - so this may be something that cannot be fixed by Bluetooth drivers or bluez updates alone, we may need to wait for another Xbox controller firmware update.

If there are no other issues like intermittent disconnects, you could try living with this 30 seconds delay until it becomes fixed at some time in the future. Otherwise, I'd recommend to go with a genuine (aka "not China clone") version of the CSR chipset which many known-brand USB dongles use and place it at the top/front of your PC case or above your computer desk, and disable the integrated Intel Bluetooth function.

benkei-kuruma, 2023-05-08:
Thanks a million for all of the work you do. Also thanks for replying to my post with such in-depth info.

I've decided to just upgrade my WiFi/Bluetooth card, since I'd rather not deal with any of these issues at all. I'm going to try an Intel AX210 and see how that goes. I'll report here just to keep the records for anyone that is having similar issues.

closed 2023-05-10 by benkei-kuruma

kakra, 2023-05-10:
So we have one more confirmation that AX201 is working now. Which kernel version are you running? (`uname -a`)

thiccaxe, 2023-07-17:
Can confirm that AX210 is working on `Linux thcwrk 6.3.12-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Jul  6 04:05:18 UTC 2023 x86_64 GNU/Linux` tested on SuperTuxKart.

latest firmware on the "Series" controller.