atar-axis/xpadneo #426: GameSir T4 pro xbox mode rumble does not work sometimes

RushingAlien, 2023-06-09:
## Version of xpadneo
 0.9.r137.g13dd267 (latest commit as of writing)

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: GameSir T4 pro, xbox gamepad mode

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [x] xboxdrv (user-space gamepad driver)
- [x] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [x] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [x] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [x] I don't know how to ... 
  - apply firmware update or if there is any
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
Rumble doesn't work on some games, work on others, what's tested so far  
Native Linux games  
- Celeste : Works
- Hollow Knight : Doesn't work

Wine/Proton games
- Sonic Generations : works via Steam, steaminput disabled and enabled
- Sonic Frontiers : doesn't work via steam, steaminput disabled and enabled
- Hyper Light Drifter : not steam, doesn't work

## Steps to Reproduce

1. Build xpadneo's latest commit as of writing via dkms and install
2. Load the module
3. Connect controller under xbox mode and pair with bluetooth (hold B + home to use xbox mode of the controller)
4. Play games listed above

## Expected Behavior

For Rumble to work in general

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a
Linux komputer-raambm 6.3.6-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 05 Jun 2023 15:12:57 +0000 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0a 15 00 25 01 75 01 95 0a 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 06 95 01 81 03 05 01 09 80 85 02 a1 00 09 85 15 00 25 01  u.................%.
000000b4: 95 01 75 01 81 02 15 00 25 00 75 07 95 01 81 03 c0 05 0f 09  ..u.....%.u.........
000000c8: 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02 15 00 25  !........%.u.......%
000000dc: 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04 91 02 09  .u......p..%du......
000000f0: 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15  Pf..U...&..u........
00000104: 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff  .&..u.....e.U..|..&.
00000118: 00 75 08 95 01 91 02 c0 85 04 05 06 09 20 15 00 26 ff 00 75  .u........... ..&..u
0000012c: 08 95 01 81 02 c0                                            ......
3445511648 1458
```

## Controller and Bluetooth Information

No dongle

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/11709034/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/11709036/xpadneo-dmesg.txt)

## Additional Context

kakra, 2023-06-09:
After connecting the gamepad, please check for a line similar to this in `dmesg` (use the last line you'll find):
```
[ 3386.918550] xpadneo 0005:045E:02E0.000C: input,hidraw4: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on 04:ea:56:56:7b:0f
```

It tells you a hidraw number, `hidraw4` in this case, we'll use it as an example. Now, without disconnecting the gamepad, run `getfacl /dev/hidraw4` (obviously, use the correct number you found).

Post the output. I suspect there might be an issue with games running through SDL2 and trying to use hidraw rumble.

RushingAlien, 2023-06-10:
```
# file: dev/hidraw4
# owner: root
# group: root
user::---
group::---
other::---
```

RushingAlien, 2023-06-10:
Also forgot to add additional context, don't know if it will be helpful 
1. On latest tagged release, the rumble runs nonstop after connecting
2. with `xpad`, controller is recognized as xbox one controller and rumble does not work at all, while with `xpadneo`, controller is recognized as xbox 360 controller with rumble as mentioned in the issue.

kakra, 2023-06-12:
Please try loading the module with quirk parameters (latest master):
```
rmmod hid-xpadneo
modprobe hid-xpadneo quirks=04:ea:56:56:7b:0f+7
```

Does it rumble correctly now? It should rumble the strong and weak motor for around 300ms each.

Termuellinator, 2024-02-08:
I haven't tested much (or compared to original xbox controller) to verify that it rumbles "correctly", but at least with the quirks +7 (though not with the address of the bluetooth controller like you gave in your example but of the gamepad directly as advertised in the line `xpadneo 0005:045E:02E0.0018: Xbox Wireless Controller [a0:5a:5d:4d:24:70] connected
` ) it doesn't rumble a long time on switching it on and instead has at least some rumble in game, so that's a good sign ;) 

Is there any (easy) way to make that quirk permanent until you release a version where it's incorporated?

Strange thing is that i didn't have that long rumble issue when i bought the thing a few months back (haven't used it in a while), but just now. xpadneo-dkms from AUR hasn't updated in the meantime, so i wonder where that change might come from?

kakra, 2024-02-08:
> Is there any (easy) way to make that quirk permanent until you release a version where it's incorporated?

```
# /etc/modprobe.d/xpadneo.conf
options hid-xpadneo quirks=a0:5a:5d:4d:24:70+7
```

> Strange thing is that i didn't have that long rumble issue when i bought the thing a few months back (haven't used it in a while), but just now.

Yeah, I actually fixed a bug that rumble capabilities weren't tested correctly and that showed that some controllers need a quirk applied - which is what you're experiencing now. Sometimes, fixing a bug uncovers quirky behavior. For games that use rumble, it doesn't matter a lot because they actually drive all motors at the same time and send explicit stop commands. But if I wanted to introduce haptics (e.g. for pressing a button), controllers with quirks won't just work correctly, so the driver needs to know about that.

IOW, it looks like this constant rumble is a driver bug but it's actually a firmware bug of your model - the driver was just testing initial rumble like a game and not like a haptic driver.

> xpadneo-dkms from AUR hasn't updated in the meantime, so i wonder where that change might come from?

It should have updated... BTW, the git version has this fix already.

> Edit: oh, and i got some warnings in dmesg (`xpadneo 0005:045E:02E0.0018: buggy firmware detected, please upgrade to the latest version`) but those are probably flukes?

This is bogus for non-Microsoft models: Those just report a firmware version they are pretending to emulate (and they are not actually accurate with that as the above example shows). Most third-party controllers have some subtle differences, one even reverses the rumble-enable bits for triggers and internal rumble motors - which just doesn't matter to games because games just program all motors at once.

This message was added for older genuine models which have Bluetooth bugs which were later fixed my MS, so people can see it before reporting a bug especially because we can do nothing about Bluetooth-related bugs from inside xpadneo. Maybe I should check if we are handling a genuine controller.

Termuellinator, 2024-02-08:
> It should have updated... BTW, the git version has this fix already.

The non-'-git' package hasn't been updated since march last year. That's why i'm surprised the behaviour changed (or maybe i was just lucky with my initial first test?)

I switched over to -git to test your suggestion after finding this issue - but i still need the quirk setting though.

kakra, 2024-02-11:
@Termuellinator Could you try the quirks flag `+3` and `+6` instead of `+7` separately and report how the initial rumble behaves? From the product description, it has only two motors so we don't need to test values for the trigger rumble.

Termuellinator, 2024-02-11:
@kakra Both `+3` and `+6` give prolonged initial rumbles (like without any quirk set) - i haven't tested ingame rumble with those though.

But i think you are confusing the Cyclone T4 **Pro** with the non-Pro. The latter indeed only has two motors, but the Pro version also has trigger rumble (though only supported in x-input mode over bluetooth. But that's the mode we're talking about here after all ^^).

kakra, 2024-02-11:
Okay, this complicates things: `+7` also disables trigger rumble. But both models probably share the same MAC OUI. So we'd need to find a way to tell both models apart. For the moment, I probably just stick with quirk flag `+5` then which leaves the trigger rumble intact. Can you test if `+5` properly works for you?

Also, this is the third controller which needs both flags `+1` (no timing parameters) and `+4` (no individual motor programming) combined, so it completely ignores the haptic parameters of the rumble commands. I probably combine that in the next release.

BTW: `+2` disables handling the trigger rumble motors. You just add these flags together.

Termuellinator, 2024-02-11:
`+5` works perfectly and indeed leaves the trigger rumble intact (always hard to tell if you don't actually know if the game supports trigger rumble or not ^^)

Let me know if you need any detailed info to differentiate the models - but for that we probably would need to have someone with a non-pro model, too...

kakra, 2024-02-11:
> `+5` works perfectly and indeed leaves the trigger rumble intact (always hard to tell if you don't actually know if the game supports trigger rumble or not ^^)

Games running under Linux cannot currently support trigger rumble because the kernel interface is lacking support. SDL with hidraw mode can work around that but that's not compatible with xpadneo, so hidraw is disabled here.

Since you still get trigger rumble, that's because xpadneo synthesizes trigger rumble from the existing input pressure and current rumble of the game. I found it working quite well but you can disable it if you don't like it, or you lower the strength of the effect. Also, as soon as the kernel supports it, xpadneo will automatically disable synthetic trigger rumble if it detects that the game actually supports it - I'm just not exactly sure how to implement that because we don't know when the game was closed.

Also, I'm not sure if Proton with current SDL in hidraw mode already supports XINPUT trigger rumble for games that implement it.

closed 2024-02-12 by kakra