atar-axis/xpadneo #427: question: Xbox Elite Wireless Controller Series 2 support for paddles

alan-barzilay, 2023-06-12:
Hi! Love the project but I have a quick question that I would like to confirm before pulling the trigger on such an expensive controller haha 

I understand that profiles and profile switching are still a bit iffy, but besides that is there any other feature missing for this controller? I really need the ability to remap LB and LT to the elite series 2 right hand paddles due to limited mobility in my left hand, my understanding is that xpadneo currently does not provide button remapping functionality but if it enables the use of the paddles I guess I could do the remapping inside of the game or through steam. Am I correct in my understanding? I never had an elite controller so I'm a bit loss about what functionalities could be missing or even exist haha

I read the docs but wasn' t sure if I was understanding it write. Also, if I buy this controller I would love to help further develop support for it :) (although I never got involved with coding for the kernel or drivers haha)

kakra, 2023-06-12:
Currently you cannot adjust this without hacking the source code. But the infrastructure is there. If you're somewhat into coding, you could hard-code a remapping yourself - it shouldn't be to difficult. We already can see and identify the paddles in the HID reports, and we do report them back to user-space in profile 0 mode.

Some more advanced, I'd like to reverse engineer how MS actually programs remappings into the controller via HID. I think I identified which HID reports do that but I didn't setup a packet capture yet.

Mapping LT to a paddle may be a bit more difficult because one is analog while the other is digital. But converting from a digital input to analog (as you probably want to do) should be easy because the value is either 0 or 1023. I'm currently working on doing it the other way around, too, for mouse emulation (converting analog to digital with some fuzzy logic).

So I'm all in for making this a perfect fit for your use-case.

kakra, 2023-06-12:
There's a proven working example in the source code how to suppress some HID events and synthesize others from it (for the Guide button and emulated profile switching).

alan-barzilay, 2023-06-13:
Makes sense! I didn' t think about the paddles being digital or analog, tbh I had no idea of which one they were haha

I'm sold on the idea, when I finally get the controller I will be back then :) Thank you for the detailed response!

kakra, 2023-06-13:
> when I finally get the controller I will be back then

It's a great piece of hardware: It feels very high-quality in your hands. You can adjust the strength of the joysticks, swap the joystick hats and lengths, remove or swap the paddles, swap the digipad, and adjust the pressing depth of the triggers or make them digital. And it comes in a nice fabric hard case with charging adapter. Some or all of these features may be missing if you buy a non-premium version of it, tho. I'm not sure.

alan-barzilay, 2023-06-15:
sounds great! I am just insecure with the rubberized hand grip, it sounds like something that will deteriorate a lot with time, I wanted something more doritos/greasy hands proof haha

kakra, 2023-06-15:
Mine is about 2 years old (not heavily used, tho) and there are no signs of deterioration, neither for the grips nor for the rest of the housing (which overall feels rubberized but doesn't seem to be, and even if I scratch it with my fingernails the markings can be wiped off with my bare fingers). I think it's not that sort of cheap thin glued rubber foil mixed with plasticizer you find on many other devices. But I think it may look a bit gross over time due to sweaty hands where the surface structure catches some grease. Usually, a soft tooth brush helps cleaning that off, at least for my mouse which has a similar rubbery thumb rest. But I'd recommend not leave it laying around in direct sun, luckily it comes with a nice case protecting from dust and light.

Also, I think it'd be a bit too expensive if it would wear off that fast. But maybe keep some cloth nearby for greasy doritos fingers. Doritos never mix well with PC equipment... ;-)

alan-barzilay, 2023-06-16:
which model of mouse do you have? to get a reference of how the rubber looks/feels like haha

kakra, 2023-06-16:
This one: https://www.logitechg.com/de-de/products/gaming-mice/g502-lightspeed-wireless-gaming-mouse.910-005567.html

WillsterJohnson, 2023-06-16:
If you have access to a Windows machine the Xbox Accessories app lets you write profiles, everything seems to work just fine for me (see above mention).

Not ideal of course, say you decide you want a different mapping, well it's back to the Windows machine and reconfigure. But at least it's possible, and evidently bringing support for this to XPadNeo is being worked on.

kakra, 2023-06-16:
I'm planning to reverse engineer that protocol, so we can have a profile programming app in Linux, too.

ahungry, 2023-12-25:
I just purchased an Xbox Series 2 Elite - Is there a way to make the back paddles report under jstest as events 11, 12, 13, 14 without modifying the source code/recompiling?

Then I could remap them to any key(s) I desire using something like antimicrox - currently they are returning as XABY (events 0 to 4).

I have not used Windows, so no HID remapping - I assume the XABY is the profile 0/default support noted on the readme.

ahungry, 2023-12-25:
I'm going to try editing the source, but how do I compile this?  Does install.sh actually compile?  configure.sh did not produce a makefile for me, nor do I see a top level build file I'm familiar with (Makefile, or cmake stuff etc.)

```
diff --git a/hid-xpadneo/src/hid-xpadneo.c b/hid-xpadneo/src/hid-xpadneo.c
index 8219d03..b99981e 100644
--- a/hid-xpadneo/src/hid-xpadneo.c
+++ b/hid-xpadneo/src/hid-xpadneo.c
@@ -905,10 +905,10 @@ static int xpadneo_event(struct hid_device *hdev, struct hid_field *field,
        if ((usage->type == EV_KEY) && (usage->code == BTN_PADDLES(0))) {
                if (gamepad && xdata->profile == 0) {
                        /* report the paddles individually */
-                       input_report_key(gamepad, BTN_PADDLES(0), value & 1 ? 1 : 0);
-                       input_report_key(gamepad, BTN_PADDLES(1), value & 2 ? 1 : 0);
-                       input_report_key(gamepad, BTN_PADDLES(2), value & 4 ? 1 : 0);
-                       input_report_key(gamepad, BTN_PADDLES(3), value & 8 ? 1 : 0);
+                       input_report_key(gamepad, BTN_PADDLES(11), value & 1 ? 1 : 0);
+                       input_report_key(gamepad, BTN_PADDLES(12), value & 2 ? 1 : 0);
+                       input_report_key(gamepad, BTN_PADDLES(13), value & 4 ? 1 : 0);
+                       input_report_key(gamepad, BTN_PADDLES(14), value & 8 ? 1 : 0);
                }
                goto stop_processing;
```

kakra, 2023-12-26:
@ahungry Yeah, profile 0 is hard-coded to report as buttons ABXY. If you switch to profile 1-3, it will use whatever was programmed into the controller, in case of factory defaults, that's native reporting of the paddles.

The `BTN_PADDLES` macro just shifts the button IDs into the `trigger happy` range. We will shift these a few offsets more soon to be compliant with SDL and in-kernel xpad.

Your patch actually applies to profile 1-3 only, and moves the buttons to `TRIGGER_HAPPY+11` and above. Also, if you want to do that, you also need to ensure the keymap is updated, too, for setting the proper bits so user-space can figure out which bits are actually present.

Compiling can be done with `make -C hid-xpadneo` for systems having the kernel headers/sources installed. Otherwise, dkms uses the Makefile in `hid-xpadneo/src`.