atar-axis/xpadneo #429: [Xbox Elite S2] Mappings configured in Xbox Accessories on Windoes *DO* work on Linux (Pop!_OS 22.04)

WillsterJohnson, 2023-06-16:
Regarding:
https://github.com/atar-axis/xpadneo/blob/cf97733f4f3d372ee80247fe7532de7b15123bc4/docs/README.md?plain=1#L193

Related issues (non-exhaustive list):
- #427 

## Version of xpadneo

`v0.9-142-gcf97733` (https://github.com/atar-axis/xpadneo/commit/cf97733f4f3d372ee80247fe7532de7b15123bc4)

## System Info

(happy to provide more on request, will update this section also)

```sh
$: uname -r
6.0.2-76060002-generic

$: dkms --version
dkms-2.8.7

$: apt-cache policy linux-headers-`uname -r`
linux-headers-6.0.2-76060002-generic:
  Installed: 6.0.2-76060002.202210150739~1666289067~22.04~fe0ce53
  Candidate: 6.0.2-76060002.202210150739~1666289067~22.04~fe0ce53
  Version table:
 *** 6.0.2-76060002.202210150739~1666289067~22.04~fe0ce53 100
        100 /var/lib/dpkg/status

$: bluetoothctl --version
bluetoothctl: 5.64
```

## Controller Model

- [ ] Xbox One S controller
- [x] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [x] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Setup

- acquire Xbox Elite Series 2 controller (mine is pink & blue!)
- acquire hardware installation of Windows 10 (VirtualBox VM didn't work for me - could be user error)
- acquire Microsoft Account, Xbox account
- download Xbox Accessories from Microsoft Store (previous two steps could be replaced with `wine` if this were a simple `.exe` instead, thanks Microsoft)
- connect controller via USB to Windows computer
- Update firmware (latest version as of ~2023-06-16-16:00-UTC+1)
- configure profiles (see below for specifics)
- run `git pull` on existing clone of this repo (or `git clone` if one does not exist, see tagged commit above)
- run `sudo ./uninstall.sh` in clone directory (skip if not previously installed)
- run `sudo ./install.sh` in clone directory (I'm aware of `update.sh` but it never hurts to be thorough)
- disconnect & forget controller with bluetooth manager (I used GUI, `bluetoothctl` or orthers are fine)
- power-on controller & enter pairing mode (button on the top edge)
- connect to Linux computer
- perform test as described below
- disconnect from bluetooth & connect via USB
- repeat perform test as described below

## Test

- launch Steam and enter big picture mode
- go to **Settings** > **Controller** > **Test Device Inputs**
- for each of the four profiles (default, slots 1-3) do the following;
  - press every button, including the Xbox button & paddles, not including the bluetooth pairing mode button
  - ensure that for the given profile the UI shows the correct button being pressed
  - ensure joystick motion is accurate for selected curve
  - ensure triggers activate and bottom out at correct positions (not applicable for my mappings)

## Profiles

Unless otherwise stated;
- all buttons are mapped to their default input for both normal and shift modes
- paddles are not set (as per default config)

<details>
  <summary>Slot 1 (<q>ABXY</q>)</summary>
  <p><strong>Joystick Curves</strong></p>
  <ul>
    <li>Left: Delayed, 50% exaggeration</li>
    <li>Right: Delayed, 50% exaggeration</li>
  </ul>
  <p><strong>Trigger Actuation</strong></p>
  <ul>
    <li>Left: 0%-100%</li>
    <li>Right: 0%-100%</li>
  </ul>
  <p><strong>Normal Mappings</strong></p>
  <ul>
    <li><kbd>P1</kbd> > <kbd>B</kbd></li>
    <li><kbd>P2</kbd> > <kbd>Y</kbd></li>
    <li><kbd>P3</kbd> > <kbd>A</kbd></li>
    <li><kbd>P4</kbd> > <kbd>X</kbd></li>
  </ul>
  <p><strong>Shift Mappings</strong></p>
  <p>Shift keys: N/A</p>
</details>

<details>
  <summary>Slot 2 (<q>ABLBRB</q>)</summary>
  <p><strong>Joystick Curves</strong></p>
  <ul>
    <li>Left: Delayed, 50% exaggeration</li>
    <li>Right: Delayed, 50% exaggeration</li>
  </ul>
  <p><strong>Trigger Actuation</strong></p>
  <ul>
    <li>Left: 0%-100%</li>
    <li>Right: 0%-100%</li>
  </ul>
  <p><strong>Normal Mappings</strong></p>
  <ul>
    <li><kbd>P1</kbd> > <kbd>B</kbd></li>
    <li><kbd>P2</kbd> > <kbd>RB</kbd> (right shoulder)</li>
    <li><kbd>P3</kbd> > <kbd>A</kbd></li>
    <li><kbd>P4</kbd> > <kbd>LB</kbd> (left shoulder)</li>
  </ul>
  <p><strong>Shift Mappings</strong></p>
  <p>Shift keys: N/A</p>
</details>

<details>
  <summary>Slot 3 (<q>Shifted</q>)</summary>
  <p><strong>Joystick Curves</strong></p>
  <ul>
    <li>Left: Delayed, 50% exaggeration</li>
    <li>Right: Delayed, 50% exaggeration</li>
  </ul>
  <p><strong>Trigger Actuation</strong></p>
  <ul>
    <li>Left: 0%-100%</li>
    <li>Right: 0%-100%</li>
  </ul>
  <p><strong>Normal Mappings</strong></p>
  <ul>
    <li><kbd>P1</kbd> > <kbd>B</kbd></li>
    <li><kbd>P3</kbd> > <kbd>A</kbd></li>
  </ul>
  <p><strong>Shift Mappings</strong></p>
  <p>Shift keys: <kbd>P1</kbd>, <kbd>P3</kbd></p>
  <ul>
    <li><kbd>P1</kbd> > <kbd>A</kbd></li>
    <li><kbd>P3</kbd> > <kbd>B</kbd></li>
    <li><kbd>A</kbd> > <kbd>DpadD</kbd></li>
    <li><kbd>B</kbd> > <kbd>DpadR</kbd></li>
    <li><kbd>X</kbd> > <kbd>DpadL</kbd></li>
    <li><kbd>Y</kbd> > <kbd>DpadU</kbd></li>
    <li><kbd>DpadU</kbd> > <kbd>Y</kbd></li>
    <li><kbd>DpadR</kbd> > <kbd>B</kbd></li>
    <li><kbd>DpadD</kbd> > <kbd>A</kbd></li>
    <li><kbd>DpadL</kbd> > <kbd>X</kbd></li>
    <li><kbd>LT</kbd> > <kbd>RT</kbd></li>
    <li><kbd>RT</kbd> > <kbd>LT</kbd></li>
    <li><kbd>LB</kbd> > <kbd>RB</kbd></li>
    <li><kbd>RB</kbd> > <kbd>LB</kbd></li>
    <li><kbd>LSB</kbd> > <kbd>RSB</kbd></li>
    <li><kbd>RSB</kbd> > <kbd>LSB</kbd></li>
    <li>Sticks swap X-axis</li>
  </ul>
</details>

## Results

Every single button on the controller does exactly what was configured in Xbox Accessories, including paddle mapping and shift layer.

[/docs/README.md](https://github.com/atar-axis/xpadneo/blob/cf97733f4f3d372ee80247fe7532de7b15123bc4/docs/README.md)

```diff
-- The other three profiles behave the same way currently, and there is no support for configuring them. This may be
-  different if profiles have been configured in Windows already, that is still untested.
+- The other three profiles behave the same way currently, and there is no support for configuring them. configurations
+  set in Windows with the Xbox Accessories app will work as expected. Tested distros: 
+  - Pop!_OS 22.04 (Ubuntu flavor)
```
## Further Info

Some issues are present when using the Xbox Elite Series 2 on Linux with XPadNeo. If I've found a workaround, I've added it as an item in a sublist below.

- The controller name set in Xbox Accessories does not show up as the device name anywhere in Linux
- The controller stops working periodically despite being enabled
  - Steam shuts down "idle" controllers. In big picture, go to **Settings** > **Controller** > **External Gamepad Settings** > **Idle Gamepad Shutdown Timeout** and set it to "Never"
- The controller UI in Steam shows up as an older model of controller on bluetooth connections (maybe 360 - I'm unsure, see images below. correct model on USB)
  - stop caring; the buttons are still there

[image]
[image]

*Steam controller input test UI shown above. Left: Bluetooth, Right: USB*

To the maintainers & contributors of XPadNeo, if you need someone to test something (XPadNeo development related) on an Xbox Elite S2, don't hesitate to contact me. Either @ me on GitHub, or see my GitHub profile for more ways to get in touch. I'm dreadful at C, and useless with Kernel modules, so unfortunately "owning an Elite S2" is the only way I can contribute.

## Your Controller Is Pink?

You read well - yes!

[image]

I called it "BubblegumButtons" but unfortunately Linux doesn't use it's name, see issues in "Further Info" above.

kakra, 2023-06-16:
I don't think we can get the name of the controller via Bluetooth: There's no report defined in the HID descriptor which would allow that. There seem to be reports defined which I'm guessing can (a) set the RGB value and brightness of the Guide LED, and there's a report defined which can send arbitrary data which I believe is for programming the profiles. But I found out in the last few days that the current firmware version no longer seems to have those reports, I need to reevaluate that.

In USB mode (wired or wireless via dongle), the protocol uses GIP. I've started an early version of a protocol decoder for Wireshark here: https://github.com/xpadneo/gip-dissector. The controller uses SSL handshakes to encrypt some data, so it will be hard to figure out what's going on. But I'm pretty sure there are ways to read the custom name, write RGB (already known how to do that), set profiles (TBD). HID and GIP are quite similar, so if they both have reports for the same purpose, data is sent and received in an almost similar format byte by byte, just the report IDs differ.

Regarding the mappings: Yes, I expected that profiles programmed in Windows will map completely fine in Linux - there's just no way of programming it in Linux.

Regarding the model: Yes, we mimic an original Xbox360 USB controller at the device interface layer of Linux. This has historic reasons: Before SDL2 became de-facto standard in games, games expected the joystick interface to have specific buttons at specific positions, similar for evdev interface. Upcoming SDL2 will have full support for xpadneo mappings including the paddles and special buttons. When it becomes widely deployed, Proton games and Steam Input will automatically work properly finally because older games can load newer SDL2 libs via a dynamic loader and compatibility layer.

So yes, we should probably amend the documentation to explicitly mention that profiles programmed in Windows will work. Feel free to open a pull request for that. No need to mention tested distributions, it finally boils down to SDL2 to detect that properly.

If using a VM with USB passthrough, the following problem occurs: The Xbox Acc app may put the controller into bootloader mode, and then it gets a different USB ID, thus disconnects from the guest. If you use the USB dongle and its wireless connection, you'll bypass that problem and it works.

WillsterJohnson, 2023-06-16:
> I don't think we can get the name of the controller via Bluetooth

Just had a look, even on Windows (hardware, not VM) it has a generic name over Bluetooth. If it's transmitting it's name, Windows & Xbox Accessories don't seem to be using it.

I'm not concerned about the model UI. Oddly enough it shows up as what appears to be the correct model in the Steam Input UI, seems like Steam can't decide what an Xbox controller looks like.

> If using a VM with USB passthrough, the following problem occurs: The Xbox Acc app may put the controller into bootloader mode, and then it gets a different USB ID, thus disconnects from the guest. If you use the USB dongle and its wireless connection, you'll bypass that problem and it works.

That sounds about right, if I had no USB filter or if I used one with default settings, it would connect then immediately disconnect. If I removed some of the filter fields, I'd get a connect/disconnect loop that near enough brought the VM to a standstill. I don't have the dongle unfortunately, but it's good to know that option is available.

I'll sort out that PR shortly.

PR #430 (2023-06-17): docs(#429): Elite S2 Profiles carry over
Fixes #429.

Amend profile switching documentation to state that profiles set to an Xbox Elite Series 2 controller using the Xbox Accessories app will load and operate correctly. Slots without set configurations inherit the default profile's behavior.
closed 2023-06-17 by kakra