atar-axis/xpadneo #431: Missing dkms.conf file, prevents dkms autoinstall from workin (included on kernel updates)

KayJay7, 2023-06-17:
## Version of xpadneo
v0.9-135-g5df12b2\
v0.9-137-g13dd267\
v0.9-143-g5970c4c\
unclear

## Version of other software
dkms: 3.0.11\
fedora: 38\
kernel: 6.3.x

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [x] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

When running `dkms autoinstall` (either manually or at kernel update) the command fails with output:

```
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/hid-xpadneo/v0.9-135-g5df12b2/source/dkms.conf does not exist.
```

the latest version installed is 143, but there are still files from versions 135 and 137 in my system. I guess `autoinstall` stops at the first error. Anyway, running `tree /var/lib/dkms/hid-xpadneo/` I can see the `dkms.conf` file is missing in the other versions as well.

I could try deleting older versions, but I don't think it will make a difference. And I'd rather not manually touch those files, as I don't know if they are also "tracked" in some dkms' database or something.

Autoinstall used to work on some previous version of kernel/dkms/driver. I cannot pinpoint which one is to blame. My best bet is a mixture of dkms and driver. Older versions still on my system did not have the file, but maybe older versions of dkms didn't care.

## Steps to Reproduce

Run `dkms autoinstall`.\
Error.

## Expected Behavior

I expected the module to be automatically installed.

## System Information

```console
# uname -a
Linux thinkpad-alvise 6.3.7-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Jun  9 15:21:11 UTC 2023 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
```

This doesn't look informative

## Additional Context

Installation output:

```
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/hid-xpadneo/v0.9-135-g5df12b2/source/dkms.conf does not exist.
* creating dkms.conf
* adding hid-xpadneo-v0.9-143-g5970c4c folder to /usr/src
* installing module (using DKMS)
Sign command: /lib/modules/6.3.7-200.fc38.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-143-g5970c4c/source -> /usr/src/hid-xpadneo-v0.9-143-g5970c4c

Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.3.7-200.fc38.x86_64 -C /lib/modules/6.3.7-200.fc38.x86_64/build M=/var/lib/dkms/hid-xpadneo/v0.9-143-g5970c4c/build/src VERSION=v0.9-143-g5970c4c modules...
Signing module /var/lib/dkms/hid-xpadneo/v0.9-143-g5970c4c/build/src/hid-xpadneo.ko
Cleaning build area...

hid-xpadneo.ko.xz:
Running module version sanity check.
 - Original module
   - This kernel never originally had a module by this name
 - Installation
   - Installing to /lib/modules/6.3.7-200.fc38.x86_64/extra/

Running the post_install script:
Not disabling ERTM, kernel version doesn't require it...
Installing modalias database...
Installing udev rules...
Reloading udev...
depmod.....
```

It looks like it's creating the file, but there is no file in the folder

kakra, 2023-06-17:
xpadneo does not have a static `dkms.conf`. It will be created by the `install.sh` script. I'm not sure if we could avoid that somehow in the future. I'll leave this opened as a reminder.

KayJay7, 2023-06-17:
The problem I had with autoinstall was solved by removing the old v0.9-135 files, apparently the source directory for the older (with it's `conf` file) was deleted from `/usr/src` but the files in `/var/lib/dkms` weren't. So the file wasn't missing, the whole folder was. I think there's some issue with the cleanup of older versions.

Running `./uninstall.sh` removes the source but not the other files.

Leaving this open as a reminder, should probably rename the issue to the real problem.

kakra, 2023-06-17:
Yeah, I think I've seen this before that older versions weren't cleaned up properly but I'm not sure why. Last time I checked, it seems because dkms changed how versions are listed, and our uninstaller could no longer properly get the list of versions. When that happened, even the fixed version couldn't clean up: The folder structure was somehow broken.

kakra, 2024-12-18:
Please test #517

closed 2024-12-23 by kakra

KayJay7, 2024-12-23:
I apologize, I couldn't test it because my system is currently half broken for other reasons

kakra, 2024-12-23:
I think I can guess the issue: `AUTOINSTALL` was set to `Y` instead of `yes`, so `dkms autoinstall` ignored xpadneo. If you combine that with users that put xpadneo into module autoload, it probably prevented successful booting after a kernel update.

KayJay7, 2024-12-23:
No, I have xpadneo installed from a copr repo that uses akmods instead of dkms. Which still doesn't compile on 6.12 tho, it's a couple weeks outdated.

Anyway, I have an issue with a different dkms driver (a horribly written one) which I shouldn't need anymore, and another with wake suspend. Long story short, I'm running a 6.10 kernel from September, and my dkms install is broken because I tried fixing something.
I'll tend to all this when I have time

kakra, 2024-12-23:
Whatever `akmods` does, if it somehow replicates what `dkms` does, it should obey `AUTOINSTALL` properly now. The other issue with kernel 6.12 has been fixed, too.

Your original post was about dkms and autoinstall, so my guessing was about that being the root cause. If you have other issues still with xpadneo, feel free to open a new bug report.

KayJay7, 2024-12-23:
Thanks, I will.

Just for reference, Akmods is basically redhat flavoured dkms, they are not interchangeable but they can coexist on the same system, often people on Fedora repackage dkms drivers for akmods so it integrates with the package manager. But when I opened the issue I still installed xpadneo manually with dkms.