atar-axis/xpadneo #432: Recommend bluetooth?

kanedait, 2023-06-24:
Can you help me get a "Bluetooth Low Energy" Dongles (or chipset) compatible with the xbox X series controller
I would like it at least 5.0
on Amazon.it
I live in Italy

kakra, 2023-06-24:
I'm using this one: https://www.amazon.it/TP-Link-UB400-Bluetooth-Computer-unterst%C3%BCtzt-Black-Gold/dp/B07NQ5YGDW/ref=sr_1_1?keywords=tp+link+bt400&qid=1687626667&sprefix=tp+link+bt%2Caps%2C265&sr=8-1

But the UB500 model should be fine, too.

kanedait, 2023-06-24:
Thank you
i take the TP-Link UB500
I'll let you know later if it works fine

kanedait, 2023-07-07:
I take the TP-Link UB500.
Sorry for wait, but I had problems.
I use Fedora
with clean install Fedora 38 and 37 I have the same problem:
the controller disconnects and reconnects losing the rumble

I tried Fedora 36 and it works perfectly.
So the dongle is ok, the problem is Fedora.
can you help me?

for  lower the "trigger rumble" 
$ sudo ./configure.sh --rumble-attenuation=50,50
the command is correct or wrong?!

kakra, 2023-07-07:
This sounds like you may run an early firmware of the controller. Could you share the output of `dmesg|grep xpadneo` after connecting the controller?

kanedait, 2023-07-07:
[  415.916025] hid_xpadneo: loading out-of-tree module taints kernel.
[  415.916073] hid_xpadneo: module verification failed: signature and/or required key missing - tainting kernel
[  415.916499] loaded hid-xpadneo v0.9-143-g5970c4c
[  415.945916] xpadneo 0005:045E:0B13.0008: BLE firmware version 5.17
[  415.945921] xpadneo 0005:045E:0B13.0008: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[  415.945924] xpadneo 0005:045E:0B13.0008: working around wrong SDL2 mappings (changed version from 0x00000517 to 0x00001130)
[  415.945927] xpadneo 0005:045E:0B13.0008: report descriptor size: 283 bytes
[  415.945929] xpadneo 0005:045E:0B13.0008: fixing up Rx axis
[  415.945931] xpadneo 0005:045E:0B13.0008: fixing up Ry axis
[  415.945932] xpadneo 0005:045E:0B13.0008: fixing up Z axis
[  415.945933] xpadneo 0005:045E:0B13.0008: fixing up Rz axis
[  415.945935] xpadneo 0005:045E:0B13.0008: fixing up button mapping
[  415.946148] xpadneo 0005:045E:0B13.0008: gamepad detected
[  415.946150] xpadneo 0005:045E:0B13.0008: enabling compliance with Linux Gamepad Specification
[  415.946330] xpadneo 0005:045E:0B13.0008: input,hidraw7: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on e8:48:b8:c8:20:00
[  415.946405] xpadneo 0005:045E:0B13.0008: consumer control added
[  415.946475] xpadneo 0005:045E:0B13.0008: keyboard added
[  415.946478] xpadneo 0005:045E:0B13.0008: controller quirks: 0x00000050
[  415.946480] xpadneo xpadneo_welcome_rumble start
[  416.937009] xpadneo xpadneo_welcome_rumble took 991ms
[  416.937024] xpadneo 0005:045E:0B13.0008: Xbox Wireless Controller [0c:35:26:39:a6:1c] connected
[  416.960320] xpadneo 0005:045E:0B13.0008: reverting to original version (changed version from 0x00001130 to 0x00000517)
[  416.960326] xpadneo 0005:045E:0B13.0008: reverting to original product (changed PID from 0x028E to 0x0B13)
[  417.005951] xpadneo 0005:045E:0B13.0008: BLE firmware version 5.17
[  417.005956] xpadneo 0005:045E:0B13.0008: pretending XB1S Windows wireless mode (changed PID from 0x0B13 to 0x028E)
[  417.005959] xpadneo 0005:045E:0B13.0008: working around wrong SDL2 mappings (changed version from 0x00000517 to 0x00001130)
[  417.005962] xpadneo 0005:045E:0B13.0008: report descriptor size: 283 bytes
[  417.005964] xpadneo 0005:045E:0B13.0008: fixing up Rx axis
[  417.005966] xpadneo 0005:045E:0B13.0008: fixing up Ry axis
[  417.005967] xpadneo 0005:045E:0B13.0008: fixing up Z axis
[  417.005969] xpadneo 0005:045E:0B13.0008: fixing up Rz axis
[  417.005970] xpadneo 0005:045E:0B13.0008: fixing up button mapping
[  417.006186] xpadneo 0005:045E:0B13.0008: gamepad detected
[  417.006187] xpadneo 0005:045E:0B13.0008: enabling compliance with Linux Gamepad Specification
[  417.006347] xpadneo 0005:045E:0B13.0008: input,hidraw7: BLUETOOTH HID v11.30 Gamepad [Xbox Wireless Controller] on e8:48:b8:c8:20:00
[  417.006413] xpadneo 0005:045E:0B13.0008: consumer control added
[  417.006474] xpadneo 0005:045E:0B13.0008: keyboard added
[  417.006478] xpadneo 0005:045E:0B13.0008: controller quirks: 0x00000050
[  417.006480] xpadneo xpadneo_welcome_rumble start
[  417.996994] xpadneo xpadneo_welcome_rumble took 991ms
[  417.997013] xpadneo 0005:045E:0B13.0008: Xbox Wireless Controller [0c:35:26:39:a6:1c] connected

kakra, 2023-07-07:
> `[ 417.005951] xpadneo 0005:045E:0B13.0008: BLE firmware version 5.17`

This looks new enough and should not have any problems. And the timing with ~990ms is very good, too, so we shouldn't fall into rumble crash territory.

Which kernel version are you running? (`uname -a`)

kanedait, 2023-07-07:
6.3.11-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Sun Jul  2 13:17:31 UTC 2023 x86_64 GNU/Linux

kakra, 2023-07-07:
Is there a chance you could test with an LTS kernel like 6.1.x?

kanedait, 2023-07-07:
i tried with 6.2.9-300.fc38.x86_64 same problem
I don't know how to install older kernels

kanedait, 2023-07-08:
I don't know if it can help.
but sometimes add and remove on bluetooth, the controller work better

kanedait, 2023-10-09:
> I use edimax bt-8500 (bt 5.0) I had problems with pairing, but then I updated the firmware of the controller via windows and now everything is fine

I solved it by copy this firmware from Fedora 35 on 38
rtl8761b_config.bin.xz
rtl8761b_fw.bin.xz
rtl8761bu_config.bin.xz
rtl8761bu_fw.bin.xz

kakra, 2024-03-04:
Closing as stale, distributions should fix this, there's nothing xpadneo could do.

closed 2024-03-04 by kakra