atar-axis/xpadneo #433: Incorrect bindings on Gulikit KingKong 2 Pro NS09(axis and buttons detected incorrectly)

CheerfulNag, 2023-07-02:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [x] Other: Gulikit KingKong 2 Pro NS09 

### Connection mode

- [x] Bluetooth connection
- [x] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [x] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [x] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [x] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

I didn't have any problems for more than half a year, but after last month my bindings are wrong on the controller. Problem persistent in all controller modes and both via USB and Bluetooth. Left Trigger moves right stick, some buttons are non existent, some messed up(tested via gamepad-tester, but not only), etc. The exact bindings depend on the controller mode, but always wrong. I spent a lot of time looking into the problem and at the end I think the main issue is - specs of the controller are detected incorrectly. If I use evtest for example(or literally anything else) it says that I have 9 axis and 11 buttons(Xbox controller mode, in other modes different).
This problem is also persistent on another controller of this module(I have 2 of them, completely identical). But no such problem on Windows.

## Steps to Reproduce

Connect controller via USB or Bluetooth, try to use it. Check axis and buttons via evtest/any tester.

## Expected Behavior

I expect that controller bindings won't be insane and it'll work as supposed to.

## Screenshots / GIFs / Videos

## System Information

Manjaro Linux, 6.1 LTS Kernel. But I also had same problem on Fedora, Linux Mint, EndeveourOS(all live cd, except Fedora). 
Laptop, Acer Swift 3 SF314-43, with 5500u.

```console
# uname -a
Linux swift-sf314-43 6.1.31-2-MANJARO #1 SMP PREEMPT_DYNAMIC Sun Jun  4 12:31:46 UTC 2023 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0

```

## Controller and Bluetooth Information

## Additional Context

Last command for file creating returns an empty file for some reason.

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/11931242/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/fil
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/11931246/xpadneo-lsusb.txt)
es/11931243/xpadneo-dmesg.txt)

kakra, 2023-07-02:
I'll test this with my Gulikit Controller. Usually, evtest should be correct always (ignore the extra axis). But if it is wrong in USB mode, too, that's not a problem of xpadneo because xpadneo doesn't support USB.

closed 2023-07-12 by CheerfulNag

kakra, 2023-07-12:
@CheerfulNag What did change that this was closed now?

CheerfulNag, 2023-07-12:
Hey. Well, I still don't know what the issue is with detection(wrong axis and button amount) and why do testers go insane, but in games it works fine(only tried steam though).
So... Not a problem anymore? 
Also considering that this issue is persistent between xpadneo and regular xpad and across distros - I doubt something is wrong with xpadneo specifically :)

kakra, 2023-07-12:
Okay, then it's fine to close.

Take note: Testers that use /dev/js* instead of evdev may be wrong anyways. That interface is hardly used anymore by modern games. And for testers using SDL, there might be a wrong controllerdb entry which messes that up. Officially, SDL removes controllerdb entries for xpadneo-compatible devices (and thus, probably also for xpad-devices) in the next release because it caused problems like yours. Instead, it will have a robust auto-detection for all mappings. For paddles, the next xpadneo release will use xpad-compatible mappings so it'll work with SDL, too. Only the main Guide button will need some extra work and will be fixed later.

Recognizer83, 2023-07-17:
I have the exact same issue. Some games work fine. Some games dont. All games used to work perfect.