atar-axis/xpadneo #437: Xbox series controllers Guide/Share button held to become mouse

FormBurden, 2023-07-27:
## Version of xpadneo
0.9.5

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ X] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Describe your feature request

- [ ] I'd like to see support for a specific model
- [ ] I'd like to see support for a specific hardware feature
- [X ] I'd like to see support for a specific software feature

**Is your feature request related to a problem? Please describe.**
This is more of a question and/or could possible be a feature request. 

**Describe the solution you'd like**
There's a feature that seems to only work while streaming on platforms (Apple TV, Raspberry Pi, etc.) when streaming your Steam game to device (using Steam Remote Play): Where when you hold the guide button  a cursor comes up, and you can use your right joystick and triggers as a mouse. I always thought this feature was neat and could become useful when using the controller in semi-controller compatible games and such. A game that is quite specific but there are others that this feature would be great is: Project Zomboid. The game supports controller support, but it's very basic and quite buggy tho, however, when it comes to mods or interfaces that can be added to the game the controller is useless and you need a mouse. That's where this feature could become useful. There are other games that are similar with limitations like that.

Anyway, this is where the question/feature request comes in: Currently is there a way to do that at all? I tried looking all over the issues and documentation, but couldn't find anything similar. 

Now I know that in 0.10 and 0.11 xpadneo is getting more customization with key-mapping and such. Would this be one of those versions or something?

Also where some other questions come to mind.

I know for the series X|S that holding the guide button will cause the controller to shutdown (after about 10 seconds), and this cannot be prevented. But if the controller sees user-feedback such as using the controller while the guide button is held down, will it still shutdown after the duration? If that's sadly the case, maybe this could be mapped to the "share" button or something instead?

I know the Xbox controllers can be used as a mouse and such when using: "steam input" for the controller, but once you play a game, those features go away and the "Xbox" controls take over. You can re-map the controls to other use-cases using their scheme customizer. But just like in other issues I've seen here, it seems Steam doesn't register the "share" button on the Xbox controller at all. 

So I would assume this would be user-space/host control than Steam controlling the controller when Steam is open. I haven't tried xpadneo while using other programs or the OS itself for navigation. I know it can be done. But my question is from above, like when we can control and change button schemes like in 0.10/11, when these schemes are changed and in effect, when you open Steam, does it get overwritten or are the changes above Steam even when Steam is open? I ask this because I know like when you are in Desktop Mode on the Steam Deck, before Steam opens the controls are being used by the OS/kernel, but when Steam opens, Steam takes over the settings/controls. 

Anyway, sorry for all these questions and partial "request", but I'm trying to figure this out if I can do it now (the guide/share button to become a mouse while held down), or have to wait until 0.10/11 update. 

Thanks for your time!

kakra, 2023-07-27:
Here's an in-progress pull-request where I'm implementing mouse mode for v0.10:
https://github.com/atar-axis/xpadneo/pull/282

Mouse mode can be toggled by holding down the Guide button, then click the Select button and releasing Guide.

I'll close your request once that becomes implemented.

kakra, 2023-07-27:
> I know for the series X|S that holding the guide button will cause the controller to shutdown (after about 10 seconds), and this cannot be prevented. But if the controller sees user-feedback such as [...]

It cannot be prevented - no matter what. That's why xpadneo implements the Guide button as a shift button, and then you'll click one of the other buttons to switch to a different mode and release the Guide button again. That way, you only need to depress it a second or two.

> I know the Xbox controllers can be used as a mouse and such when using: "steam input" for the controller, but once you play a game, those features go away and the "Xbox" controls take over. You can [...]

The feature comes from Steam Input. If you disable it, games will use the controller as-is without Steam overrides. I didn't test if Steam Input overrides xpadneo mouse mode but I think it depends on whether Steam reads HID raw (then it would still see the "real" inputs) or Steam reads evdev (then it sees only what xpadneo wants it to see).

xpadneo currently maps the Share button to F12 so it can be used to take screenshots.

CaptainCoward, 2024-02-09:
> xpadneo currently maps the Share button to F12 so it can be used to take screenshots.

Sorry for  hijacking. But is this still the case. On my Xbox One Series S Pad Share doesn'T seem to do anything on steam.

kakra, 2024-02-13:
> > xpadneo currently maps the Share button to F12 so it can be used to take screenshots.
> 
> Sorry for hijacking. But is this still the case? On my Xbox One Series S Pad Share doesn'T seem to do anything on steam.

This is in the master branch. On v0.9.x, it reports as a gamepad button, and Steam will not make use of it. If you need some help with it, try our Discord community, or find an open bug report about the Share button function itself.

Closing as duplicate of https://github.com/atar-axis/xpadneo/pull/282

closed 2024-02-13 by kakra