atar-axis/xpadneo #439: Xbox Wireless Controller w/ Intel AX210 on Fedora 38: Cannot pair after OS re-install

acerspyro, 2023-09-08:
## Version of xpadneo
v0.9-144-g9b3b696

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

On the Intel AX210, if you pair the device on Linux, then re-install your operating system, the pairing keys are lost on the OS side, but the controller seems to remember the Bluetooth hardware's MAC address and the old pairing key, and refuses to re-pair.

You can temporarily connect the controller without pairing, but it will not work, the controller will simply ignore the fact that it is connected and eventually turn off. Interestingly, you do get the connection "buzz buzz" that xpadneo sends out. The light on the controller goes solid, but it turns off less than a minute later, and doesn't send inputs.

I was able to pass this device to a Windows 10 VM and pair the controller perfectly fine. I then picked up the Bluetooth pairing keys from Windows, implanted them into Linux, and it worked until you unpair the device and try to re-pair it again. It then has the same behaviour as mentioned earlier.

Oddly enough, on my previous installation of F38 (not sure if it was a previous version of xpadneo or the same one), I could unpair and re-pair controllers to my heart's content, without hiccup. It would "just work".

I tried it with the ASUS USB-BT500 dongle, and it seems to pair and unpair just fine.

## Steps to Reproduce
0. Use Intel AX210 chip
1. Install Fedora 38 with xpadneo
2. Pair Xbox Wireless Controller
3. Re-install Fedora 38 with xpadneo
4. Attempt to pair Xbox Wireless controller

## Expected Behavior
It should pair and stay connected.

## Screenshots / GIFs / Videos

## System Information

```console
# uname -a

```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
[veryloud@fedora xpadneo]$ xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
xxd: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor: No such file or directory
4294967295 0
# I can't find a node called "report_descriptor" or "0005:045E:*" below /sys/module/hid_xpadneo/drivers/hid:xpadneo/
[veryloud@fedora xpadneo]$ ls /sys/module/hid_xpadneo/drivers/hid\:xpadneo/
bind  module  new_id  uevent  unbind
```

## Controller and Bluetooth Information

## Additional Context

[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/12555233/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/12555234/xpadneo-lsusb.txt)
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/12555235/xpadneo-btmon.txt)

kakra, 2023-09-08:
> On the Intel AX210, if you pair the device on Linux, then re-install your operating system, the pairing keys are lost on the OS side, but the controller seems to remember the Bluetooth hardware's MAC address and the old pairing key, and refuses to re-pair.

Yes, this is exactly what we previously observed in the early days when connection were still very unreliable. The controller needs to be paired to a different MAC first, and then it will work again.

> You can temporarily connect the controller without pairing, but it will not work, the controller will simply ignore the fact that it is connected and eventually turn off. Interestingly, you do get [...]

For some reason the controller accepts the connection for a few seconds, enough to send the initial rumble greeting, but then disconnects for some reason. As mentioned above, this can be solved by pairing to a different MAC first, e.g. to your smartphone. Then clear out all Bluetooth configuration from the cache in the PC Bluetooth service and start over. It should pair now.

I'm not sure what Windows does differently that it successfully connects. It may be a driver thing, we've not observed that behavior with the very common CSR 4.0 chipset (as long as it is genuine and not a China clone). Modern kernels usually do better. The version of bluez also has some influence on the pairing process itself.

I'll keep this open because you described the issue very precise and well, and will eventually put it into the docs, then close it.

acerspyro, 2023-09-08:
> Yes, this is exactly what we previously observed in the early days when connection were still very unreliable. The controller needs to be paired to a different MAC first, and then it will work again.

I thought I read this affected the AX200, but now that I think about it, they're probably closely related enough that the bug applies to both hardwares.

> I'm not sure what Windows does differently that it successfully connects.

It would be interesting to use WireShark to grab the Bluetooth frames and the USB communication and compare the communication from the OS to the chipset, and from the chipset to the controller, to figure out what Windows drivers are doing differently than Linux drivers, or even compare the Bluetooth frames between the AX200/210 and the ASUS USB-BT500 to find out what one is doing that the other one isn't.

> As mentioned above, this can be solved by pairing to a different MAC first, e.g. to your smartphone. Then clear out all Bluetooth configuration from the cache in the PC Bluetooth service and start over. It should pair now.

Unfortunately, this does not appear to work in my case. I have paired the controller to my phone, cleared the Bluetooth cache relating to its MAC address (it was not paired, so there was no entry directly under the chipset MAC address), restarted the Bluetooth service, and attempted to re-pair it to my computer after unpairing it from my phone, I get the same behaviour. My Bluetooth configuration is attached. [main.conf.txt](https://github.com/atar-axis/xpadneo/files/12563721/main.conf.txt)

> I'll keep this open because you described the issue very precise and well, and will eventually put it into the docs, then close it.

Thank you haha, this issue has been driving me insane for days as I thought I was encountering a Fedora bug, so I had time to get familiar with it :P

rp86, 2023-09-14:
I had a similar problem with Fedora 38 and a motherboard with AX200 chipset. Xbox controllers (1x Elite 2 and 1x Series X) weren't showing up at all in Gnome bluetooth settings, however there were several MAC addressed showing up in bluetoothctl with no device names attached.

Solution for me:
- Paired to a windows laptop to get the correct controller MAC addresses
- Paired with linux machine using bluetoothctl as per wiki
- Controllers now show up in Gnome bluetooth settings, however are not actually connected
- Put controllers back in pairing mode then connect (not re-pair) to them using Gnome bluetooth settings

acerspyro, 2023-09-15:
> I had a similar problem with Fedora 38 and a motherboard with AX200 chipset. Xbox controllers (1x Elite 2 and 1x Series X) weren't showing up at all in Gnome bluetooth settings, however there were [...]

It sounds like your issue is different than mine, both my XBWC show up in GNOME Bluetooth with the name "Xbox Wireless Controller", I just can't connect to them, even using bluetoothctl. The OS refuses to pair, and the controller ignores the fact that it is connected, and just shuts off, as (I assume) its firmware is programmed to not stay connected if it's not paired.

Joomsy, 2023-12-12:
> You can temporarily connect the controller without pairing, but it will not work, the controller will simply ignore the fact that it is connected and eventually turn off. Interestingly, you do get [...]

Not sure how wildly mileage varies with this, but I recently had this problem under Pop_OS with an Intel 3168NGW. The conditions weren't exactly the same, but ultimately, I had the pairing issue you described. The controller would vibrate twice, and then just disconnect when trying to pair it. This happened after upgrading the firmware on my controller (XB1S), and I found that forgetting the device is what led to the problem. Prior to this, I had updated a spare controller as a test, and it paired without issue. I still had my original controller paired, though. After forgetting both, the Bluetooth manager under GNOME wouldn't even list it. To work around this, I found that pressing the sync button for two to three seconds, releasing, and then pressing and holding it again allowed Pop to see and pair with it. It might be worth forgetting the controller, and then trying this. Sounds like you don't really have anything to lose, and worst case, you just end up needing to use a USB cable (I assume you likely already are).

closed 2024-03-04 by kakra