atar-axis/xpadneo #440: Build error on Archlinux kernel version 6.5.4 (Cannot found `generated/autoconf.h`)

pe200012, 2023-09-27:
## Version of xpadneo

v0.9-144-g9b3b696

## Controller Model

- [x] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [x] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [x] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

During building dkms module, `make` complains about not finding `generated/autoconf.h`

## Steps to Reproduce

```
git clone https://github.com/atar-axis/xpadneo.git
cd xpadneo
sudo ./install.sh
```

## Expected Behavior

No issue should happen.

## Screenshots / GIFs / Videos

```
> sudo ./install.sh                                    
* creating dkms.conf
* adding hid-xpadneo-v0.9-144-g9b3b696 folder to /usr/src
* installing module (using DKMS)
Sign command: /usr/lib/modules/6.5.4-zen2-1-zen/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Creating symlink /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/source -> /usr/src/hid-xpadneo-v0.9-144-g9b3b696

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.5.4-zen2-1-zen -C /usr/lib/modules/6.5.4-zen2-1-zen/build M=/var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src VERSION=v0.9-144-g9b3b696 modules...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.5.4-zen2-1-zen (x86_64)
Consult /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/make.log for more information.

> cat /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/make.log
DKMS make.log for hid-xpadneo-v0.9-144-g9b3b696 for kernel 6.5.4-zen2-1-zen (x86_64)
2023年  9月 27日 水曜日 10:56:58 CST
make: ディレクトリ '/usr/lib/modules/6.5.4-zen2-1-zen/build'　に入ります
cp /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/hid-xpadneo.c /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo.c
  CC [M]  /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo/core.o
  CC [M]  /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo/consumer.o
  CC [M]  /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo/keyboard.o
  CC [M]  /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo.o
次のファイルから読み込み:  <コマンドライン>:
././include/linux/kconfig.h:5:10: 致命的エラー: generated/autoconf.h: そのようなファイルやディレクトリはありません
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
次のファイルから読み込み:  <コマンドライン>:
././include/linux/kconfig.h:5:10: 致命的エラー: generated/autoconf.h: そのようなファイルやディレクトリはありません
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
次のファイルから読み込み:  <コマンドライン>:
././include/linux/kconfig.h:5:10: 致命的エラー: generated/autoconf.h: そのようなファイルやディレクトリはありません
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
コンパイルを停止しました。
コンパイルを停止しました。
コンパイルを停止しました。
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo/consumer.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo/keyboard.o] エラー 1
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo/core.o] エラー 1
次のファイルから読み込み:  <コマンドライン>:
././include/linux/kconfig.h:5:10: 致命的エラー: generated/autoconf.h: そのようなファイルやディレクトリはありません
    5 | #include <generated/autoconf.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~
コンパイルを停止しました。
make[2]: *** [scripts/Makefile.build:243: /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src/xpadneo.o] エラー 1
make[1]: *** [/usr/lib/modules/6.5.4-zen2-1-zen/build/Makefile:2032: /var/lib/dkms/hid-xpadneo/v0.9-144-g9b3b696/build/src] エラー 2
make: *** [Makefile:234: __sub-make] エラー 2
make: ディレクトリ '/usr/lib/modules/6.5.4-zen2-1-zen/build' から出ます
```

## System Information

[image]

```console
# uname -a
Linux pecomputer 6.5.4-zen2-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Thu, 21 Sep 2023 12:54:31 +0000 x86_64 GNU/Linux
```

## Controller and Bluetooth Information

## Additional Context

kakra, 2023-09-27:
This looks like a problem outside of xpadneo: `include/kconfig.h` includes `generated/autoconfig.h` which does not exist. Do you have the matching kernel include sources properly installed?

pe200012, 2023-09-27:
@kakra You are correct. I somehow messed up my kernel source(probably because of some previously installed dkms modules).
Closing.

closed 2023-09-27 by pe200012