atar-axis/xpadneo #442: Series X|S controller not listed in bluetooth devices Ubuntu 22.04.3

DoobleD, 2023-10-27:
Hi folks,

Thanks for providing this. 

My series X|S controller is not listed in the available bluetooth devices on Ubuntu 22.04.3 using either the system bluetooth GUI or `bluetoothctl` from the command line as instructed [here](https://github.com/atar-axis/xpadneo#connection). 

My bluetooth supports BLE, and pairing with the controller works fine from my Windows partition on the same computer.

```
$ btmgmt info
Index list with 1 item
hci0:	Primary controller
	addr C8:94:02:C1:67:1E version 11 manufacturer 70 class 0x10010c
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech 
	current settings: powered connectable discoverable bondable ssp br/edr **le** secure-conn 
	name <obfuscated>
	short name 
hci0:	Configuration options
	supported options: public-address 
	missing options: 
```

Any idea?

kakra, 2023-10-27:
Try updating the Xbox controller to latest firmware in Windows. This works by USB or with the original Xbox radio dongle, you cannot update the firmware via Bluetooth connection (there's no such support in the protocol).

After updating the firmware, if it still does not work, try pairing to a smartphone first. Also, try double clicking or short holding (1s) the connect button (only works with firmware 5.x) to switch it between Xbox dongle and Bluetooth mode. I'm currently not sure which connect button press switches to which mode.

Also, look if maybe your Bluetooth chipset needs a firmware package to be installed. Usually, most firmware is provided in the linux-firmware package but sometimes there are extra packages for specific hardware.

DoobleD, 2023-10-28:
Thank you @kakra!

I had upgraded the controller to the latest firmware already from Windows with USB, and I have also paired with my Android smartphone without issue.

I did enable pairing mode by holding the connect button for ~1 sec (the controller light starts blinking fast), is that what you mean? I'm not sure I get the difference between dongle and bluetooth mode. I don't have any external dongle, just the controller and a bluetooth chipset in my computer.

I tried looking around for a specific Linux driver for my chipset but had no much luck with that.

deathxxx123, 2024-05-03:
The controller is no more working after lastest firmwre. Im thinking for xbox dongle and end this kind of problems for once.

closed 2024-12-21 by github-actions[bot]