atar-axis/xpadneo #444: Vibration / rumble stops working after reconnecting the controller [Xbox Series S]

arnxxau, 2023-11-13:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [x] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [ ] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [x] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

Vibration / rumble stops working when reconnecting the controller. For example if it turns off because of inactivity and then you power it on again

## Steps to Reproduce

- Open a game with the gamepad already connected
- check that vibration / rumble is working
- turn off the controller
- turn on the controller

## Expected Behavior

Vibration / rumble should start working after restart of the game

## Observations / workaround
I found out that I could re-enable the vibration / rumble by closing and opening the game again.
I also observed that the welcome rumble start plays correctly even after the restart of the controller

## System Information

```console
# uname -a
6.6.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 08 Nov 2023 16:05:38 +0000 x86_64 GNU/Linux
```
I also tried with the latest Nobara Linux Kernel (6.59 fsync 201)

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363

```

## Controller and Bluetooth Information

Bus 003 Device 004: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/13340672/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/13340673/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/13340674/xpadneo-lsusb.txt)

kakra, 2023-11-13:
This is a known issue due to how xpadneo offers rumble support to games, and games using the wrong SDL API to query rumble support. We can probably work around this from the xpadneo side, but the SDL project lately stated that games really should use a different API to query for rumble support.

Until then, you should be able to work around it by disabling the welcome rumble in the driver. This tightens the time frame between device detection and offering the rumble feature which will probably work around that race condition in SDL - at least most of the time.

arnxxau, 2023-11-13:
Thank you :)! Is there a way to disable the welcome rumble without commenting the function in the code? I can't seem to find any option in the configure.sh script

kakra, 2023-11-13:
```text
# modinfo hid-xpadneo
[...]
parm:           ff_connect_notify:(bool) Connection notification using force feedback. 1: enable, 0: disable. (bool)
[...]
```

;-)

So you actually add that parameter to the module load options.

arnxxau, 2023-11-13:
Working! Thank you again !  : )

kakra, 2023-11-13:
Thanks for confirming. Please leave this opened so I can implement a fix.

orbea, 2024-01-18:
> but the SDL project lately stated that games really should use a different API to query for rumble support.

The newer API is also not applicable for some use cases, for example some emulators emulate controllers that have wildly different layouts than SDL2 expects for modern games. Was it ever confirmed that the newer API even fixes this?

RobotRoss, 2024-02-17:
Just wanted to say that I was also having this issue and that disabling ff_connect_notify also fixed this for me on a Gulikit KingKong2 Pro controller.

Disabled using `echo 0 | sudo tee /sys/module/hid_xpadneo/parameters/ff_connect_notify` (found in the xpadneo docs here - https://github.com/atar-axis/xpadneo/blob/master/docs/CONFIGURATION.md )

kakra, 2024-02-19:
@RobotRoss I closed your other report as duplicate then. You can apply the setting permanently:
```
# /etc/modprobe.d/xpadneo.conf
options hid-xpadneo ff_connect_notify=0
```

RobotRoss, 2024-02-19:
> @RobotRoss I closed your other report as duplicate then. You can apply the setting permanently:
> 
> ```
> # /etc/modprobe.d/xpadneo.conf
> options hid-xpadneo ff_connect_notify=0
> ```
> 
> The next version will disable that by default, or switch to a different method.

That issue should be reopened, seems to be a different problem. See comment on the other issue.

kakra, 2026-02-10:
> I'd like to add that setting ff_connect_notify=0 does not solve the problem; it introduces a new problem. As comments above, disabling this does re-enable rumbling, but the rumbling is not functioning properly.

Well, then by definition, it still fixes the problem. Also, it doesn't introduce a new problem because without this parameter, your controller won't rumble at all. As outlined above, the problem is that we expose rumble support only a few hundred milliseconds after the controller appears on the bug. A proper fix will be deployed during v0.11 development.

Your issue is probably very different. The controller has exactly two frequencies: a heavy strong motor with slow vibrations, and a weaker fast motor with fast vibrations. If you are seeing just one frequency, the rumble commands do not properly operate, which is a completely different issue. Please open a new report for it, you may reference this report.