atar-axis/xpadneo #449: New information about bluetooth pair issues on Raspberry Pi

lukakama, 2024-01-03:
Hi,
I would like to share that I found a way to pass the initial controller pairing issue at least some type of controllers on a Raspberry  Pi 3b+.

Background: I have two wireless controllers, one old Xbox One S (USB type A and missing share button) and a newer Xbox Series X one (with share button and USB type C connector). Both controllers has been updated with latest firmwares. The Raspberry is a 3b+ freshly installed with "bullseye" Lite OS on 12-2023, using the built-in adapter with just "disable_ertm" enabled. The Xbox Series X controller paired without much issues (it struggled to connect the first time, but after a bunch of time it paired successfully), but the Xbox One S one persistently refused to pair (bluetoothctl was reporting a connected status, but he controller was still in pairing mode with the led blinking).

To successfully pair the Xbox One S controller I had to:
1. Set the controller into pairing mode
2. Perform che standard pairing procedure on the Raspberry using "bluetoothctl"
3. Once connected with the controller still in pairing mode, disconnect the controller issuing a "disconnect XX:XX:XX:XX:XX" command on "bluetoothctl"
4. Open the "/var/lib/bluetooth/[ADAPTER_MAC]/[XBOX_CONTROLLER_MAC]/info" file (device metadata file of Bluez)
5. Add the following section to the file (copied from the "info" file of the Xbox Series X controller): 
```
[ConnectionParameters]
MinInterval=6
MaxInterval=6
Latency=0
Timeout=300
```
6. Save and close the file
7. Re-connect to the controller using "bluetoothctl" (this time the controller successfully moved to the "connected" mode exiting the "pairing" mode without issues) and completing the pairing procedure.

Probably it would works also creating the file manually before the first pairing attempt, but I don't know if the format would be different on different versions of Bluez (I didn't found much info about these files).

Maybe this could be added to the Troubleshooting section. I hope that this could be useful to someone.

kakra, 2024-01-03:
Interesting that copying the interval and latency settings fixes things. This still looks like bluez doesn't propagate these attributes properly to the connection. FWIW, modifying these parameters has been part of the xpadneo documentation but we use a bit more relaxed values so Bluetooth mice and other peripherals would still work correctly.

With fairly recent versions of the kernel (everything from the last 24 months) you should not use disable_ertm any longer because it can cause sticky button inputs or disconnects with rumble.

I'll leave this open to review it and merge into documentation. Please don't close.

kakra, 2024-03-04:
Deferring to v0.11 to investigate RPi compatibility.

d-cglt, 2026-10-12:
Confirmed working beyond Raspberry Pi: Fedora 44 with kernel 7.2.9, BlueZ 5.87, a Realtek Bluetooth adapter (USB ID `0bda:4853`), xpadneo `v0.10-pre-259-gfc1b13a`, and an Xbox Series S|X controller reporting firmware 5.09.

Before the fix, BlueZ reported paired and connected and created a gamepad device, but the controller kept blinking and sent no input. USB input worked.

Applied these settings to `/etc/bluetooth/main.conf`:

```ini
[General]
Privacy = device
JustWorksRepairing = confirm

[LE]
MinConnectionInterval = 7
MaxConnectionInterval = 9
ConnectionLatency = 0
ConnectionSupervisionTimeout = 200
```

Restarted Bluetooth, removed the controller's failed pairing, then paired and trusted it again. Verified button and stick input over Bluetooth with USB unplugged, plus automatic reconnection with a steady light after switching the controller off and on.

No firmware update or per-device connection-parameter edits were needed. In particular, `JustWorksRepairing = confirm` worked here rather than `always`. These settings were applied together, so I haven't isolated which ones are necessary. Long-term stability remains untested.

Thanks for documenting this workaround!

closed 2026-10-25 by kakra