atar-axis/xpadneo #450: Xbox Elite 2 Controller Disconnecting on Arch Linux

Pejno, 2024-01-07:
## Version of xpadneo
0.9.r144.g9b3b696 (same thing happens on 0.9.5 from aur)

## Controller Model

- [ ] Xbox One S controller
- [ *] Xbox Elite 2 controller
- [ ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ *] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ *] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [* ] I've applied the latest firmware update to the controller
- [* ] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug

The Xbox Elite 2 controller randomly disconnects and then reconnects after a few seconds. The disconnections occur regardless of whether the controller is idle or in use. The issue can manifest within seconds or minutes after connecting. The problem does not occur when using Windows or when connecting via USB.

## Steps to Reproduce
Connect the controller
Launch a game and play or let it sit

## Expected Behavior
The controller will disconnect at random intervals and then reconnect

## System Information

```console
# uname -a
Linux archlinux 6.6.10-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 a1 01 0a 85 00  ...%.u..............
000000c8: 15 00 26 ff 00 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95  ..&....u........&...
000000dc: 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 0a 81 00 15 00  .u.....%...u........
000000f0: 26 ff 00 95 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 c0  &....u.....%...u....
00000104: 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02  ...!........%.u.....
00000118: 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04  ..%.u......p..%du...
0000012c: 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02  ...Pf..U...&..u.....
00000140: 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15  ....&..u.....e.U..|.
00000154: 00 26 ff 00 75 08 95 01 91 02 c0 05 0c 09 01 85 0c a1 01 0a  .&..u...............
00000168: 9e 00 15 00 26 ff 00 95 01 75 08 81 02 0a a1 00 15 00 26 ff  ....&....u........&.
0000017c: 00 95 01 75 08 81 02 0a a2 00 15 00 26 ff 00 95 01 75 08 81  ...u........&....u..
00000190: 02 0a a3 00 15 00 26 ff 00 95 01 75 08 81 02 c0 c0 05 01 09  ......&....u........
000001a4: 06 a1 01 85 05 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81  .........)...%.u....
000001b8: 02 95 01 75 08 81 03 95 06 75 08 15 00 25 65 05 07 19 00 29  ...u.....u...%e....)
000001cc: 65 81 00 c0                                                  e...
2376472410 2192

```

## Controller and Bluetooth Information

ID 2357:0604 TP-Link TP-Link UB500 Adapter

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/13854480/xpadneo-btmon.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/13854481/xpadneo-lsusb.txt)

## Aditional note
While I've fully embraced Linux for my everyday computing, I still consider myself a Linux 'noob'. I'm more than willing to share any additional information if needed and am open to following instructions for any potential solutions. 
Thanks in advance for your assistance!

kakra, 2024-01-07:
This is a problem we cannot really control from the driver. There are multiple causes for this:

- Bluetooth driver problem or chipset incompatibility: the UB500 falls into this category, UB400 may work better
- too fast rumble commands from user-space bypassing the driver crashes the firmware: if SDL or other hidraw clients use direct rumble commands, we cannot control this, disable Steam Input and/or access to hidraw
- xpadneo really is a HID driver, it doesn't see Bluetooth at all, leaving us with no chance to control anything Bluetooth-related

Depending on the firmware version, the firmware can crash on rumble:

- via USB (includes the Xbox ACC dongle) if intervals fall below 10ms (Windows drivers seem to force intervals of at least 10ms)
- via Bluetooth on older firmware if intervals fall below 20ms (all modern SDL versions should work around this)
- via Bluetooth on newer firmware if intervals fall below 50ms (recent SDL versions should work around this)
- the Linux kernel uses 20ms intervals but doesn't prevent the issues because 20ms is per effect, not per command
- xpadneo currently forces intervals of at least 50ms, this causes frequency aliasing with ff_memless resulting in uneven rumble effects (xpadneo v0.10 is planned to work around this)

You can check if rumble is the problem by turning rumble off: set attenuation to 100 in the module options. If games still rumble, SDL/Steam Input use hidraw: Try turning that off. If rumble is completely gone now, you can turn driver rumble back on and see if the problem persists.

But I believe you should rather look into Bluetooth issues:

- if possible, try a different chipset, e.g. UB400 (CSR-based) instead of UB500 (Realtek-based)
- look into bluez latency and interval settings: https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md

Pejno, 2024-01-08:
Tried the tweaks you suggested. 

Changed the attenuation to 100 in the /etc/modprobe.d/xpadneo.conf
Added the 'SDL_USE_HIDAPI=0' in the /etc/environment

Quick rundown:

1. Disconnected after about a minute.
2. Reconnected, and the Steam Controller Tester's rumble option vanished, played nice for 5-10 minutes (the longest uninterrupted session so far).
3. Disconnected again, rumble option came back.
4. Repeat.

Thanks for the quick response and guidance. Apologies if I misunderstood any steps.
Will try to get a different adapter soon, and I'll report back if anything changes.

closed 2024-04-03 by github-actions[bot]