atar-axis/xpadneo #452: Xbox one S controller has R input on it's own

StoffelCPR, 2024-01-21:
## Version of xpadneo

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [x] Xbox Series X|S controller ( XB1S )
- [ ] Other:

### Connection mode

- [x] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [x] OpenRGB (may mess up mappings and rumble stability)
- [ ] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [x] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [x] Steam Proton games are having issues
- [x] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [x] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [x] I don't use native games or did not try
- [x] Other software is having issues (describe software and issues below)
- [x] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [x] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [x] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [x] I've read the docs and the bug reporting instructions
- [x] I've applied the latest firmware update to the controller
- [x] I've tried disabling or running without above mentioned software
- [ ] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [x] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
So on two different systems I had the same problem.. Generally everything works fine, except, most times my right stick input goes up ( negative values in jstest, evtest ) so menus scroll up and my camera `looks` up on it's own when unused ( ONLY when unused, while using the right stick everything works fine and the random scroll up stops, starts after a few seconds without right stick input )

I've tested in several games ( Monster Hunter World, Jedi: Fallen order, Jedi: Survivor, Hogwarts Legacy )
I've tested on my GFs windows Xbox Accessories and it also show a right stick input w/o me inputting
I've tested in jstest
I've tested in evtest

All show the same behaviour 9 Axis ( which is correct apparently? ) all working fine.. And after using the right stick input ( Axis 5 ) and letting go of it it's `neutral` ( value: 0 ) and after a short moment the axis' value goes 'down' ( which means the game inputs an up command ( value of axis5: -32k ).

I've tried having the controller calibrate itself and doing inputs but that doesn't fix it.. I've searched several forums, reddits and issue here.

## Steps to Reproduce
unsure as I haven't figured out what's causing the input.. For me to reproduce I just need to connect the controller and input onto the axis. It'll move on it's own afterwards

## Expected Behavior
Well.. Nothing, I guess? It'd be easier to play if I wouldn't have to constantly worry about keeping constant stick input as to prevent the camera moving upwards on it's own.

## Screenshots / GIFs / Videos

FOR BOTH: -> I only input `down` on axis five ( right stick ) and then let go. The rest happens without my inputting anything.

[JSTest input values on my personal google drive](https://drive.google.com/file/d/1nmRy72MOQUejfE0AuL7sBFMPYfxaFZdW/view?usp=sharing)

[Hogwarts Legacy settings menu scrolling on my personal google drive](https://drive.google.com/file/d/1CTUfErhQgl99rD1h4Y15kgZ4JCgRDAuT/view?usp=sharing)

P.S. It'd be nice to get a notice if these videos aren't needed anymore so I can take the links out for privacy reasons

## System Information

```console
# uname -a
Linux gs-ravenclaw-beta 6.7.0-204.fsync.fc39.x86_64 #1 SMP PREEMPT_DYNAMIC TKG Wed Jan 17 09:24:55 UTC 2024 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a 24 02 15 00 25 01 95 01 75 01 81  u........$...%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 85 02 a1 01 05  ...%.u..............
000000c8: 0c 0a 23 02 15 00 25 01 95 01 75 01 81 02 15 00 25 00 75 07  ..#...%...u.....%.u.
000000dc: 95 01 81 03 c0 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75  ........!........%.u
000000f0: 04 95 01 91 02 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25  .......%.u......p..%
00000104: 64 75 08 95 04 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75  du......Pf..U...&..u
00000118: 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00  .........&..u.....e.
0000012c: 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91 02 c0 05 06 09 20  U..|..&..u......... 
00000140: 85 04 15 00 26 ff 00 75 08 95 01 81 02 c0                    ....&..u......
```

## Controller and Bluetooth Information

( huge consistent output of axis' after input the upwards event spams )

```console
Bus 001 Device 004: ID 8087:0029 Intel Corp. AX200 Bluetooth
```

[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/14003037/xpadneo-lsusb.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14003038/xpadneo-dmesg.txt)
[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14003039/xpadneo-btmon.txt)

## Additional Context

After writing this I do realize that the problem might not be XPadNeo related -> Kept sending the issue because maybe you can tell me where exactly the issue is and keep it closed as knowledge base in case someone searches for this on google.

kakra, 2024-01-22:
I can see a similar issue for some games while my HOTAS is connected because SDL misidentifies it as a gamepad and thus feeds XINPUT data to such games. But since you're seeing it in jstest, I think your issue is different.

xpadneo offers an additional axis which combines both triggers into a rolling axis for simulators. I probably remove that with the next release and maybe that is interfering?

If you have additional game input devices, you may need to exclude them from SDL for gamepad-focused games:
```
SDL_GAMECONTROLLER_IGNORE_DEVICES=0x044F/0xB10A,0x044F/0xB687,0x231D/0x0200,0x231D/0x0201,0x26CE/0x01A2
```
would exclude my VKB HOTAS, Thrustmaster HOTAS and my mainboard RGB controller because that is detected as a HID input device identified as a gamepad. You can use `lsusb` to find the device IDs. Maybe your G733, mouse to Razer device interferes. Your MSI Mystic light and Cooler Master RGB are also possible candidates. All of these should not introduce phantom input into `jstest` or `evtest`, tho.

Also, please upgrade your controller firmware, it is known to cause bugs with sticky inputs or rumble crashes:
```
[ 1022.600747] xpadneo 0005:045E:02FD.000B: buggy firmware detected, please upgrade to the latest version
```

Also check if turning Steam Input off makes any difference. It is active as soon as the Steam Client is running.

kakra, 2024-01-22:
> FOR BOTH: -> I only input `down` on axis five ( right stick ) and then let go. The rest happens without my inputting anything.
> 
> [Google Drive Links]
> 
> P.S. It'd be nice to get a notice if these videos aren't needed anymore so I can take the links out for privacy reasons

I've played Hogwards Legacy myself with Steam Input disabled, it runs fine for me with xpadneo. The 9th axis is a combination of both trigger axes and should idle at 0 if no trigger is pressed. It won't interfere with games because wine only looks at the first 8 axes.

StoffelCPR, 2024-01-22:
## System Information

```console
# uname -a
Linux ws-cpelzer-ravenclaw-alpha 6.6.6-76060606-generic #202312111032~1702306143~22.04~d28ffec SMP PREEMPT_DYNAMIC Mon D x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0f 09 21 85 03 a1 02 09  ...%.u........!.....
000000c8: 97 15 00 25 01 75 04 95 01 91 02 15 00 25 00 75 04 95 01 91  ...%.u.......%.u....
000000dc: 03 09 70 15 00 25 64 75 08 95 04 91 02 09 50 66 01 10 55 0e  ..p..%du......Pf..U.
000000f0: 15 00 26 ff 00 75 08 95 01 91 02 09 a7 15 00 26 ff 00 75 08  ..&..u.........&..u.
00000104: 95 01 91 02 65 00 55 00 09 7c 15 00 26 ff 00 75 08 95 01 91  ....e.U..|..&..u....
00000118: 02 c0 c0                                                     ...
2986910699 1363
```

## Controller and Bluetooth Information

```console
Bus 001 Device 007: ID 8087:0029 Intel Corp. AX200 Bluetooth
```

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14015477/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14015478/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/14015479/xpadneo-lsusb.txt)

## Additional Context

This is on my PopOS system ( I tried nobara yesterday while deleting the windows disk and it had the same error and I got frustrated and opened the issue; PopOS is my main system currently )

The controller is updated to the latest firmware. Steam off and it still show the input.

Is it possible that the controller just has some kind of issue? Any issues known with older Xbox one S controller versions?

Would it be advisable to buy a new controller?

kakra, 2024-01-22:
You're using the latest in-development version. Does this happen also with v0.9.5 of xpadneo?

StoffelCPR, 2024-01-26:
Installed v0.9.5:

```console
# uname -a
Linux ws-cpelzer-ravenclaw-alpha 6.6.10-76060610-generic #202401051437~1704728131~22.04~24d69e2 SMP PREEMPT_DYNAMIC Mon J x86_64 x86_64 x86_64 GNU/Linux
```

```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

Dunno why this happens?

```console
zsh: no matches found: /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor
4294967295 0
```

[xpadneo-btmon.txt](https://github.com/atar-axis/xpadneo/files/14068367/xpadneo-btmon.txt)
[xpadneo-dmesg.txt](https://github.com/atar-axis/xpadneo/files/14068369/xpadneo-dmesg.txt)
[xpadneo-lsusb.txt](https://github.com/atar-axis/xpadneo/files/14068370/xpadneo-lsusb.txt)

###

EVTEST can't find the device:

```console
No device specified, trying to scan all of /dev/input/event*                                                                                                                                                                                                  
Available devices:                                                                                                                                                                                                                                            
/dev/input/event0:      Power Button                                                                                                                                                                                                                          
/dev/input/event1:      Power Button                                                                                                                                                                                                                          
/dev/input/event2:      Logitech G733 Gaming Headset Consumer Control                                                                                                                                                                                         
/dev/input/event3:      Logitech G733 Gaming Headset                                                                                                                                                                                                          
/dev/input/event4:      Logitech G733 Gaming Headset                                                                                                                                                                                                          
/dev/input/event5:      Razer Razer Huntsman Elite                                                                                                                                                                                                            
/dev/input/event6:      Razer Razer Huntsman Elite Keyboard                                                                                                                                                                                                   
/dev/input/event7:      Razer Razer Huntsman Elite                                                                             
/dev/input/event8:      Razer Razer Huntsman Elite                                                                             
/dev/input/event9:      MOSART Semi. Trust Wireless Mouse                                                                      
/dev/input/event10:     MOSART Semi. Trust Wireless Mouse                                                                      
/dev/input/event11:     MSI MYSTIC LIGHT                                                                                       
/dev/input/event12:     Cooler Master Technology Inc. AMD SR4 lamplight Control                 
/dev/input/event13:     Cooler Master Technology Inc. AMD SR4 lamplight Control System Control                                 
/dev/input/event14:     Cooler Master Technology Inc. AMD SR4 lamplight Control Consumer Control                               
/dev/input/event15:     HDA NVidia HDMI/DP,pcm=3                                                                               
/dev/input/event16:     HDA NVidia HDMI/DP,pcm=7          
/dev/input/event17:     HDA NVidia HDMI/DP,pcm=8          
/dev/input/event18:     HDA NVidia HDMI/DP,pcm=9          
/dev/input/event19:     HD-Audio Generic Front Mic        
/dev/input/event20:     HD-Audio Generic Rear Mic         
/dev/input/event21:     HD-Audio Generic Line             
/dev/input/event22:     HD-Audio Generic Line Out Front   
/dev/input/event23:     HD-Audio Generic Line Out Surround
/dev/input/event24:     HD-Audio Generic Line Out CLFE  
/dev/input/event25:     HD-Audio Generic Front Headphone
```

StoffelCPR, 2024-01-26:
And the controller loses connection after a minute or two

kakra, 2024-01-26:
Sorry to read that the firmware upgrade negatively affected the connection stability. Usually, the later versions improve Bluetooth stability.

You should be able to downgrade the firmware in Windows running this URL from the "run command" dialog:
```
xboxaccessories://firmwareupdate?legacyDowngrade=true
```

It will revert the controller back to version 4.x.

Alternatively, if possible, you could try a CSR-based Bluetooth dongle (e.g. TP-LINK UB400). The AX200 chipset is known to have some problems with Xbox controllers.

Also, since firmware 5.x switches from classic Bluetooth to BLE, it may help to adjust some settings in bluez:
https://github.com/atar-axis/xpadneo/blob/master/docs/TROUBLESHOOTING.md#gamepad-does-not-connect-properly

CaptainCoward, 2024-01-31:
I want to share my Experience on this issue since  i had the very same issue with the 8bitdo Pro 2 controller. (which now has a mechanical defect - not related to the issue.). However when i paired the Pad and then moved the right Stick around... the issue would be gone and not happen in games. Just like you would "re-calibrate" the pad that way.
The TP-LINK UB400  doesn't fix the issue (i own that very adapter).  On my Xbox One  Series X Controller this issue isn't present. Though instead i now seem to have random button presses (very rarly) right after i pressed a button.

kakra, 2024-02-06:
The 8BitDo Ultimate can be switched to xinput mode when using Bluetooth, it then works correctly with xpadneo.

I cannot replicate any of the issues here with my UB400 Bluetooth dongle and the 8BitDo controller - neither phantom inputs nor connection drops. But OTOH, I didn't try with current kernels, so I'll retest this here.

StoffelCPR, 2024-02-06:
All right. Can you tell me ( or link me ) to how I switch to xinput mode for BT? ( The 8BitDo controller works flawlessy on both 2.4GHz and Bluetooth; only that BT is reporting the controller as switch controller which is annoying to me )

The XBox wireless controller is the one with faults.

kakra, 2024-02-06:
Hmm, after doing some search, it seems 8BitDo Ultimate and 8BitDo Pro 2 are two very different controllers. My Pro 2 has 4-position switch next to the battery cover with the letters SADX where X is for x-input mode. All four modes support Bluetooth on this model. From picture, it looks like yours only has a 2-position switch on the back for x-input mode and d-input mode, so it probably supports these only with the 2.4GHz dongle. Other sources say to use SELECT+X for 5 seconds to switch to x-input mode, or SELECT+B for (maybe) Bluetooth mode, but that Bluetooth mode only supports d-input which makes the triggers digital. Or in other words: x-input mode switches to the 2.4GHz receiver and thus connects the controller via USB/GIP mode, and then it should be supported by xpad instead of xpadneo.

I wonder why they are making so vastly different models.

kakra, 2024-02-13:
Can you retry with the latest version?

StoffelCPR, 2024-02-16:
Sure. I can test it next week

closed 2024-04-03 by github-actions[bot]