atar-axis/xpadneo #454: guid button not working proerly

powl999, 2024-01-27:
## Version of xpadneo
v0.9-144-g9b3b696

## Controller Model

- [ ] Xbox One S controller
- [ ] Xbox Elite 2 controller
- [X ] Xbox Series X|S controller
- [ ] Other:

### Connection mode

- [ X] Bluetooth connection
- [ ] USB cable (not yet supported)
- [ ] Xbox Dongle connection (not yet supported)

## Installed Software

- [ ] Anti-Micro (may affect button mappings)
- [ ] OpenRGB (may mess up mappings and rumble stability)
- [ X] Steam Input (enabled by default via Steam Desktop client)
- [ ] Steam Link (usually via Raspberry Pi or other micro computers)
- [ ] devices with QMK firmware (may affect udev rules, similar to OpenRGB)
- [ ] netstick (shares input devices via network similar to Steam Link)
- [ ] xboxdrv (user-space gamepad driver)
- [ ] xone (kernel-space gamepad driver using the Xbox dongle or USB)
- [ ] xow (alternative driver using the Xbox dongle)

## Protocol Information

Please help us identify at which layer the problem can be found if you want
to report mapping errors or if the controller fails to be detected:

- [ ] Steam Proton games are having issues
- [ ] Steam Linux-native games are having issues
    - [ ] I don't use Steam or did not try
- [ ] games running through Lutris, wine and/or Bottles are having issues
    - [ ] I don't use Lutris, Bottles, wine or did not try
- [ ] Linux-native games are having issues
    - [ ] I don't use native games or did not try
- [ X] Other software is having issues (describe software and issues below)
- [ ] Running `evtest` is showing issues (describe the issues below)
    - Keep in mind that `BTN_NORTH` and `BTN_WEST` are intentionally swapped
- [ ] Running `jstest` is showing issues (describe the issues below)
    - [ ] I don't have this tool or don't know how to use it
- [ ] Running `gamepad-tool` is showing issues (post console output below)
    - [ ] I don't have this tool

Please describe how it is failing below in the next sections.

## Severity / Impact

- [ ] I've read the docs and the bug reporting instructions
- [ ] I've applied the latest firmware update to the controller
- [ ] I've tried disabling or running without above mentioned software
- [ X] It does not work at all
- [ ] It used to work in a previous version
- [ ] It mostly works but sometimes it doesn't
- [ ] I found a work-around
- [ ] I probably didn't figure it all out but it's too early to give up
- [ ] I don't know how to ... 
- [ ] It's too complicated
- [ ] Fantastic work but ... 
- [ ] I can code and I want to help

## Describe the Bug
so the problem is the guide button in steam big oicture mode when i use it and i want to open the quick accses menu the only thing that opens is the right steam menue i assume the button press goes through after i realse the guide button i had the same issue on endevour and now on tumbleweed

## Steps to Reproduce
use steam bigpicture mode and try to open the qam

## Expected Behavior
should open by pressing guide and B or A i forgot which button i'm really sure B but either work

## Screenshots / GIFs / Videos

## System Information

                                                                               
 OS: openSUSE Tumbleweed x86_64

```console
# uname -a

```Linux linux.fritz.box 6.7.1-1-default #1 SMP PREEMPT_DYNAMIC Sat Jan 20 15:39:39 UTC 2024 (4959dd8) x86_64 x86_64 x86_64 GNU/Linux

<!-- Paste the output below the line prepended with # -->
```console
# xxd -c20 -g1 /sys/module/hid_xpadneo/drivers/hid:xpadneo/0005:045E:*/report_descriptor | tee >(cksum)

```00000000: 05 01 09 05 a1 01 85 01 09 01 a1 00 09 30 09 31 15 00 27 ff  .............0.1..'.
00000014: ff 00 00 95 02 75 10 81 02 c0 09 01 a1 00 09 33 09 34 15 00  .....u.........3.4..
00000028: 27 ff ff 00 00 95 02 75 10 81 02 c0 05 01 09 32 15 00 26 ff  '......u.......2..&.
0000003c: 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01 81 03 05 01 09  ...u.....%.u........
00000050: 35 15 00 26 ff 03 95 01 75 0a 81 02 15 00 25 00 75 06 95 01  5..&....u.....%.u...
00000064: 81 03 05 01 09 39 15 01 25 08 35 00 46 3b 01 66 14 00 75 04  .....9..%.5.F;.f..u.
00000078: 95 01 81 42 75 04 95 01 15 00 25 00 35 00 45 00 65 00 81 03  ...Bu.....%.5.E.e...
0000008c: 05 09 19 01 29 0c 15 00 25 01 75 01 95 0c 81 02 15 00 25 00  ....)...%.u.......%.
000000a0: 75 01 95 04 81 03 05 0c 0a b2 00 15 00 25 01 95 01 75 01 81  u............%...u..
000000b4: 02 15 00 25 00 75 07 95 01 81 03 05 0c 09 01 a1 01 0a 85 00  ...%.u..............
000000c8: 15 00 26 ff 00 95 01 75 08 81 02 0a 99 00 15 00 26 ff 00 95  ..&....u........&...
000000dc: 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 0a 81 00 15 00  .u.....%...u........
000000f0: 26 ff 00 95 01 75 04 81 02 15 00 25 00 95 01 75 04 81 03 c0  &....u.....%...u....
00000104: 05 0f 09 21 85 03 a1 02 09 97 15 00 25 01 75 04 95 01 91 02  ...!........%.u.....
00000118: 15 00 25 00 75 04 95 01 91 03 09 70 15 00 25 64 75 08 95 04  ..%.u......p..%du...
0000012c: 91 02 09 50 66 01 10 55 0e 15 00 26 ff 00 75 08 95 01 91 02  ...Pf..U...&..u.....
00000140: 09 a7 15 00 26 ff 00 75 08 95 01 91 02 65 00 55 00 09 7c 15  ....&..u.....e.U..|.
00000154: 00 26 ff 00 75 08 95 01 91 02 c0 05 0c 09 01 85 0c a1 01 0a  .&..u...............
00000168: 9e 00 15 00 26 ff 00 95 01 75 08 81 02 0a a1 00 15 00 26 ff  ....&....u........&.
0000017c: 00 95 01 75 08 81 02 0a a2 00 15 00 26 ff 00 95 01 75 08 81  ...u........&....u..
00000190: 02 0a a3 00 15 00 26 ff 00 95 01 75 08 81 02 c0 c0 05 01 09  ......&....u........
000001a4: 06 a1 01 85 05 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81  .........)...%.u....
000001b8: 02 95 01 75 08 81 03 95 06 75 08 15 00 25 65 05 07 19 00 29  ...u.....u...%e....)
000001cc: 65 81 00 c0   

## Controller and Bluetooth Information
<!-- Also follow these steps to create addition information
     about your Bluetooth dongle and connection: -->
standard bluetooth package tumbleweed
<!-- First, disconnect the controller. -->

<!-- Run `sudo btmon | tee xpadneo-btmon.txt` and connect the controller. -->

<!-- Run `dmesg | egrep -i 'hid|input|xpadneo' | tee xpadneo-dmesg.txt`. -->

<!-- Run `lsusb` and pick the device number of your dongle. -->

<!-- Run `lsusb -v -s## | tee xpadneo-lsusb.txt` where `##` is the device
     number picked in the previous step -->

<!-- Attach the resulting files, do not bundle the files into a single
     archive. If some files are too big, gzip them individually. Drag
     and drop the files below. -->

## Additional Context
<!-- Add any other context about the problem here. -->

kakra, 2024-01-28:
Guide+A, +B, +X, +Y is reserved by xpadneo. I will make this behavior optional in the next driver version.

powl999, 2024-01-28:
Thats nice i thougt that i was an Idiot because of this i've tried everything for a short time i thought about this could BE the Problem but i couldnt find the setting now i know it dosent exost thanks for the Help and sorry fory Bad Englisch i'm Not a nativ speaker. Bug thanks dou you know when the next Driver Version gets realesed so Just a rough time frame?

kakra, 2024-01-28:
No plans yet... I tried to release it during the last year but lacking a lot of time. I'll try to get it ready in the next two months.

The feature is mentioned here:
https://github.com/atar-axis/xpadneo?tab=readme-ov-file#profile-switching

There's an ongoing PR that merges mouse emulation, which will also use Guide + a hotkey to switch mouse mode on and off.

If you don't need these functions, the kernel will support your controller out of the box with proper Steam support - no xpadneo needed (this was different some years ago).

powl999, 2024-01-28:
ok thanks

powl999, 2024-01-28:
i have the problem my controller gets recognized by everything exept for steam without the driver

kakra, 2024-01-28:
Interesting... I never heard someone having problem with only Steam detecting the controller.

powl999, 2024-01-28:
Me to thats the Problem i cant find a sulution and i cant Code If u want to See someblogs or Something Just ASK me uonlynmust say me what to do(:

kakra, 2024-01-28:
Try disabling Steam Input controller support in the Steam client and see, if that helps. You don't need it for games to support Xbox controllers, you just need it to remap controls. I'm running most games without it. There are a few games (mostly Playstation ports) which need Steam Input to detect a controller, you can enable that per game in that case.

If it still doesn't work, run `gamepad-tool` (it's part of SDL, I think, but you may need to install it from an extra package). See, if it detects your controller. If it does, be sure to not save any remappings. Steam Input internally uses SDL, so if `gamepad-tool` sees the controller, Steam should, too. Also, Proton internally uses SDL for controllers, or evdev as a fallback. If rawhid is present, it is preferred for some devices.

If even the tool doesn't show your controller, run `evtest` and see, if it lists your controller. Select the controller from the list to test if it sees button presses and axis movements. Proton should see this list of devices as part of its evdev fallback.

rawhid mode is not currently supported with xpadneo (mappings may be wrong if enabled). There's not test utility for it.

Disabling Steam Input forces Proton to use SDL or evdev directly without going through Steam Input virtual controllers. It's the mode xpadneo was originally developed for.

closed 2024-04-22 by kakra

GloriousEggroll, 2024-10-09:
I've found a problem related to this. If you disable profile switching, the QAM menu shortcut works fine on the xbox elite (guide + a) however it does not work for the standard xeries x/s.